Klipper3d/klipper #77: Briefly Worked -- Timeout on Serial Connect -- Maker Slect v2 Melzi Board

recuso, 2017-12-28:
I used a raspberry pi and this how-to to directly flash the hex file to my Maker Select v2: https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview . This allowed me to get around the lack of a bootloader allowing direct USB flashing (from what I understand?)

Anyway, after going through everything it wasn't connecting. I'd just get tons of 

Timeout on Serial Connect

Then, out of the blue:

`Timeout on serial connect
Loaded 51 commands (v0.5.0-56-g1bdebea-20171228_075725-octopi / gcc: (GCC) 4.8.1 binutils: (GNU Binutils) 2.24)
MCU config: ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 SOFT_PWM_MAX=256 SERIAL_BAUD=115200 MCU=atmega1284p STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (1024 moves)
Stats 840.3: gcodein=0 print_time=33.769 buffer_time=0.249 print_stall=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=855 bytes_read=2558 bytes_retransmit=143 bytes_invalid=0 send_seq=93 receive_seq=88 retransmit_seq=9 srtt=0.011 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15995192
Stats 841.3: gcodein=0 print_time=33.769 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=861 bytes_read=2656 bytes_retransmit=143 bytes_invalid=0 send_seq=94 receive_seq=94 retransmit_seq=9 srtt=0.010 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 freq=15994732
Stats 842.3: gcodein=0 print_time=33.769 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.023 mcu_task_avg=0.000146 mcu_task_stddev=0.000553 bytes_write=867 bytes_read=2772 bytes_retransmit=143 bytes_invalid=0 send_seq=95 receive_seq=95 retransmit_seq=9 srtt=0.010 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15994731
Stats 843.3: gcodein=0 print_time=33.769 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.023 mcu_task_avg=0.000146 mcu_task_stddev=0.000553 bytes_write=873 bytes_read=2887 bytes_retransmit=143 bytes_invalid=0 send_seq=96 receive_seq=96 retransmit_seq=9 srtt=0.009 rttvar=0.005 rto=0.028 ready_bytes=0 stalled_bytes=0 freq=15994726
`

It came to life and I was able to control the Z axis. I got carried away an it started bumping into the top of the frame, so I shut it down. I Haven't been able to connect since. I issued the command:

`
Recv: // FIRMWARE_RESTART: Restart firmware, host, and reload config` 

Because it suggested that, but it didn't do anything. It seemed like a fluke that it connected.

I'm using a Maker Select v2 with a melzi 2.0 Board atmega1284p via USB.

chrzr, 2017-12-28:
You could try flashing a bootloader to your Melzi. I did that for the SMelzi (which also did not come with a bootloader  installed) on my CR-10. You just need an Arduino Uno as a flashing device. See: https://youtu.be/ejpSniiJejI
Then you can just use the "avrdude -p atmega1284p -c arduino -P /dev/ttyUSB0 -U out/klipper.elf.hex" in order to flash the firmware, which worked out just fine for me.

KevinOConnor, 2017-12-28:
On Thu, Dec 28, 2017 at 01:59:03AM -0800, recuso wrote:
> I used a raspberry pi and this how-to to directly flash the hex file to my Maker Select v2: https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview . This allowed [...]

To get help with an issue, we need the full klippy.log file (as
described in docs/Contact.md).  Also, check if the steps in the FAQ
item "When I restart my micro-controller the device changes to
/dev/ttyACM1" helps.

-Kevin

recuso, 2017-12-28:
chrzr: Thanks, I already figured as much in terms of flashing a bootloader allowing USB programming. I got around it by using the raspberry Pi as a programmer instead, so it's not a hassle.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1591514/klippy.log)

KevinOConner: Here's the fully klippy log. Not sure why it would magically work that one time...

ls -l /dev/serial/by-id shows my once usb Device at /dev/ttyUSB0.

Thanks for the help - I'm super excited to try to get this working and show others with this printer how to get it going :)

jmichelsen, 2017-12-28:
Hey Recuso,

Would you mind sharing your printer.cfg too? I'm running the same printer, and am encountering similar problems.

How did you use the pi as a programmer? I'd love more detail on what you did specifically. I know this is off-topic for this git issue so maybe we can exchange those details in a different forum

recuso, 2017-12-28:
[klippy.workedonce.log](https://github.com/KevinOConnor/klipper/files/1591538/klippy.workedonce.log)

Here's the log from the time it connected and worked, if that's at all helpful either. I'm trying to read to see if there's something settings-wise I'm missing...

recuso, 2017-12-28:
jmichelsen: I basically did this, but used the klipper hex firmware file instead when it came to the step of uploading to the board.

 https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview

Feel free to message me (does github do messaging? I haven't checked...) to chat about that or anything.  I'm excited about that specifically because I think lots of people would be willing to try updating their firmware on these printers if it were one step simpler.

chrzr, 2017-12-28:
From reading through your logfiles it looks like you did not specify the baudrate in your printer.cfg file. You have to add the desired baudrate to the [mcu] section, otherwise it will just default to 200000.
```
[mcu]
serial = /dev/ttyUSB0
baud: 115200
```

jmichelsen, 2017-12-28:
@chrzr which printer are you using? Just curious

@recuso I'd love to discuss it further and maybe we can simplify the process of flashing for this application specifically (the arduino docs you followed leave a lot of options open to you that in our case are unnecessary and can confuse). My email addy is in my git profile

chrzr, 2017-12-28:
@jmichelsen I am using a Creality CR-10 (non s version) with added bootloader for easy firmware flashing and a z probe for bed leveling.

recuso, 2017-12-28:
chrzr: Huzzah, thank you much! You're right, I just copied the melzi config and didn't add it in. 

Can you or anyone clarify my understanding: are bootloaders included on "newer" printer firmwares (repeteir, klipper) as a standard and hence when you flash one of these you can flash via USB afterwards? It's my understanding that the original limitation of the included firmware was that it didn't have a bootloader, forcing ISP use. But it seems to be taken for granted that flashing a new firmare takes care of that: is that because any other firmware will have a bootloader, or is the bootloader separate/special in some way.

Sorry for not posting my full log immediately -- I did even read the readme, too, heh.

jmichelsen: Ok, I'll try to get to you today. Email me at mine if I don't! It can definitely be simplified into a handful of simple steps. Most of the challenge was sorting through and doing research (reading klipper docs, reading bits of what's discussed about firmware upgrades for this board/printer).

jmichelsen, 2017-12-28:
@recuso I don't see your email in your profile. If you just shoot me an email as an intro, I'll reply with specific questions that I think could get us a good start on a detailed walk through for Wanhao/MakerSelect owners, or anyone without a bootloader.

@chrzr I just got my first 3d printer last month, and figured the MakerSelect was the most bang for my buck (on sale for $250 for the v2.1). Kinda wish I knew the CR-10 existed. That build volume is crazy and it's not much more expensive, plus everyone I talk to that has one loves it

chrzr, 2017-12-28:
@recuso The bootloader is not really a part of the firmware. It is stored in a separate part of the memory. If the manufacturer of the printer did not pre-install the bootloader to the MCU, you won't be able to flash anything to it. That is why you need an external flashing device (serial programmer, arduino, RPi) in order to flash firmwares. The bootloader allows for flashing without such an external device. Luckily you can just burn the bootloader to these chips via the on board serial header!
@jmichelsen Yea, it really is a great printer for the money! Great parts/construction, easy to mod and potentially expandable!

closed 2017-12-28 by recuso

jmichelsen, 2017-12-29:
@recuso contact me when you get a chance :) jmichelsen4@gmail.com