Klipper3d/klipper #78: anet a8

manotroll, 2017-12-28:
has support for anet a8
do you have support for automatic leveling?
Do you have temperature protection?

KevinOConnor, 2017-12-29:
On Thu, Dec 28, 2017 at 08:43:39AM -0800, manotroll wrote:
> has support for anet a8
> do you have support for automatic leveling?
> Do you have temperature protection?

I don't know what electronics are in the anet a8, so I can't be sure.

Automatic bed leveling still needs to be implemented (see the
docs/Todo.md file).

The micro-controller software will detect if a heater is outside the
min_temp/max_temp range and disable the heaters if so.  Some more
advanced detetion of subtle heater failures (eg, the thermistor
falling off the heater) have not yet been implemented.

-Kevin

rewolff, 2017-12-29:
FYI, while building two new 3D printers over the last few months, I've had about three times that the marlin firmware detected something odd going on and went into "safe mode". All false positives, but still it is nice to know that it works. 

Now I must say that Marlin's handling of that situation is far from ideal: It shuts down completely, does not report temperatures anymore, and sometimes a whole reboot of the octopi machine is necessary to get things going again. 

So... learning from that. When such an error condition is detected, keep responding to commands, but make sure that the heaters are in a "safe" mode for a while. Make sure that "manual intervention" happens before you enable them again. So maybe require that the temperature is set to zero before another temperature target can be set?

For context as to when something like this could happen: 
I have an ATX powersupply that powers my machine. Apparently it has a feedback loop on the 5V and not on the 12V line: When the 3D printer loads the 12V line it drops to 11V and even to 10V when both bed and hotend are on.  At 10V my hotend cannot reach the target temperature for my plastic: 225 degrees. A workaround is to load the 5V with one coil of a stepper motor I have.... So when Marlin detects: "I turned on the heater, but it took too long for the temperature to go up", I can reconnect the dummyload and get things to work better. Fault cleared, it would be nice to be able to allow the system to continue without having to reboot.

manotroll, 2017-12-29:
> I don't know what electronics are in the anet a8, so I can't be sure.

Automatic bed leveling still needs to be implemented (see the
docs/Todo.md file).

The micro-controller software will detect if a heater is outside the
min_temp/max_temp range and disable the heaters if so.  Some more
advanced detetion of subtle heater failures (eg, the thermistor
falling off the heater) have not yet been implemented.

right
as soon as the customs finish reviewing the taxes I will test I already have a pi 3 + b ready
the videos I saw on YouTube show an incredible speed
I have completely discharged the marlin

> FYI, while building two new 3D printers over the last few months, I've had about three times that the marlin firmware detected something odd going on and went into "safe mode". All false positives, but still it is nice to know that it works.

Now I must say that Marlin's handling of that situation is far from ideal: It shuts down completely, does not report temperatures anymore, and sometimes a whole reboot of the octopi machine is necessary to get things going again.

So... learning from that. When such an error condition is detected, keep responding to commands, but make sure that the heaters are in a "safe" mode for a while. Make sure that "manual intervention" happens before you enable them again. So maybe require that the temperature is set to zero before another temperature target can be set?

For context as to when something like this could happen:
I have an ATX powersupply that powers my machine. Apparently it has a feedback loop on the 5V and not on the 12V line: When the 3D printer loads the 12V line it drops to 11V and even to 10V when both bed and hotend are on. At 10V my hotend cannot reach the target temperature for my plastic: 225 degrees. A workaround is to load the 5V with one coil of a stepper motor I have.... So when Marlin detects: "I turned on the heater, but it took too long for the temperature to go up", I can reconnect the dummyload and get things to work better. Fault cleared, it would be nice to be able to allow the system to continue without having to reboot.

uses a better source with more amperage or a separate source
set to 13.8V

rewolff, 2017-12-29:
Yeah, for a little more margin, i ordered a 15V, 40A  powersupply. It arrived several minutes after writing the above comment.

manotroll, 2017-12-30:
and solved it?

rewolff, 2017-12-30:
Solved what? 

I explained that I've seen from Marlin. This is intended to improve klipper: The same mistakes should be avoided. 

The supply was advertised with "fan makes a little noise". Which could be interpreted as chinglish for "fan makes little noise: very quiet". However, this seems to be "fan is a little noisy" which is an euphemism for "fan is very loud (and always-on)".

C4g2, 2017-12-30:
Just got my Anet A8 via Sain Smart! The mainboards IC is an Atmel branded ATMEGA1284p AU1720

Any other info you need to confirm compatibility? I'm VERY interested if this could run my machine.

manotroll, 2017-12-31:
> Solved what?

I explained that I've seen from Marlin. This is intended to improve klipper: The same mistakes should be avoided.

The supply was advertised with "fan makes a little noise". Which could be interpreted as chinglish for "fan makes little noise: very quiet". However, this seems to be "fan is a little noisy" which is an euphemism for "fan is very loud (and always-on)".

you did not have a drop problem for 10V and the 15V source arrived if you solved your reset problem

> Just got my Anet A8 via Sain Smart! The mainboards IC is an Atmel branded ATMEGA1284p AU1720

Any other info you need to confirm compatibility? I'm VERY interested if this could run my machine.

C4g2 
I think trying the same ramps settings might be a good way to start
perhaps looking at both electrical schematics of both is a good alternative

rewolff, 2017-12-31:
I'm slowly starting to look into sources of klipper. It seems that the "total shutdown" that I dislike from Marlin is NOT present in klipper. I've found the code to maintain certain commands even when in "error shutdown". Good stuff.

C4g2, 2017-12-31:
Apologies for newbing all over the place, but I thought this was a thread about if the Anet A8 could run the Klipper firmware. Id be more than willing to experiment with my hardware if no ones gotten it working with this model, but Im afraid ill need a bit more instruction. This is my first 3d printer and I've only done enough research to know that Klipper is the firmware I want to run/learn.

In that light:
What same ramp settings? a stock a8? a similar printer that does run klipper thats 'close enough'?
And again, which 'both' schematics?

I very much appreciate the response none the less!

ps. Your original post, 'has support for a8' was that a statement or a question? I think that kicked off my confusion.

KevinOConnor, 2018-01-01:
On Sun, Dec 31, 2017 at 11:52:21PM +0000, C4g2 wrote:
> Apologies for newbing all over the place, but I thought this was a thread about if the Anet A8 could run the Klipper firmware. Id be more than willing to experiment with my hardware if no ones [...]

Earlier, you indicated the electronics used an ATMEGA1284p.  This is a
micro-controller that Klipper supports.  You should be able to follow
the instructions in the docs/Installation.md document to compile and
flash your electronics.

Your next step is going to be putting together a config file for the
printer.  This requires knowing which pins are attached to which
hardware features.  (For example, the x axis step pin might be
"ar54".)  This is pretty straight forward if you've got access to a
schematic for the board, or if you know of some other common firmware
that has support for the board that you can use as a reference.

> In that light:
> What same ramp settings? a stock a8? a similar printer that does run klipper thats 'close enough'?
> And again, which 'both' schematics?

RAMPS is a common type of board (config/generic-ramps.cfg).  I would
verify that your board uses the same pin mapping before trying to use
it though.

-Kevin

rewolff, 2018-01-01:
> Atmel  branded ATMEGA1284p AU1720
That's an atmega-128-4-p-AU . 128 for 128 kb of memory. 4 for the 44-pin variant. p for the newer version that probably runs from 1.8V up to 5V and up to 20Mhz and AU for "TQFP" package. Yes, that should be compatible. 
It was produced 2017, week 20.  (may 14-may 20)

kotfarepra, 2018-01-01:
Very much interested if this would have Anet A8 config/support as well . 
its a replica printer ( prusa mk2 i believe )  and a low price (  ~ 180 USD ) so its popular for entry-level 3D printer  https://www.banggood.com/Anet-A8-3D-Printer-DIY-Kit-1_75mm-0_4mm-Support-ABS-PLA-HIPS-p-1130694.html

rewolff, 2018-01-01:
I've seen even cheaper. (but maybe I didn't look carefully enough that something essential/expensive is missing). 
Yes, it would be nice if someone would create a config file for these. However, this is not easily done "over the internet". You need to be close to the device to reverse-engineer the pinout. 

Kevin, for initial config it would be nice to have some sort of debugging mode. In that mode it would be nice to be able to toggle and pulse arbitrary inputs and outputs to try and find those that control things. As I understand things now, the protocol between the server and the hardware already allows this. Not sure if the /tmp/printer protocol allows it. Then a separate application (web based?) can be used to show all inputs and allow control over all outputs. Click on output "D8" and suddenly the analog value on some analog pin (I don't know which by heart) starts to change and... the bed starts heating up. That sort of stuff. Of course, without feedback this is a bit dangerous, so proper warnings should be issued. ....

C4g2, 2018-01-01:
Thanks for the detailed reply! Knowing that it Will work with the A8 chip is great to hear, very encouraging!

News when Im stuck or have it.

manotroll, 2018-01-01:
in marlin has the ramps settings files and A8 can be a good place to base

lenne0815, 2018-01-01:
Reowulf, theres a working A8 pinout for marlin, pins can just be copied from there.

KevinOConnor, 2018-01-01:
On Mon, Jan 01, 2018 at 12:50:27PM +0000, Roger Wolff wrote:
> Kevin, for initial config it would be nice to have some sort of debugging mode. In that mode it would be nice to be able to toggle and pulse arbitrary inputs and outputs to try and find those that [...]

There is a console tool that can be used to issue low-level commands
(see docs/Debugging.md).  I wouldn't recommend trying to find pins by
trial and error though.

As lenne0815 stated, there appears to be a Marlin config for an anet
board, so I'd start with that.

-Kevin

kotfarepra, 2018-01-02:
Dont know how relevant this might be. Output of Octoprint when connected to printer 

```
2018-01-02 18:06:16,896 - octoprint.util.comm - INFO - Printer reports firmware name "ANET_A8_20160525  http://www.anet3d.com/"
2018-01-02 18:06:16,898 - octoprint.util.comm - INFO - Detected Repetier firmware, enabling relevant features for issue free communication
```

lenne0815, 2018-01-02:
I think to make it work you need to let klipper generate the firmware hex and flash these via atmel studio to the board, maybe someone can confirm this ?

dragonnn, 2018-01-02:
First of - learn how to flash you board for example by flashing Marlin with is official supported. then try to learn how clipper works.

tilsonm, 2018-01-03:
> I think to make it work you need to let klipper generate the firmware hex and flash these via atmel studio to the board, maybe someone can confirm this ?

@lenne0815 
 I managed to compile the hex, pull it to my Windows desktop, and flash the board using atmel studio, but it soft-bricked my board. I was able to re-flash Marlin 1.1.8, though. Tried with avrdude, and same result. 

Here is the output from avrdude.
```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file "C:\Users\tilso\Desktop\klipper.elf.hex"
avrdude: writing flash (13338 bytes):

Writing | ################################################## | 100% 3.78s

avrdude: 13338 bytes of flash written
avrdude: verifying flash memory against C:\Users\tilso\Desktop\klipper.elf.hex:
avrdude: load data flash data from input file C:\Users\tilso\Desktop\klipper.elf.hex:
avrdude: input file C:\Users\tilso\Desktop\klipper.elf.hex contains 13338 bytes
avrdude: reading on-chip flash data:

Reading |                                                    | 0% 0.00savrdude: stk500_loadaddr(): (a) protocol error, expect=0x14, resp=0x49

avrdude: stk500_paged_load(): (a) protocol error, expect=0x14, resp=0x4e
Reading | #                                                  | 1% 0.05savrdude: stk500_cmd(): programmer is out of sync
avr_read(): error reading address 0x0000
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2

avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: safemode: Sorry, reading back fuses was unreliable. I have given up and exited programming mode
avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x2e

avrdude done.  Thank you.
```

tilsonm, 2018-01-03:
@lenne0815 I've successfully flashed my Anet v1.0 with klipper. I have no idea what changed, but after flashing back to Marlin, I tried avrdude again and it flashed flawlessly. I _**think**_ that it worked the first time, but I mistakenly thought that it not being able to read back the flash data and then the white squares on the display that it was soft-bricked.

Testing now.

C4g2, 2018-01-03:
@Tilsonm Thats great to hear, what config file did you use when building the hex? Best of luck on the test!

@Lenne0815 Hey, just recognized your name from Hackaday comments on this firmware, thanks for the videos you posted!

tilsonm, 2018-01-03:
@C4g2 I ran menuconfig with Atmega AVR, atmega1284p, 16MHz, 115200 baud rate. When I flashed the board, I had to modify the flashing baudrate to 57600, and it has to be flashed manually with avrdude or with Atmel Studio, or at least that's the only way I could get it to flash.

tilsonm, 2018-01-03:
Good news, all. I've got klipper working on the stock Anet v1 board and have verified the board configuration. I'm going to do some testing to see if I can work out some "optimal" settings, but it works.
I will try to write up a quick guide in the next few days so others can try to replicate it on their Anet v1 boards.

Quick question: any hopes of the LCD screen working? Has a generic framework been implemented yet?

C4g2, 2018-01-03:
Across my research I think I've found mention of atleast 3 different layouts, yet still called Anet V1 boards, with mainly just the addition of trim pots for some or all of the stepper drivers. The specific board I have has a single trim pot for the Z stepper driver, if anyone has suggestions how to differentiate between them when talking, please help. For now I've been calling mine what is printed on the board, even tho it makes no sense, the "Anet V1-5." One through Five? One dash Five? whatever...

Point is I found a schematic
https://github.com/ralf-e/ANET-3D-Board-V1.0/blob/master/ANET3D_Board_Schematic.png
and have put together what i believe is a correct config file based on this schematic and a few online calculators for step per mm and whatnot. I'll be reporting back with how the first run goes as soon as I have it up.

@tilsonm I thought the 1284p was a 20mhz chip? Thats too bad to hear about the RaspPi flash not working tho.

C4g2, 2018-01-03:
File doesnt wanna attach, and I'd like corrections on it, so here it is:

# This file is for the Anet A8 boards, based on the Anet V1-5 board.
# All versions come populated with the Atmel ATMEGA1284P.
#
# Pin names may be preceded by an '!' to indicate that a reverse
# polarity should be used (eg, trigger on low instead of high). Input
# pins may be preceded by a '^' to indicate that a hardware pull-up
# resistor should be enabled for the pin.

# The stepper_x section is used to describe the stepper controlling
# the X axis in a cartesian robot.
[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC2
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_y]
step_pin: PC7
dir_pin: PC6
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC3
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_z]
step_pin: PB4
dir_pin: PB2
enable_pin: !PA5
step_distance: 0.004
endstop_pin: ^PC4
#position_min: 0
position_endstop: 0.25
position_max: 240
#homing_speed: 5.0
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The extruder section is used to describe both the stepper
# controlling the printer extruder and the heater parameters for the
# nozzle.
[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .002
# The remaining variables describe the extruder heater.
nozzle_diameter: 0.400
filament_diameter: 1.750
#max_extrude_cross_section: 0.64
#max_extrude_only_distance: 50.0
#max_extrude_only_velocity:
#max_extrude_only_accel:
#pressure_advance: 0.0
#pressure_advance_lookahead_time: 0.010
heater_pin: !PD5
#max_power: 1.0
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA7
#pullup_resistor: 4700
#adc_voltage: 5.0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
#pid_deriv_time: 2.0
#pid_integral_max: 1.0
#min_extrude_temp: 170
min_temp: 0
max_temp: 250

# The heater_bed section describes the heated bed.
[heater_bed]
heater_pin: !PD4
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA6
control: watermark
max_delta: 2.0
min_temp: 0
max_temp: 100

#Front Mounted Part Cooling Fan
[fan]
pin: PB4
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

#Side Mounted Extruder Fan
[heater_fan extruder_fan]
pin: PD0
heater: extruder
#heater_temp: 50.0
#fan_speed: 1.0
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

# Micro-controller information.
[mcu]
serial: /dev/ttyACM0
baud: 115200
#pin_map: -Removed 'arduino'
#restart_method: arduino

# The printer section controls high level printer settings.
[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
#max_accel_to_decel: 1500
max_z_velocity: 5
max_z_accel: 100
#motor_off_time: 600
#junction_deviation: 0.02

C4g2, 2018-01-03:
File doesnt want to attach, and I would appreciate corrections, so here it is:

# This file is for the Anet A8 boards, based on the Anet V1-5 board.
# All versions come populated with the Atmel ATMEGA1284P.
#
# Pin names may be preceded by an '!' to indicate that a reverse
# polarity should be used (eg, trigger on low instead of high). Input
# pins may be preceded by a '^' to indicate that a hardware pull-up
# resistor should be enabled for the pin.

# The stepper_x section is used to describe the stepper controlling
# the X axis in a cartesian robot.
[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC2
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_y]
step_pin: PC7
dir_pin: PC6
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC3
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_z]
step_pin: PB4
dir_pin: PB2
enable_pin: !PA5
step_distance: 0.004
endstop_pin: ^PC4
#position_min: 0
position_endstop: 0.25
position_max: 240
#homing_speed: 5.0
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The extruder section is used to describe both the stepper
# controlling the printer extruder and the heater parameters for the
# nozzle.
[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .002
# The remaining variables describe the extruder heater.
nozzle_diameter: 0.400
filament_diameter: 1.750
#max_extrude_cross_section: 0.64
#max_extrude_only_distance: 50.0
#max_extrude_only_velocity:
#max_extrude_only_accel:
#pressure_advance: 0.0
#pressure_advance_lookahead_time: 0.010
heater_pin: !PD5
#max_power: 1.0
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA7
#pullup_resistor: 4700
#adc_voltage: 5.0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
#pid_deriv_time: 2.0
#pid_integral_max: 1.0
#min_extrude_temp: 170
min_temp: 0
max_temp: 250

# The heater_bed section describes the heated bed.
[heater_bed]
heater_pin: !PD4
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA6
control: watermark
max_delta: 2.0
min_temp: 0
max_temp: 100

#Front Mounted Part Cooling Fan
[fan]
pin: PB4
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

#Side Mounted Extruder Fan
[heater_fan extruder_fan]
pin: PD0
heater: extruder
#heater_temp: 50.0
#fan_speed: 1.0
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

# Micro-controller information.
[mcu]
serial: /dev/ttyACM0
baud: 115200
#pin_map: -Removed 'arduino'
#restart_method: arduino

# The printer section controls high level printer settings.
[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
#max_accel_to_decel: 1500
max_z_velocity: 5
max_z_accel: 100
#motor_off_time: 600
#junction_deviation: 0.02

C4g2, 2018-01-03:
No clue what I'm doing wrong when I post this, sorry.

C4g2, 2018-01-03:
# This file is for the Anet A8 boards, based on the Anet V1-5 board.
# All versions come populated with the Atmel ATMEGA1284P.
#
# Pin names may be preceded by an '!' to indicate that a reverse
# polarity should be used (eg, trigger on low instead of high). Input
# pins may be preceded by a '^' to indicate that a hardware pull-up
# resistor should be enabled for the pin.

# The stepper_x section is used to describe the stepper controlling
# the X axis in a cartesian robot.
[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC2
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_y]
step_pin: PC7
dir_pin: PC6
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC3
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_z]
step_pin: PB4
dir_pin: PB2
enable_pin: !PA5
step_distance: 0.004
endstop_pin: ^PC4
#position_min: 0
position_endstop: 0.25
position_max: 240
#homing_speed: 5.0
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The extruder section is used to describe both the stepper
# controlling the printer extruder and the heater parameters for the
# nozzle.
[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .002
# The remaining variables describe the extruder heater.
nozzle_diameter: 0.400
filament_diameter: 1.750
#max_extrude_cross_section: 0.64
#max_extrude_only_distance: 50.0
#max_extrude_only_velocity:
#max_extrude_only_accel:
#pressure_advance: 0.0
#pressure_advance_lookahead_time: 0.010
heater_pin: !PD5
#max_power: 1.0
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA7
#pullup_resistor: 4700
#adc_voltage: 5.0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
#pid_deriv_time: 2.0
#pid_integral_max: 1.0
#min_extrude_temp: 170
min_temp: 0
max_temp: 250

# The heater_bed section describes the heated bed.
[heater_bed]
heater_pin: !PD4
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA6
control: watermark
max_delta: 2.0
min_temp: 0
max_temp: 100

#Front Mounted Part Cooling Fan
[fan]
pin: PB4
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

#Side Mounted Extruder Fan
[heater_fan extruder_fan]
pin: PD0
heater: extruder
#heater_temp: 50.0
#fan_speed: 1.0
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

# Micro-controller information.
[mcu]
serial: /dev/ttyACM0
baud: 115200
#pin_map: -Removed 'arduino'
#restart_method: arduino

# The printer section controls high level printer settings.
[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
#max_accel_to_decel: 1500
max_z_velocity: 5
max_z_accel: 100
#motor_off_time: 600
#junction_deviation: 0.02

C4g2, 2018-01-03:
Thank you for the clarification, that could have been a small headache to figure out! My last post just doesnt wanna work for me. I tried a few things, including the code button, before giving up and blaming it on my mac running an old version of chrome. But thank you for trying.

I'll try flashing with the rasppi just to make sure it wasnt a problem with tilsonm's specific hardware, if it still fails, avrdude then load this mess of a config file and give it a go!

C4g2, 2018-01-03:
One last try at config file:

Anet_A8.cfg
```
# This file is for the Anet A8 boards, based on the Anet V1-5 board.
# All versions come populated with the Atmel ATMEGA1284P.
#
# Pin names may be preceded by an '!' to indicate that a reverse
# polarity should be used (eg, trigger on low instead of high). Input
# pins may be preceded by a '^' to indicate that a hardware pull-up
# resistor should be enabled for the pin.

# The stepper_x section is used to describe the stepper controlling
# the X axis in a cartesian robot.
[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC2
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_y]
step_pin: PC7
dir_pin: PC6
enable_pin: !PD6
step_distance: 0.15385
endstop_pin: ^PC3
#position_min: 0
position_endstop: 0
position_max: 220
homing_speed: 25
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The stepper_z section is used to describe the stepper controlling
# the Z axis in a cartesian robot.
[stepper_z]
step_pin: PB4
dir_pin: PB2
enable_pin: !PA5
step_distance: 0.004
endstop_pin: ^PC4
#position_min: 0
position_endstop: 0.25
position_max: 240
#homing_speed: 5.0
#homing_retract_dist: 5.0
#homing_positive_dir: False
#homing_stepper_phases: 0
#homing_endstop_accuracy: 0.200
#homing_endstop_phase: 0
#homing_endstop_align_zero: False

# The extruder section is used to describe both the stepper
# controlling the printer extruder and the heater parameters for the
# nozzle.
[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .002
# The remaining variables describe the extruder heater.
nozzle_diameter: 0.400
filament_diameter: 1.750
#max_extrude_cross_section: 0.64
#max_extrude_only_distance: 50.0
#max_extrude_only_velocity:
#max_extrude_only_accel:
#pressure_advance: 0.0
#pressure_advance_lookahead_time: 0.010
heater_pin: !PD5
#max_power: 1.0
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA7
#pullup_resistor: 4700
#adc_voltage: 5.0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
#pid_deriv_time: 2.0
#pid_integral_max: 1.0
#min_extrude_temp: 170
min_temp: 0
max_temp: 250

# The heater_bed section describes the heated bed.
[heater_bed]
heater_pin: !PD4
sensor_type: NTC 100K beta 3950
sensor_pin: ^PA6
control: watermark
max_delta: 2.0
min_temp: 0
max_temp: 100

#Front Mounted Part Cooling Fan
[fan]
pin: PB4
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

#Side Mounted Extruder Fan
[heater_fan extruder_fan]
pin: PD0
heater: extruder
#heater_temp: 50.0
#fan_speed: 1.0
#max_power: 1.0
#hard_pwm: 0
#kick_start_time: 0.100

# Micro-controller information.
[mcu]
serial: /dev/ttyACM0
baud: 115200
#pin_map: -Removed 'arduino'
#restart_method: arduino

# The printer section controls high level printer settings.
[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
#max_accel_to_decel: 1500
max_z_velocity: 5
max_z_accel: 100
#motor_off_time: 600
#junction_deviation: 0.02
```

manotroll, 2018-01-03:
how many people testing and I still do not have my printer to help with the tests: \

how to solve the recording problem since the board runs at 20mhz had to change something in arduino IDE?

lenne0815, 2018-01-03:
@tilsonm Theres neither need nor plans to implement the lcd displays afaik. Look at Printoid for a great gui instead.

tilsonm, 2018-01-04:
@C4g2 I suggest doing an ls -l /dev/serial/by-id and getting the exact path to your printer. I turn my printer off but leave my Pi on, and when I restart the symlinked /dev/tty* path has changed. Oddly, mine is /dev/ttyUSB*, not /dev/ttyACM*

See section from FAQ for setting the address more concretely
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md

C4g2, 2018-01-04:
Will do, mine is /ttyUSB0 as well, and thank you for posting your config file! Copied most your axis measurements & offsets (Z isn't ~240?), and also helped me notice a typo at my z step pin.

About the sensor type; I've not found much info on what comes stock in this MK8 extruder, I did find more than one place mentioning the NTC 3950, but that could of been meant as an upgrade from stock equipment.
If Im reading the schematic correctly, both of the sensors have 4.7k pull ups on them. But you confirmed End and Bed temp both reporting correctly?

Anyone else lurking this thread now or later, if you're looking to program your printers board and dont have a programmer, apparently you can [program your AVR via RaspPi's GPIO] (https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview) have yet to try it, but it looks solid.

C4g2, 2018-01-04:
I can also confirm that make flash does not work via rasppi. Tried both /dev/ttyUSB0 and the /by-id/**** at 250k 115k and 57k, regardless, printer restarts then times out.

Is this a Bootloader issue, or something else, hopefully more easily fixed?

rewolff, 2018-01-04:
"make flash" tries to program the MCU in the printer. Normally, a program like "AVRDUDE" talks to "programming hardware", which then programs the target CPU. The original thinking in "arduino" was that an arduino would be simple because you would not need a separate programmer. This was accomplished by burning a program into the arduino that would PRETEND to be a programmer, while infact programming the CPU itself. So ESP8266 and STM32 have a boot loader protocol that is different from what hardware progammer modules use, but here the bootloader simply emulates being a programmer. 

On my system this worked: my MCU was tricked to boot into the bootloader, and my pi talked to the MCU in the "programmer-protocol" while the bootloader pretended to be programming a different MCU while in fact it was self-programming. 

This works because my board is actually an arduino with a bootloader. But if you are selling hundreds of A8 printers, you get to chose: "Do we program the bootloader AND the default firmware, or do we just program the firmware?" 

You can bet your ass that the choice was made to program just the firmware. 

So when you issue "make flash" the makefile will try to program your MCU by trying to get it into the bootloader (not present, so it will immediately boot back into the default firmware) and then try to talk to it using the bootloader/programmer protocol. That'll time out.

C4g2, 2018-01-04:
So first thank you for the answer! But that leads me to the question is it worth putting a 'standard' bootloader on there? Is there even such a thing? Cause if klippers firmware portion is gonna be undergoing many revisions, it sounds like it might be worth the time to get the rasppi flashing it properly. I know this is almost outside the scope of klipper, but im curious : )

KevinOConnor, 2018-01-04:
On Thu, Jan 04, 2018 at 02:40:23AM -0800, C4g2 wrote:
> So first thank you for the answer! But that leads me to the question is it worth putting a 'standard' bootloader on there? Is there even such a thing? Cause if klippers firmware portion is gonna be [...]

Yes.  If your board did not come with a bootloader installed, then you
will need to use an external programmer to program the board.  In that
process, you should write a bootloader to the board (there is no
downside to having a bootloader).  Then it should be possible to
program future firmware updates without requiring an external
programmer.

-Kevin

tilsonm, 2018-01-05:
@C4g2, lenne0815 reminded me today that I never shared my "secret formula" of avrdude commands to flash the hex file to my Anet board.

Here goes (written as if I was doing this again for the first time):

1. ssh into pi using PuTTY.
2. `git clone https://github.com/KevinOConnor/klipper`
3. `./klipper/scripts/install-octopi.sh`
4. `cd ~/klipper/`
5. `make menuconfig`
6. Select options: MCU = Atmega AVR; Processor Model = atmega1284p; Processor speed = 16MHz; Baud Rate = 115200. Save using default file name suggested by utility. Exit utility.
7. `make`
8. `sudo service klipper stop`
### **Here's where the REAL magic starts**
9. **_` avrdude -v -p atmega1284p -c arduino -P /dev/ttyUSB# -b 57600 -D -U flash:w:/home/pi/klipper/out/klipper.elf.hex:i`_**

Explanation of arguments (to the best of my ability):
- avrdude calls avrdude
- -v means the program will display 'verbose' feedback as it programs the board
- -p means the "part type" (board model, which in this case is atmega1284p. It should match the make menuconfig.)
- -c means the programmer used. In this case it's arduino. All the research I can find says it should be avrispmkII, but I got the terminal command from ArduinoIDE on a successful flash of Marlin and even though I selected avrispmkII in ArduinoIDE, the command programmed with arduino.
- -P means the path to your device. In my case it's /dev/ttyUSB0 or /dev/ttyUSB1 (it changes each time I power up my printer). I've read that most cheap Chinese boards will show up as ttyUSB#.. 
- -b mean baud rate. Even though the printer communicates at 115200, it apparently gets programmed at 57600. Another parameter that I got from ArduinoIDE.
- -D means disable auto erase for flash memory - from what I've read, it will prevent the programmer from wiping the board if the flash fails for some reason.
- -U means a memory operation, in this case "flash memory, write file /home/pi/klipper/out/klipper.elf.hex in hex file format" [[http://www.nongnu.org/avrdude/user-manual/avrdude_4.html]]

Then you can run `sudo service klipper start` and follow the rest of Kevin's instructions in the Installation page.

C4g2, 2018-01-05:
@Tilsonm Awesome guide, the flag explanations help with whats in man avrdude. I would advise anyone like me new to flashing chips that the Anet board is not particularly friendly with its programmer header; upside down and only the middle 6, good [pics](https://discuss.toms3d.org/anet-a8-cloned-prusa-tronxy-etc-f19/things-to-think-about-if-you-are-going-to-buy-a-an-t208.html) and other sage advice about this printer there too.

@Kevin Programming the board doesnt seem to be any problem, but might as well do it "hard" once and easy the rest. Optiboot was the first loader I found, hes looking for folk with the 1284p to test it, but I have no idea what to look for in terms of requirments or compatibility, essentially if Optiboot is even an option for this project.

Other news, I got the link I posted previously about AVR programming with RaspPi GPIO pins working, and just had my first success at a -v and getting a response from the chip with a 40-pin ribbon cable and 6 dupont male to female wires. Figured I'd wait to find something like Tilsonm's guide, thanks again, before I finally flashed anything. Now I think I'll try a bootloader option first and see if the rasppi can do this via usb.

Any suggestions?

hg42, 2018-01-05:
you can try any boot loader, you can always flash another.

You eventually need to program the fuses, because different boot loaders often stsrt st different addresses.
https://embedderslife.wordpress.com/2012/08/20/fuse-bits-arent-that-scary/
http://www.engbedded.com/fusecalc/

You can fake to flash a boot loader with the arduino ide. This will output command lines in the terminal that you can use as orientation, which values you need. You have to choose the correct board and cpu. Eventually you need to install additional board managers for this.

These command lines are for the Arduino standard boot loader, so you may have to change the fuses for the start address. You can see which meaning the values have and also change options and calculate new values with the fusecalc in the second link.

rewolff, 2018-01-05:
> -c means the programmer used. In this case it's arduino. All the research I can find says it should be avrispmkII, but I got the terminal command from ArduinoIDE on a successful flash of Marlin and even though I selected avrispmkII in ArduinoIDE, the command programmed with arduino.

I explained that the bootloader in the arduino pretends to be a programmer. Well, my memory for such details is bad, but that programmer is apparently 'avrispmkII'. When you specify "arduino" as the programmer, then avrdude will follow the AVRISPMKII protocol, but before starting the protocol avrdude will issue a few tricks that will cause an arduino to restart-into-bootloader. If your anet responds to this with a succesfull flash, you've proven me wrong in that there IS a bootloader in your board. :-) 

>  In my case it's /dev/ttyUSB0 or /dev/ttyUSB1 (it changes each time I power up my printer). 

The Linux kernel will chose the first free /dev/ttyUSBxx with xx being a number starting at 0. Now if you unplug the printer while it is connected as "ttyUSB0", it is likely that SOME program is still holding a filedescriptor to that device. So at that moment, the /dev/ttyUSB0 device file will be removed, but the index "0" is not yet marked as free. If you replug the device, it will get index 1: /dev/ttyUSB1. At that moment you  notice something isn't working and you reconfigure klipper to use /dev/ttyUSB1 and it releases the filedescriptor and the index becomes free again.

rewolff, 2018-01-05:
> (there is no downside to having a bootloader).  
There is **very little** downside to having a bootloader.... The small downside is that you lose a few bytes of your flash memory to the bootloader. With the firmware fitting easily in what's left, for clipper there is indeed no downside.... :-)

tilsonm, 2018-01-05:
@rewolff that makes sense. What you said the first time didn't quite click, but thanks for repeating it. It makes more sense now.

How can I fix the ttyUSB#. I thought the suggestion I made to c4g2 about ls -l /dev/serial/by-id would take care of it, but it still seems confused when I power my printer off and back on and switches between USB0 and USB1. I've tried symlinking to /dev/anetv1 by using USB characteristics, but for some reason it's still not recognizing and automatically adding that symlink. Or maybe I'm going about this all wrong.

KevinOConnor, 2018-01-05:
On Fri, Jan 05, 2018 at 12:41:14PM +0000, tilsonm wrote:
> How can I fix the ttyUSB#. I thought the suggestion I made to c4g2 about ls -l /dev/serial/by-id would take care of it, but it still seems confused when I power my printer off and back on and [...]

Did you follow the directions at:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

If so, can you provide the klippy.log file with the failure (along
with `ls -l /dev/serial/by-id/` after the failure).

-Kevin

KevinOConnor, 2018-01-05:
On Thu, Jan 04, 2018 at 04:22:05PM -0800, tilsonm wrote:
> @C4g2, lenne0815 reminded me today that I never shared my "secret formula" of avrdude commands to flash the hex file to my Anet board.
> 
> Here goes (written as if I was doing this again for the first time):
[...]

Thanks.  I've updated the Klipper build so that it will always use "-c
arduino" for the atmega1284p (commit 5a86391f).  So, once one gets a
bootloader on the board, it should be possible to use the standard
"make flash" instructions to flash Klipper.

I don't have the board, but this link seems helpful for getting a
bootloader installed:
http://www.instructables.com/id/Flashing-a-Bootloader-to-the-CR-10/

-Kevin

C4g2, 2018-01-05:
@kevin Thanks, good link too, I found the [bootloader](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot/optiboot_atmega1284p.hex) the talking about and judging by the file name, it looks to support the 1284p!

So the .hex is what I flash with avrdude using `-U boot:w:~/optiboot_atmega1284p.hex:i`?  There is also a .lst file in the parent folder of the .hex that has the exact same name, do I need to do anything with that?

C4g2, 2018-01-05:
I was thinking about Tilsonm's last post, I'm not at home right now so I cant try this myself, but I was thinking about symlinking the /dev/serial/by-id to /dev/anet? udev should have a rules file you could try hardcoding it in if you're brave!

tilsonm, 2018-01-06:
@c4g2 I tried using udev and couldn't get it to work. Lol. You might have better luck.

rewolff, 2018-01-06:
> There is also a .lst file in the parent folder of the .hex that has the exact same name, do I need to do anything with that?

No. The .lst will list where things have ended up in memory. i.e. if you declare a variable "int steps_to_make", the LST will describe what address in memory is now used for storing that variable.

C4g2, 2018-01-06:
So I've been reading about fuses and locks on the chip and I have a feeling this isn't something I'm gonna immediately pick up on and feel comfortable just doing, but I'm far from quitting. I have found enough to come up with this monstrosity: `sudo avrdude -p atmega1284p -C ~/avrdude_rpgpio.conf -c rpgpio -v -v -U lock:w:0x3F:m -U lfuse:w:0xFF:m -U hfuse:w:0xDE:m -U efuse:w:0xFD:m -U flash:w:~/optiboot_atmega1284p.hex:i -U lock:w:0x2F:m`

The RPGPIO is the nonsense I set up to flash the chip, both -C and -c should be changed by anyone not using the Raspberry Pi to program. -v -v, for very verbose, cause I like when text scrolls by fast.

The next bit Avrdude will preform all -U commands in the order presented it. So;

`-U lock:w:0x3F:m` unlocks the chip and allows fuses to be written.
`-U lfuse:w:0xFF:m` deals with clock settings? The options here are frankly overwhelming, I have yet to find anything that says other than 0xFF for the 1284p, but would appreciate conformation on this one as it seems to be the most dangerous fuse to mess with.
`-U hfuse:w:0xDC:m` should be setting how large the bootloader reserve is. Optiboot is ~702 words, 1.37kB, needing 1024 words of space, seems straight forward.
`-U efuse:w:0xFC:m` seems to be used exclusively for brownout detection, 4.3v for a +5 board, intuitive, unless its not?
-U flash is where Optiboot gets loaded onto the chip...
`-U lock:w:0x2f:m` locks fuses and was a suggestion by an optiboot author so that anything flashed after can read the bootloader area and report on it.

This may be gross oversimplification, but hopefully not so much as to be wrong.

C4g2, 2018-01-06:
So I've read up on fuses and locks enough to know I wont be immediately picking up on this and also wont feel comfortable just doing it, that said I havnt given up. I found enough detail to put together this monstrosity: `sudo avrdude -p atmega1284p -C ~/avrdude_rpgpio.conf -c rpgpio -b 57600 -v -v -U lock:w:0x3F:m -U lfuse:w:0xFF:m -U hfuse:w:0xDC:m -U efuse:w:0xFC:m -U flash:w:~/optiboot_atmega1284p.hex:i -U lock:w:0x2F:m'

The RPGPIO nonsense at -C and -c is for the rasppi programmer I'm using. -b 57600 as per tilsonm. And -v -v for very verbose, cause I like when text scrolls past quickly. Avrdude will process the following -U in the order presented it.

`-U lock:w:0x3F:m` Unlocks the chip or area to write the bootloader.
`-U lfuse:w:0xFF:m` Sets the clock. This is frankly the most overwhelming list I've seen in quite a while. Fortunately I have yet to find a guide that says anything other than 0xFF, still, considering the danger when messing with this fuse I'd like conformation before setting.
`-U hfuse:w:0xDC:m` Seems to be reserving space for the bootloader? Optiboot is 1.37kB, ~702 words, needing 1024 words.
`-U efuse:w:0xFC:m` This one looks to exclusively deal with the brownout detection? +5v means this should be set to 4.3v yes?
-U flash:w to load the Optiboot code on the chip...
`-U lock:w:0x2F:m` Locks again. The 0x2F was suggested by an Optiboot author so anything flashed after can read the boot area.

This may be a gross oversimplification, but hopefully not enough to be wrong.

C4g2, 2018-01-06:
So I've read up on fuses and locks enough to know I wont be immediately picking up on this and also wont feel comfortable just doing it, that said I havnt given up. I found enough detail to put together this monstrosity: `sudo avrdude -p atmega1284p -C /avrdude_rpgpio.conf -c rpgpio -b 57600 -v -v -U lock:w:0x3F:m -U lfuse:w:0xFF:m -U hfuse:w:0xDC:m -U efuse:w:0xFC:m -U flash:w:/optiboot_atmega1284p.hex:i -U lock:w:0x2F:m`

The RPGPIO nonsense at -C and -c is for the rasppi programmer I'm using. -b 57600 as per tilsonm. And -v -v for very verbose, cause I like when text scrolls past quickly. Avrdude will process the following -U in the order presented it.

-U lock:w:0x3F:m Unlocks the chip or area to write the bootloader.
-U lfuse:w:0xFF:m Sets the clock. This is frankly the most overwhelming list I've seen in quite a while. Fortunately I have yet to find a guide that says anything other than 0xFF, still, considering the danger when messing with this fuse I'd like conformation before setting.
-U hfuse:w:0xDC:m Seems to be reserving space for the bootloader? Optiboot is 1.37kB, ~702 words, needing 1024 words.
-U efuse:w:0xFC:m This one looks to exclusively deal with the brownout detection? +5v means this should be set to 4.3v yes?
-U flash:w to load the Optiboot code on the chip...
-U lock:w:0x2F:m Locks again. The 0x2F was suggested by an Optiboot author so anything flashed after can read the boot area.

This may be a gross oversimplification, but hopefully not enough to be wrong.

mabl, 2018-01-06:
@KevinOConnor Thanks for this fix (5a86391). However the ANET board also needs its baud rate set to 57600, which is not currently menuconfig option, correct? It might make sense to make this configurable too. When doing so, I can successfully flash klipper on my ANET board.

KevinOConnor, 2018-01-06:
On Sat, Jan 06, 2018 at 03:17:44PM +0000, Matthias Blaicher wrote:
> @KevinOConnor Thanks for this fix (5a86391). However the ANET board also needs its baud rate set to 57600, which is not currently menuconfig option, correct? It might make sense to make this configurable too. When doing so, I can successfully flash klipper on my ANET board.

Okay, thanks.  Can you run the following commands and copy-and-paste
the output for me?

cd ~/klipper/
avrdude -patmega1284p -carduino -P/dev/ttyUSB0 -D -Uflash:w:out/klipper.elf.hex:i
avrdude -patmega1284p -carduino -b57600 -P/dev/ttyUSB0 -D -Uflash:w:out/klipper.elf.hex:i

KevinOConnor, 2018-01-06:
On Sat, Jan 06, 2018 at 03:17:44PM +0000, Matthias Blaicher wrote:
> @KevinOConnor Thanks for this fix (5a86391). However the ANET board also needs its baud rate set to 57600, which is not currently menuconfig option, correct? It might make sense to make this configurable too. When doing so, I can successfully flash klipper on my ANET board.

Which bootloader did you flash to your ANET board?  Looking at the
directions from:

http://www.instructables.com/id/Flashing-a-Bootloader-to-the-CR-10/

it recommends writing the bootloader from:

http://lauszus.github.io/Sanguino/

which has a boards.txt file with an unpload speed of 115200:

https://github.com/Lauszus/Sanguino/blob/master/avr/boards.txt

-Kevin

lucashpandolfo, 2018-01-06:
¿Does the Anet A8 uses a Melzi too?

I have a Tronxy p802ma (also uses Melzi) and also had some problems flashing the firmware.

First, as someone else said, it appears the only baudrate supported is 57600. Surprisingly enough, just changing the baudrate worked for me and allowed me to flash the firmware. I later reflashed the original firmware to test something. When I wanted to flash Klipper again, I couldn't do it. I tried for hours. In the end I had to reflash a bootloader.

Using my raspbery and [this guide](https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview) I successfully flashed a bootloader to the Melzi ([this one from a guide at reprap.org](https://github.com/reprappro/Marlin/blob/master/sanguino/bootloaders/atmega/ATmegaBOOT_1284P.hex) ) but for some reason the Melzi still refused to let me flash the firmware.

In the end I flashed [this other bootloader](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot/optiboot_atmega1284p.hex) and it worked flawlessly. But again, I could not get it working with another baudrate besides 57600.

[49 more replies]

closed 2018-01-21 by KevinOConnor