Klipper3d/klipper #79: Does klipper work with all ARM microcontrollers ?

kuthedk, 2017-12-28:
will Klipper work with all ARM microcontrollers? For example, http://www.st.com/en/microcontrollers/stm32f103c8.html that one that is used on the MonoPrice Select Mini v2? I know this isn't really an issue but I couldn't really find a better place to ask this if there is a better place to ask this please direct me to it.

Thanks!

KevinOConnor, 2017-12-29:
On Thu, Dec 28, 2017 at 11:09:17AM -0800, kuthedk wrote:
> will Klipper work with all ARM microcontrollers? For example, http://www.st.com/en/microcontrollers/stm32f103c8.html that one that is used on the MonoPrice Select Mini v2? I know this isn't really [...]

Klipper needs to be ported to each type of micro-controller.  That
particular micro-controller is not currently supported.

-Kevin

rewolff, 2017-12-29:
FYI, I'm a fan of the STM32 family of processors, so I am interested in getting this working.

kuthedk, 2017-12-31:
out of curiosity @KevinOConnor , how difficult/easy of a process would that be, I'm fairly capable of making changes and even writing the code to do so, just kinda want to know how big of a task that would be.

KevinOConnor, 2017-12-31:
On Sat, Dec 30, 2017 at 04:44:08PM -0800, kuthedk wrote:
> out of curiosity @KevinOConnor , how difficult/easy of a process would that be, I'm fairly capable of making changes and even writing the code to do so, just kinda want to know how big of a task that would be.

If you've never done that sort of thing, I'd say it is high
difficulty.  It requires reading through the register level
specification for the processor and programming the low-level devices.
It also requires the right equipment and tricks to debug a platform at
a low-level.  If you've done a few processor ports then it's not too
bad to do another for Klipper - Klipper only requires a handful of
architecture specific definitions (console, timer, gpios).

-Kevin

closed 2018-01-04 by KevinOConnor