Klipper3d/klipper #80: Z Axis Steps too big -- How to determine value? (Also extruder inverted)

recuso, 2017-12-28:
[klippylog.txt](https://github.com/KevinOConnor/klipper/files/1592013/klippylog.txt)

So I finally got around my other issues and was able to get klipper to start printing with my Maker Select v2 (melzi). 

The first two issues I encountered were: heated bed not responding, but reading temp (I must need to change the pin?) and the extruder was operating in reverse. I resolved the extruder issue temporarily by reversing the motor wiring, but would inverting the pin (using !) in the printer.cfg be the solution to that?

Most importantly, though:

It takes huge steps between layers. Like 2mm or something -- just straight up printing air.

I'm using the suggested step in the config:

`step_distance = 0.00025`

How do I go about figuring out why it's taking such huge steps or what the "right" steps are? I'm trying to figure out how to google the correct terms to find out the proper heated bed pin and the correct step_Distance setting, anyone have an insight?

Thanks!

chrzr, 2017-12-28:
I actually ran into the same issues when setting up my printer yesterday! A really good resource for the pin bindings etc. are the pins_boardname.h (e.g. pins_melzi.h) files in the Marlin firmware! (see: https://github.com/MarlinFirmware/Marlin/tree/1.1.x/Marlin ) There you can just crosscheck with the bindings in your printer.cfg!

1) The pin for the heater is probably set wrong. Just check with the pins_boardname.h file in Marlin!
2) For the extruder running in reverse you should just add a ! to the specified pin in order to reverse the direction.
3) The steps/unit can also be found in the stock Marlin firmware for your printer. You can just calculate the step distance for your printer.cfg by: step distance = 1/steps per unit.
Here are some values I found in a Marlin Version for your printer:

```
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 100 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 100, 100, 5, 25 }
```

source: https://www.reddit.com/r/3Dprinting/comments/6bm1zg/help_with_configuring_marlin_for_mp_maker_select/

I also attached the printer.cfg from my CR-10/SMELZI for reference. Maybe the settings also work out for you.
[printercfg.txt](https://github.com/KevinOConnor/klipper/files/1592066/printercfg.txt)

lenne0815, 2017-12-28:
Pins:
Pins are configured a bit different from the common Marlin / Repetier scheme:

ar6 f.e. is pin d6

^ar6 is pin6 with pullup if available

!ar6 is active low / high ( can be used to reverse steppers f.e. if you change the dir pin )

recuso, 2017-12-29:
I got it working using this wiring shematic:

http://reprap.org/mediawiki/images/7/7d/Melzi-circuit.png

I had to change the default z stepper distance from .00025 to .0025 and change the hotbed pin from PD2 to PD4 (from the supplied melzi config). Thanks!

closed 2017-12-29 by recuso