Klipper3d/klipper #81: enhancement: Support configurable PWM rates on Extruder/Bed PID controllers in RAMPS-type config

andrewsil1, 2017-12-29:
I have a printer (Felixprinters Tec4) which uses a cloned RAMPS 1.4 controller and Repetier firmware as "Stock."  After a few hours of tinkering I was successful in getting good initial prints from Klippy. However, this printer needs a higher PID PWM rate - 

On this particular printer, PWM for the bed is a must in order to control temp more tightly, as otherwise bed expansion/contraction causes waves in layer-to-layer Z-height when using bang-bang type heat management. However, at the low PWM rate being used by the PID code in most firmware, PWM rate there is a fairly large voltage sag (visible flicker in the 12V power supply LEDs, and noise (slight random instability) in the measured temps from the extruder and the bed, as well as sporadic shutdowns of the board.  In the stock firmware release notes, there is the following note: 
"Updated frequency from 16 hz to 128hz. This fixed also the bug where PID control at 70% dutycycle caused the control board to turn off, when stepper drivers are turned off."  

When this change was made in the Repetier fork for the Felix printer, the voltage droop, flicker, and noise in temp readings pretty much vanished.  

Would it be possible enable a similar feature in Klipper to raise the PWM rate for PID?

KevinOConnor, 2017-12-29:
On Fri, Dec 29, 2017 at 03:42:13AM +0000, Andy Silverman wrote:
> I have a printer (Felixprinters Tec4) which uses a cloned RAMPS 1.4 controller and Repetier firmware as "Stock."  After a few hours of tinkering I was successful in getting good initial prints from [...]

It's not currently configurable.  As a test, you could change the code
- in klippy/heater.py, change the line with "PWM_CYCLE_TIME = 0.100"
to 0.010 and then do a manual restart (sudo service klipper restart).
Also, be sure to set the heater_bed max_power field if that's
necessary.

If you need faster than 10ms PWM rates then hardware pwm is
necessary.  That's also not difficult, but would require some further
code changes.

-Kevin

andrewsil1, 2017-12-29:
Since filing the request I took a look at the code and found the variable you mentioned.  To get to 128hz would require a 7ms+change rate, which is what I set it to in order to try it out, and it’s definitely much happier.  If it’s actually doing 10ms even though I set it to 7, then it’s good enough for the purpose anyway – it’s working much better at this setting than the default value.

Either I can keep it as a private fork, or if you’re interested I suppose I could work on wiring up the necessary config file changes to support setting it there (and documentation in the example.cfg) and submit a pull.

What do you think?

KevinOConnor, 2017-12-29:
On Fri, Dec 29, 2017 at 04:37:57AM +0000, Andy Silverman wrote:
> Since filing the request I took a look at the code and found the variable you mentioned.  To get to 128hz would require a 7ms+change rate, which is what I set it to in order to try it out, and it’s [...]

If you set it to 7ms, then it will be 7ms, and that's fine.  Much
faster and it adds processing load to the MCU.

> Either I can keep it as a private fork, or if you’re interested I suppose I could work on wiring up the necessary config file changes to support setting it there (and documentation in the example.cfg) and submit a pull.
> 
> What do you think?

I think the pwm max_power/cycle_time/hard_pwm parameters that are
implemented in fan.py and chipmisc.py need to be refactored so that
the same code can be used in heater.py.  Not sure the best way to do
that.  I've made a note on my local list to take a look.  You're
welcome to jump in if you want.

-Kevin

KevinOConnor, 2018-01-04:
commit b22a81cd

closed 2018-01-04 by KevinOConnor