Klipper3d/klipper #83: custom kinematics

ekaggrat, 2017-12-29:
hi, 
i am intending on using klipper for gus simpson 1.5. i like the idea of klipper being very easy to flip between different kinematics with just the config file. Can you guide me as to exactly how to replace the equations in the delta.py for a custom one? 

It is a difficult for me to pin point the exact function for it as my programming skills are very noobish. 
thanks.

KevinOConnor, 2018-01-04:
Adding custom kinematics is a big task that will require software development - it's not something that can be described.

closed 2018-01-04 by KevinOConnor

ekaggrat, 2018-01-05:
i want to run it with gus simpson which has the kinematics very close to a linear delta.. just a few equations need to be changes.. people have done it for marlin and smoothie without much modifications to the firmware.. smoothie is the easiest to change or make custom kinematics. . thanks