Klipper3d/klipper #87: Make Flash Failure on Rev B Beaglebone

benwis, 2017-12-30:
Hi,
I tried to install Klipper on the Rev B Beaglebone Black, and I got this error. Any chance you can advise on a fix? It's being run on the latest debian image.

benwis@beaglebone:~/klipper$ make flash
  Flashing
[sudo] password for benwis:
Installing firmware to /lib/firmware/
Attempting PRU restart...
[   78.775588] bone_capemgr bone_capemgr: slot #5: override
[   78.801454] bone_capemgr bone_capemgr: slot #5: 'Override Board Name,00A0,Override Manuf,BB-ADC'
[   78.836133] bone_capemgr bone_capemgr: slot #5: BB-ADC conflict P9.39 (#4:BB-ADC)
[   78.861886] bone_capemgr bone_capemgr: slot #5: Failed verification
benwis@beaglebone:~/klipper$ cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-ADC

BlackStump, 2017-12-30:
You are using a BBB to control what sort of MCU ?
 because it looks to me that you have a ADC cape installed.
Have you read the appropriate docs?
Installation checks.
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md
FAQ
 https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md
and contact page.
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

you need to supply the klippy.log

KevinOConnor, 2017-12-30:
On Sat, Dec 30, 2017 at 09:35:35AM -0800, benwis wrote:
> Hi,
> I tried to install Klipper on the Rev B Beaglebone Black, and I got this error. Any chance you can advise on a fix? It's being run on the latest debian image.
>
> benwis@beaglebone:~/klipper$ make [...]

It's warning that it can't install BB-ADC because it's already
installed.  I don't see that on my test board, but it seems this "cape
manager" stuff is constantly changing.  I'd proceed by ignorning the
warning - it shouldn't adversely impact the rest of the install.

Separately, are you using a replicape or a cramps board?

-Kevin

benwis, 2017-12-30:
Currently I have nothing attached to it. The plan was to use a replicape board, but I don't have one yet. I never enabled the ADC, this is a fresh install save for your instructions. Interesting. Maybe I'll try reflashing it.

benwis, 2017-12-30:
Here is the klipper.log file:

Starting Klippy...
Args: ['/home/benwis/klipper/klippy/klippy.py', '/home/benwis/printer.cfg', '-l$
Git version: 'v0.5.0-56-g1bdebea'
CPU: 1 core ARMv7 Processor rev 2 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Sat Dec 30 21:13:10 2017 (1514668390.4 10493.7)
Config error
Traceback (most recent call last):
  File "/home/benwis/klipper/klippy/klippy.py", line 193, in _connect
    self._load_config()
  File "/home/benwis/klipper/klippy/klippy.py", line 169, in _load_config
    config_file,))
Error: Unable to open config file /home/benwis/printer.cfg

KevinOConnor, 2017-12-30:
On Sat, Dec 30, 2017 at 09:08:30PM +0000, benwis wrote:
> Currently I have nothing attached to it. The plan was to use a replicape board, but I don't have one yet. I never enabled the ADC, this is a fresh install save for your instructions. Interesting. Maybe I'll try reflashing it.

That's not necessary - the Klipper install itself will load the BB-ADC
module.  On my board, it doesn't complain if the software asks it to
load it again.

-Kevin

closed 2018-01-04 by KevinOConnor