Klipper3d/klipper #88: Fail to Generating load graphs

TylonHH, 2017-12-30:
As mentioned [here](https://github.com/KevinOConnor/klipper/blob/3af87e1c42ece2a36773520daa06612888cb9e29/docs/Debugging.md#generating-load-graphs) I tried to generate a graph after some prints.
But I got this error:
```
pi@octopi:~/klipper $ ~/klipper/scripts/graphstats.py /tmp/klippy.log loadgraph.png
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/graphstats.py", line 122, in <module>
    main()
  File "/home/pi/klipper/scripts/graphstats.py", line 119, in main
    plot_mcu(data, MAXBANDWIDTH, outname, graph_awake=options.awake)
  File "/home/pi/klipper/scripts/graphstats.py", line 92, in plot_mcu
    fig, ax1 = plt.subplots()
  File "/usr/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 1078, in subplots
    fig = figure(**fig_kw)
  File "/usr/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 435, in figure
    **kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/backend_tkagg.py", line 81, in new_figure_manager
    return new_figure_manager_given_figure(num, figure)
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/backend_tkagg.py", line 89, in new_figure_manager_given_figure
    window = Tk.Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1813, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable
```

What did I wrong?

TylonHH, 2017-12-30:
I added
```
import matplotlib
matplotlib.use('Agg')
```
to very beginning of the graphstats.py
[image]

KevinOConnor, 2017-12-30:
On Sat, Dec 30, 2017 at 11:08:51PM +0000, TylonHH wrote:
> I added
> ```
> import matplotlib
> matplotlib.use('Agg')
> ```
> to very beginning of the graphstats.py
> [image]

Thanks!  Should be fixed now (commit 7db6fa7b).

Note that the host buffer statistics are misleading during the start
and end of a print (it's graphing that nothing is buffered, but that's
okay at the very start and end of a print).  I haven't gotten around
to fixing that yet.

-Kevin

closed 2018-01-04 by KevinOConnor