Klipper3d/klipper #91: Octoprint reports print failed

Helovi, 2018-01-01:
Hello,
I’ve been using the Klipper software. Since than the octoprint interface gives the error “print failed” whenever my print is 100% successful.  

In the terminal my gcode sends the M117 command, saying: “M117 Print done”

Is there any way to fix this?

megablue, 2018-01-01:
Don't use unsupported Gcode (most of them are not support yet) M117 is one of them,

Helovi, 2018-01-01:
Thanks. I will try that. 

Is there any way to add those gcode myself to let it silently pass or anything else?

lenne0815, 2018-01-01:
Helovy look at your slicers and script / code and delete the M117 from it.

KevinOConnor, 2018-01-01:
On Mon, Jan 01, 2018 at 10:43:14AM +0000, Helovi wrote:
> Hello,
> I’ve been using the Klipper software. Since than the octoprint interface gives the error “print failed” whenever my print is 100% successful.  
> 
> In the terminal my gcode sends the M117 command, saying: “M117 Print done”
> 
> Is there any way to fix this?

To report an issue, please provide the /tmp/klippy.log file as
described in docs/Contact.md.  Also, follow the instructions for
issuing an M112 immediately after the undesired behavior.

FYI, an unknown command does not report an error to Octoprint.

-Kevin

Helovi, 2018-01-01:
@KevinOConnor I just printed something again, but the klippy.log file is nowhere to be found, neither is the /tmp folder.

Helovi, 2018-01-01:
@KevinOConnor , here is my log file:
I took a look at my codes, command M117 is nowhere to be found in there. Somehome the printer makes this command up by itself or something?
Still my prints are correct while octoprint tells me the print failed, this must be caused from the host?
[klippy.log](https://github.com/KevinOConnor/klipper/files/1596304/klippy.log)

KevinOConnor, 2018-01-02:
On Mon, Jan 01, 2018 at 06:27:38PM +0000, Helovi wrote:
> @KevinOConnor , here is my log file:
> I took a look at my codes, command M117 is nowhere to be found in there. Somehome the printer makes this command up by itself or something?
> Still my prints are [...]

That log didn't contain the M112 command.  The only error reported in
the log is a "Must home axis first" message.  Check your g-code end
sequence to make sure there isn't a movement (G1 command) after a
motor off (M84 or M18 command).  See also "Why can't I move the
stepper before homing the printer?" in the FAQ for further info.

-Kevin

closed 2018-01-11 by KevinOConnor