Klipper3d/klipper #92: M900 support

cbxbiker61, 2018-01-01:
Marlin supports M900 for linear extruder advance.  The latest Slic3r Prusa 
Edition has integrated support for this in it's material configurations (set 
per material).

Do you think we could add support for M900 in Klipper and interpolate the M900 
command to your pressure advance algorithm?

KenshiHH, 2018-01-01:
Klipper already got Pressure Advance
https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md

cbxbiker61, 2018-01-01:
Yes...of course, but it's "statically" defined for the printer.

M900 allows the slicer to specify the amount of advance based on the material....PLA is not the same as...ABS...PETG...

KenshiHH, 2018-01-01:
from what i gatheres he doesn't like nonreadable commands like M900.

but a possible command like "pressure_advance: 0.1" would be nice.
with that you only need to modify your startscript in the slicer

dragonnn, 2018-01-01:
I think we should make some generic solution with allows to set most of the config settings, somethink like:
SET CONFIG extruder pressure_advance=0.1
SET CONFIG extruder step_distance=0.004242
And so on. Even redefining the pins on fly could be somehow usefully in some scenarios (swapping multiple extrudes on live for example).

cbxbiker61, 2018-01-02:
dragonn's idea sounds good.  The main reason why I suggested M900 is to be compatible with the way existing slicers set this.

We could have a klipper specific way as well as an M900 compatible way to make it work with existing slicer setups.  It would require just a bit of math to go from M900 to klipper specific.

dragonnn, 2018-01-02:
When we have the SET CONFIG solution we could just setup some "alias" for M900. But they is small question, is Klipper pressure_advance value equal to Marlin pressure_advance?

cbxbiker61, 2018-01-02:
Not straight across equal.  As per my initial post, we'd have to interpolate.

KevinOConnor, 2018-02-25:
I'm going to close this issue as it doesn't appear that anyone is actively working to implement this feature.

closed 2018-02-25 by KevinOConnor

KevinOConnor, 2018-09-03:
It's possible to configure pressure advance using the SET_PRESSURE_ADVANCE extended g-code command.  See https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md for the details.

NoxIX, 2020-05-30:
So you can't adjust settings on the fly in Klipper? This is akin to having to reflash marlin to change LA. Seems rather...archaic.

theopensourcerer, 2020-05-31:
@NoxIX I have SET_PRESSURE_ADVANCE code in my slicer (it's different for PLA vs PETG) - so yes you can adjust this on-the-fly.

flagoss, 2021-01-30:
Juste Add this to your Printer.cfg
`[gcode_macro M900] 
default_parameter_K: 0.3
gcode :
  SET_PRESSURE_ADVANCE ADVANCE={K}`