Klipper3d/klipper #93: Stops during print

AndreVanKammen, 2018-01-01:
Hello,

First i must say nice software, good idea. I've noticed i can get better and higher printing speeds. But there is one issue that bugs my prints and that is sudden stops of around 2 or 3 seconds during printing. The filament keeps oozing from built up pressure and creates a blob on that spot.

I've printed an empty square of 100mm by 100mm using Cura and OctoPrint, here is the stl file:

[100x100_size_test.zip](https://github.com/KevinOConnor/klipper/files/1596275/100x100_size_test.zip)

It has around 8 blobs in it caused by stops, this is the klippy log:

[klippy_square.zip](https://github.com/KevinOConnor/klipper/files/1596281/klippy_square.zip)

I've printed it a second time and send a M112 during the second stop:

[klippy_square_M112.zip](https://github.com/KevinOConnor/klipper/files/1596282/klippy_square_M112.zip)

I'm running octopi on a Raspberry PI Zero W with camera. Slicing was done using Cura on Windows 10. Klippy is running on a RAMPS compatible board (MKS Gen_L V1.0). It is connected through the hardware serial (/dev/ttyS0). Here is my configuration:

[printer.zip](https://github.com/KevinOConnor/klipper/files/1596284/printer.zip)

I'm having this problem in all kinds of prints but it's most visible in this simple square, that's why i choose it. I hope you can help me.

Friendly Greetings,

André

KevinOConnor, 2018-01-01:
On Mon, Jan 01, 2018 at 06:12:38PM +0000, André van Kammen wrote:
> Hello,
>
> First i must say nice software, good idea. I've noticed i can get better and higher printing speeds. But there is one issue that bugs my prints and that is sudden stops of around 2 or 3 [...]
[...]
> I'm running octopi on a Raspberry PI Zero W with camera. Slicing was done using Cura on Windows 10. Klippy is running on a RAMPS compatible board (MKS Gen_L V1.0). It is connected through the hardware serial (/dev/ttyS0). Here is my configuration:

The recommended hardware is a Raspberry Pi 2 or 3.  I don't recommend
the single core Pi (the RPi1 or Zero) because I've found they don't
have enough CPU power to run Octoprint.  Specifically, I've found that
on some prints Octoprint is unable to send g-codes to Klipper fast
enough to keep the printer busy - thus Klipper is forced to pause
waiting for more g-codes.

The logs you sent aren't conclusive that this is the problem.  I
recommend you ssh into the RPi and run the unix "top" command during a
print - if you see Octoprint (and/or the camera threads) at the top of
the CPU and the CPU reports zero idle for sustained periods - it
indicates this problem.

-Kevin

AndreVanKammen, 2018-01-01:
I've checked with top and OctoPrint reaches about 70% utilization but i didn't see full use. Since the issue occurs on different spots for the same print (just checked) i think you are right, or it must be some other timing issue.

Was afraid it would be something like that,  I will order a Raspberry Pi 3 and try again. Found that Octoprint was also very sluggish on a Pi Zero W.

Thanks for the quick response.

rewolff, 2018-01-02:
Note that "sending GCODE" is a very very simple task. Something an arduino like machine could even do at sufficient speed (it can normally receive, interpret and execute them!).  So something is not coded right in Octoprint. In fact, I think Octoprint is too much a single process that might be "distracted" by a web request or something like that. IMHO the send commands task should be a separate process that runs with higher priority. (and klipper, when used, should run at even higher priority).

AndreVanKammen, 2018-01-02:
I just verified that idle goes to 0% octoprint hugs most of the CPU.

I stopped octoprint and just did

CAT gcodefile > /tmp/printer

Which worked to my surprise and it printed without stops :)

Now i'm waiting for my Raspberry Pi 3 to be arrive.

rewolff, 2018-01-04:
No. it is not specific to the unix socket vs real serial port. The normal state is also that you need a faster pi to allow octoprint to keep pace with the printer.

AndreVanKammen, 2018-01-06:
I got my PI 3 and it's printing fine now, so a PI zero is to slow for Octoprint+klipper.

I got it to print with the PI zero by turning of Octoprint and using klipper as command line tool with the debugfile input. I had to make some changes to gcode.py to have it wait for heating and retracts, because that was turned of during filedebug.

The PI zero W stays above 80% idle in that scenario, so it's definitively a problem with octoprint taking up to much CPU

closed 2018-01-06 by AndreVanKammen

Nickduino, 2019-03-22:
What's surprising is that Prusa has provision for a Pi Zero W on its Mk3 to run Octoprint