Klipper3d/klipper #94: Automatic fan control.

rewolff, 2018-01-02:
I've been "contributing" to the issue tracker for a few days already but finally took the plunge and installed klipper. (it was already compiled, I just needed to flash the firmware and change the octoprint configuration). 

After a few minutes of "careful moves" things looked promising. Bed and heater temp looked good, so I started a test-print. 

That didn't go too well. My fan is supposed to be "always on when the hot end is above 50C". As far as I can tell this functionality is not yet implemented. 
I also found out what happens if you don't follow that rule: The hot end clogs up. :-(

dragonnn, 2018-01-02:
https://github.com/KevinOConnor/klipper/commit/3c5649219fb120c2ac1b98cf22adfefba5dd4c10 just define some pin and add proper g-code to start and end g-code.

LucaBertuolo, 2018-01-02:
You have to definite the heater fan pin and some other few parameter in the printer.cfg
Look at the example-extras.cfg in [klipper/config/example-extras.cfg](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg)

```
[heater_fan my_nozzle_fan]
pin: ar9
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
```
ar9 is the D9 output on RAMPS

I'm testing it right now...
I've found that the fan start when a temperature > 50° is set for the extruder...
I think that is possible to remove the target_temp check in the fan routine so the fan will be turned on only when the hotend reached a temperature > heater_temp, like in marlin/repetier...

dragonnn, 2018-01-02:
Oh I didn't know that 'heater_temp' is already defined, nice.

rewolff, 2018-01-02:
OK. Already implemented. Case closed.

closed 2018-01-02 by rewolff