Klipper3d/klipper #95: Print cooling fan does not respond to speed changes expectedly

sapell, 2018-01-02:
When issuing command M106 the print cooling fan turn on at full speed as expected as well as turn off as a result to command M107. However, if issuing M106 S246 after M106 S247 there is a significant decrease in speed. At 247 the speed is still somewhat maximum and at 246 the fan starts to whine and almost stalls. I'm using Klipper on BeagleBone Black and Replicape. Suspecting the PCA9685 register values are being written somehow incorrectly. This issue will likely produce print quality issues due to stalling fan when the fan speed is controlled by slicer generated g-code.

Klippy log after playing around with the M106 with different values followed by M112:
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1598807/klippy.log.gz)

jakep82, 2018-01-02:
I have a Replicape running Klipper, and I'm not seeing this issue.  M106 S128 runs my fan at what I would estimate is around half speed.  I also have my other fans set up as PWM outputs, and changing the output value to numbers between 0.25 and 1.0 has the desired effect.  My fans stop spinning at a PWM value around 0.2, but I would consider this normal.

KevinOConnor, 2018-01-03:
On Tue, Jan 02, 2018 at 09:58:47PM +0000, Simo Apell wrote:
> When issuing command M106 the print cooling fan turn on at full speed as expected as well as turn off as a result to command M107. However, if issuing M106 S246 after M106 S247 there is a [...]

Does the problem persist if you do a full reboot?  (Reboot, startup,
issue "M106 S247", wait 10 seconds, issue "M160 S246"?)  Does the
problem persist if you remove the "multi_pin" and just run the
above sequence on fan?

The log indicates the host correctly instructed the linux mcu process
to set the specified values.  So, something must have gone awry after
that.

-Kevin

KevinOConnor, 2018-01-03:
On Tue, Jan 02, 2018 at 03:05:48PM -0800, jakep82 wrote:
> I have a Replicape running Klipper, and I'm not seeing this issue.  M106 S128 runs my fan at what I would estimate is around half speed.  I also have my other fans set up as PWM outputs, and [...]

Can you test the sequence with the same exact fan speeds - "M106
S247", wait 10 seconds, "M106 S246"?

-Kevin

jakep82, 2018-01-03:
I tried M106 S247, followed by M106 S246.  I was unable to detect a change in speed, and both ran very close to full speed.  Out of curiosity I tried a bunch of other speeds, and my fan continued running until I dropped below S35 which is about 14%.  I was using a cheap 30mm 12V fan for these tests with a bone stock Replicape config file.

jakep82, 2018-01-03:
@sapell , have you tried enabling the hard_pwm setting?  Perhaps the fan you're using doesn't like the default 10ms cycle time of the software PWM.

KevinOConnor, 2018-01-03:
On Wed, Jan 03, 2018 at 05:35:20PM +0000, jakep82 wrote:
> @sapell , have you tried enabling the hard_pwm setting?  Perhaps the fan you're using doesn't like the default 10ms cycle time of the software PWM.

FYI, the hard_pwm setting doesn't do anything on the replicape power
pins - everything on the pca9685 is "hardware" pwm.

-Kevin

sapell, 2018-01-03:
Thanks for the responses! I suspected the multi_pin at first and tried with only one print cooling fan, but the behavior was the same. Also tried the hard_pwm, but as Kevin already mentioned, it does no good with the PCA9685 on the Replicape, in fact the Klipper considers that as an error in config file and halts (turning the coldend fan on). What PWM frequency/period is the PCA9685 set to use by default? Can it be altered by a config parameter like the software PWM in order to test what jake82 proposed? I'm too tired now to carry out any testing today, hopefully I have time tomorrow.

sapell, 2018-01-03:
Clicked the Close and comment button by mistake, time to sleep.

KevinOConnor, 2018-01-03:
On Wed, Jan 03, 2018 at 12:51:40PM -0800, Simo Apell wrote:
> Thanks for the responses! I suspected the multi_pin at first and tried with only one print cooling fan, but the behavior was the same. Also tried the hard_pwm, but as Kevin already mentioned, it [...]

It's set to 1ms currently - change
chipmisc.py:REPLICAPE_PCA9685_CYCLE_TIME and "sudo service klipper
restart" if you want to try a different value.  Not sure how it would
be related though.

-Kevin

sapell, 2018-01-04:
I did some more testing and measured fan PWM frequency and duty-cycle with various fan speed settings. Turned out that Klipper and PCA9685 do their jobs as they should, so I'll close the ticket. Seems that my 30mm brushless fans are incompatible with Replicape for some reason. It's weird as they worked fine with the RAMPS+Repetier combo. Must be something different on the Replicape fan output implementation. Also a 60mm fan for cooling the electronics on fan output 1 runs happily at constant 75% duty-cycle.

closed 2018-01-04 by sapell

jakep82, 2018-01-04:
This is information I found on the issue on another forum where I lurk.  This is why I suggested changing the PWM frequency.  I can't say for certain that it will solve the issue, but several people have reported success in situations like yours.

> Most of the fans are brushless, they use a fixed reference frequency circuit inside to generate commutator waveforms. A lot of these have chips that reset when power goes off, and require a certain time before they come out of reset, so a fast PWM signal of less than 100% duty will continually reset these chips, that's why they don't work with fast PWM. Soft PWM slows the PWM frequency right down so that they come out of reset and start to drive the motor, it's then a classic duty cycle controlled fan. A brushed motor works because the coils are inductive and that can cope with fast PWM, there are no electronics to get in the way.

jakep82, 2018-01-04:
Also an article with a little more info, and the relevant section quoted below.

[https://www.maximintegrated.com/en/app-notes/index.mvp/id/1784](url)

> Pulse-width modulating (PWM) the fan directly involves turning the fan's power supply on and off at a fixed frequency. Duty-cycle adjustments are made to control the speed of the fan. The larger the duty cycle, the faster the fan spins. Choosing the appropriate frequency for this method can be somewhat tricky. If the frequency of the PWM signal is too slow, the fan's speed will noticeably oscillate within a PWM cycle. To illustrate this point, take the ridiculous extreme of a 50% duty-cycle, 0.01Hz drive signal. The fan will spin to full speed during the first 50 seconds and then stop within the next 50 seconds. The frequency can also be too high, as commutation is done electronically using circuits that are powered off the fan's plus and minus terminals. Using PWM with the fan and therefore the internal commutation electronics too quickly can cause the internal commutation electronics to cease functioning correctly. Remember that these electronics were not designed to run on anything but DC supplies. Thus, useful frequencies range from 20Hz to 160Hz. In addition, the PWM rise and fall times must be sufficiently slow to ensure long-term reliability of the fan.

sapell, 2018-01-05:
Thanks jakep82 for the tips! I only went down to 100Hz with the frequency, hopefully the ~24Hz is low enough, PCA9685 can't do lower than that. I've understood that Repetier has 16Hz PWM for the fans.

sapell, 2018-01-05:
I configured the REPLICAPE_PCA9685_CYCLE_TIME to 0.05 (actually lower than the PCA9685 can go, but wanted to make sure the frequency is as low as it can, seems to work) and got the both print fans to work down to 25% duty cycle fairly linearly, that is acceptable. Measured that the PWM frequency without a fan is ~25Hz with this setting. During these tests I noticed that my static pwm 'board_fan' also responds to M106 and M107. I thought it should turn on when Klipper starts and stay on regardless of M106 and M107 or have I misunderstood it? I will create another ticket about it though.