Klipper3d/klipper #100: problem Building firmware under Ubuntu 16.04

MrWu-at-msv, 2018-01-04:
On ubuntu 16.04 when building firmware
user@user-X501A1:~/klipper$ make
  Build Kconfig config file

*** Error during update of the configuration.

/home/user/klipper/scripts/kconfig/Makefile:36: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 1
Makefile:98: recipe for target 'out/autoconf.h' failed
make: *** [out/autoconf.h] Error 2
user@user-X501A1:~/klipper$

khseal, 2018-01-06:
I'm compile Lubuntu 16.04 all okey...

MrWu-at-msv, 2018-01-06:
Did you take peculiars attentions to somethings? Dependency library?

khseal, 2018-01-08:
There were no problems with dependencies.
There were obvious problems with dependencies on Linux Mint.
The problems were obvious and resolved quickly.
Perhaps the problems were with the octoprint.

MrWu-at-msv, 2018-01-08:
Ok will try again on fresh install

KevinOConnor, 2018-01-11:
There are some hints on installing on other OSes at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

Beyond that, you'll need to figure out how to get the software to work on a non standard setup.

closed 2018-01-11 by KevinOConnor