Klipper3d/klipper #96: Is Bed leveling high up in the todo list?

w1ebr, 2018-01-04:
I am not sure where to ask the question, but, is any form of multi-point bed leveling coming in the near future?
Thank you!

Westra, 2018-01-10:
First of all,

Great work! heard about it and had to install it, one evening later it is up and running and printing better and quieter then before. Hopefully my TMC2100 will arrive soon, very curious about the sound performance. So far its the best upgrade I have done (and yes i had it for only a few days now, but i will continue to print with it). 

coming back to bed levelling, I build my printer with a difficult/non-adjustable bed, because the z probe and auto level did a wonderful job. 
I too am very interested in bed levelling, I'm not a programmer (just some very basic Arduino stuff) but I can test stuff on my printer (H-bot with an inductive z probe).
Again thanks for this great project.

KevinOConnor, 2018-01-11:
There's a lot of interest in auto bed leveling, so I'm sure it will happen.  I don't have any kind of time frame for it though.

w1ebr, 2018-01-11:
Hi Kevin, I thought about this a bit and I am wondering if you see the problem to be solved the same way I do.

My current understanding of the problem is that incorporating z movements during a single gcode x,y,extrusion movement for any bed leveling compensation scenario other than a simple linear "bed is flat but not level" adjustment would have to occur as a part of the motion processing (vs something that could be handled as a gcode post-process step that added a single z adjustment for every extrusion movement. The reason I bring this up is because the simple case of a flat bed that is not level in either the x and/or y directions could be compensated for by post-processing the slicer's gcode to include a z coordinate before your software interpreted the gcode. Handling bed warping or a "bump map" might mean that the interpretation of a single gcode command might include more than a single destination in the z axis ( which would require more sophisticated post-processing capable of segmenting the single gcoded linear extrusion into several smaller extrusions with individual z axis destinations (is this what your code would have to do if it handled bed leveling? Does this mean that bed leveling processing actually could be a step that occurs between the slicer and your code and might be a good way to keep the solution from messing up your code? )

Gene

hg42, 2018-01-11:
I think the general case of a bump map like leveling could indeed be handled on the gcode layer.
But this would have to handle all the gcode pecularities like switching absolute/relative positioning, offsets for any axis, etc.

The bed leveling must be tied to the real bed position, usually defined by some hardware = end stops.

Between the virtual coordinate system of the gcode and the steppers are several layers. The leveling implementation could be inserted as a new layer before any layer that uses the real coordinate system.
As you already said, a longer single straight move would have to be split into several segments to create a curved move instead.

Then there is still the question, why everyone wants this to be solved by software and hence accept the "bump mapped" bottom surface on **every** print instead of using a proper flat heated bed...
(which already assumes the bump map is faded out by height, otherwise the whole print would even show the bump map). My perception is, that manufacturers push bed leveling, because they can sell cheaper hardware when everybody accepts that. There are methods to prevent or adjust sagging (I never would accept a "bump map"). Better or thicker materials or a better construction create some costs and adjustments take some time, both not wanted by the manufacturer. Instead the burden is given to the user, who buys bed level sensors etc. to find a so called solution that still leads to bad prints (in my opinion).

Or in simpler words: Why do you want to print a cup/vase/whatever that wiggles instead of standing flat on a table?

jakep82, 2018-01-11:
In the US you can get a 6 pack of 12" x 12" (305mmx305mm) mirror tiles for $10.  I've seen a lot of concern about only using borosilicate glass, but I've cycled one of these tiles from 20C to 80C and back dozens of times with no issues.  They're dead flat, and I get perfect first layers across the entire surface.  I actually stopped using my BLTouch and went back to a simple endstop switch because the probing wasn't necessary.

w1ebr, 2018-01-11:
"Bed leveling" does more than handle a bed which is not flat, it also handles flexion in the x and/or y gantries. My 280 mm x axis deflects about 50 microns at the center. For a  >= 0.2 mm layer height, it doesn't matter much. For <=0.05 layer height, it is critical. Bed leveling also compensates for this.

hg42, 2018-01-11:
makes more sense with that layer height...
may be I would use a thicker first layer or a raft.

I just noticed that segmenting is also needed for sagging.
So a "bump map" is the same according to segmenting. Obviously, it needs more work with the calculation, array of points etc.

hg42, 2018-01-12:
@w1ebr 
do you use bed leveling to correct the flexion of your x axis?
or because of adhesion issues?

Westra, 2018-01-12:
It's true, bed leveling is not a solve-it-all solution. Good printer design and materials are necessary for high quality prints. My printed is a milled flat aluminium plate coated with PEI, however i'm still interested in compensating the slight tilt in the system. I only used 3 point leveling, because i build the z axis without adjustment which resulted in a stiffer construction.

Regarding the wiggling print, I believe Prusa has a pretty solution, they compensate the leveling of the bed in the first few layers. After that the print is level.

hg42, 2018-01-12:
@Westra you probably do not want to move Z all the time while printing...

I also have a PEI coated (diffused) aluminum plate (8mm). Mine is 3-point adjustable though.
Currently I use springs and screws that are exactly vertical (to prevent shifting of the bed while printing) but I am thinking of leveling screws without springs.

I could also use three lead screws to do a physical leveling instead of constantly moving the Z axis.
Given that most boards only support 4-6 stepper drivers,
Klipper is the perfect companion for such a project (one of the reasons I use it).
It's easily extendable with cheap electronics and you may even have some old boards laying around.

Westra, 2018-01-12:
@hg42 I haven not had any problems with my z axis moving. 
Using three lead lead screws is something i would like to do too, but never did due to the many stepper drivers involved. Klipper does sound like a very good solution for the many stepper drivers. 
(just like for multi material (2+) printing). 

If Klipper is aiming to support this, It would be nice to build the next printer (there is always the next :) ) that way.

w1ebr, 2018-01-13:
My print bed is borosilcate float glass and is very flat. There are still good reasons for bed leveling, IMHO.

- Flex in the x and y gantries. My 300 mm x axis sags by about 50 microns in the middle.
- Thermal expansion. Aluminum and plastic both like to expand. I have used a Flex3drive bed on top of the glass and the plastic bed does expand.
- Before the Flex3drive bed, which I can't always use, I had a PEI sheet and the adhesive began to slowly out-gas, causing micro bumps in the bed.
- Overall leveling - my bed does not use springs to fine tune the height in each corner.
- Accuracy and automation - I can do a quick matrix bed level and be sure that my prints will stick. I have dual z-axis motors that are stepped simultaneously, but when the printer is off, the ends of the bed could get moved and be "out of sync"

Gene
-

stoto, 2018-01-16:
I am signing up for this as well! I would love to see Mesh bed leveling come alive. Even if you have to provide the mesh in a config file by yourself and there is no automatic probing at start.

I am watching Klipper for a while now, waiting for bed leveling to appear before I switch over from Marlin.

I have a 300x300 Bed with CoreXY mechanics and while my bed is perfectly flat there is some sagging in the middle of the X gantry (2x 8mm hardened steel rods).

OneDayFlie, 2018-01-16:
Looking forward to that option :p

BETLOG, 2018-01-17:
The recent interest in 32bit boards, and the currently limited selection, is generating a lot of side mentions of klipper. It's my experience that everyone who loves the idea of klipper also immediately notices that there is no z-probing, and that this often puts that person back into assess and wait mode for 32bit boards.
The addition of z-probing would immediately attract those of us in a 32bit holding pattern to implement klipper.
Just saying.

OneDayFlie, 2018-01-17:
BETLOG is right..
Almost forgetting THANKS! for the wonderfull work you allready did KevinOConner! :) 
But are you alone working on this project? I hope you have or will getting some help.  This show must go on!! ;)

sbts, 2018-01-19:
@KevinOConnor Depending on the amount of deviation on the bed it may be possible to compensate by changing flowrate@fixed speed, speed@fixed flow, or a combination of both.
That would work nicely for small deviations without needing to "remap" the Z height or modify the gcode.

In-particular simply speeding up, or slowing down movement or extruder steps (with compensating drop or add of extruder steps) should have the desired result in many if not all cases with deviation <~2/3 layer height

The only downside to this I can come up with is it would take some careful testing to get even first layer "Squish" so apparent line width when viewed from the bottom remained the same.

The upside, it should only be relevant on the first layer. Beyond that, you should have a "flat" surface (relative to the extruder's path in Z) to print on

Dealing with the effects of Sag in X or Y above the 1st layer isn't something that can currently be compensated for anyway as there is no current method of testing and mapping that.
In the event a means of mapping that was devised this same method could be used to compensate to some extent.
Although I really don't know that it would be a good idea, except on "top" surface layers perhaps.

@KevinOConnor by the way, while I take no responsibility for any possibility that the above concept may/maynot breach a patent, in the event it doesn't I'll claim this as prior art against any future patents that may be issued.
The above statement not withstanding, I offer the concept to klipper for unencumbered use.
:-))

KevinOConnor, 2018-01-21:
I've added basic support for "bed tilt compensation" to the work-probe-20170609 branch.  If all works well, this code should make it possible to handle printers that don't have bed leveling screws and instead use a Z probe to determine the bed tilt.  See the bed_tilt section of docs/example-extras.cfg for information on how to use it.

git fetch ; git checkout origin/work-probe-20170609 ; sudo service klipper restart

I'm looking for feedback on this support.

sgraber, 2018-01-22:
I have to agree with @BETLOG as well:  I've got a delta that has z-probe capabilities and I don't want to switch that one over to Klipper until it's implemented.  Manually leveling a delta is an exercise in frustration.  :/

KevinOConnor, 2018-01-23:
On Mon, Jan 22, 2018 at 08:07:34AM -0800, Shane Graber wrote:
> I have to agree with @BETLOG as well:  I've got a delta that has z-probe capabilities and I don't want to switch that one over to Klipper until it's implemented.  Manually leveling a delta is an exercise in frustration.  :/

The delta calibration code is pretty close to completion on the
work-probe-20170609 branch.  For the details, see the description of
the [delta_calibrate] section of config/example-delta.cfg on that
branch.

-Kevin

w1ebr, 2018-01-24:
Kevin, thanks for working so quickly on this!

stoto, 2018-01-24:
Meanwhile I was looking into scipy's 2d interpolation feature. @KevinOConnor do you think it would be okey to use it for proper mesh bed leveling? It would allow to have like 15x15 measured points and 2D interpolate a Z height at any position on the bed. 
An interpolation could be done just before every toolhead movement. I don't know currently how heavy is the computation part of the interpolation, but If it turns out to be too heavy the interpolation can be done one time only right after the bed measurement for every mm of the bed. That would mean interpolating a 300x300 matrix from a 15x15 measurement if the bed is 300x300mm. This way selecting the proper z compensation value would be very simple, just rounding the X and Y coordinate to whole numbers and using those as indexes from the interpolated matrix.

mediumo, 2018-01-25:
Hi Kevin,

I also tested the bed-tilt feature of the probe branch.
didn't print yet but calibration seems to work.

2 things I observed:
- The Calibration does a G28 homing first, which I think makes the calibration more realiable and assures to habe the correct  z=0 as base. But: the calibration uses the fixed G28 method, if you use homing override (which I use for bl-touch probe to home Z in the center) it doesn't work.
For now I just removed the G28 on calibration and do it manually.
Would it be a good way to trigger G28 with gcode.process_command or via gcode_handlers.get?

- I don't get the speed handling. At the moment it's a bit unreliable because to fast. I have a homing speed for each axis, for probing and for bed_tilt claibration. Have to dig deeper on which speed is used at which part of the calibration process. Because I want faster moves to the xy positions and very slow probing speed. Maybe a double-touch feature as in the normal homing would be a good idea?

Regards
Olaf

KevinOConnor, 2018-01-25:
On Thu, Jan 25, 2018 at 11:43:17AM +0000, mediumo wrote:
> 2 things I observed:
> - The Calibration does a G28 homing first, which I think makes the calibration more realiable and assures to habe the correct  z=0 as base. But: the calibration uses the fixed [...]

Thanks - good catch.  I've updated the work-probe-20170609 branch to
fix that.  (It also updates homing_override to allow moves prior to
homing via set_position_x config options.)

cd ~/klipper ; git fetch ; git checkout origin/work-probe-20170609 ; sudo service klipper restart

> - I don't get the speed handling. At the moment it's a bit unreliable because to fast. I have a homing speed for each axis, for probing and for bed_tilt claibration. Have to dig deeper on which [...]

When using the probe:z_virtual_endstop mechanism, then endstop_x.speed
controls the homing speed and bed_tilt.speed/probe.speed have no
impact.  When running a bed tilt calculation, the probe.speed controls
the descent speed, and bed_tilt.speed controls the xy move speed.

-Kevin

KevinOConnor, 2018-01-25:
On Tue, Jan 23, 2018 at 10:02:09PM -0800, Peter Stojcsics wrote:
> Meanwhile I was looking into scipy's 2d interpolation feature. @KevinOConnor do you think it would be okey to use it for proper mesh bed leveling? It would allow to have like 15x15 measured points and 2D interpolate a Z height at any position on the bed. 

You could try.  I suspect that simple linear interpolation is
sufficient though.

-Kevin

mediumo, 2018-01-25:
Know I got it with the speeds.

Everything worked with the bed_tilt until I added the x_adjus and y_adjust to to config :-).
As soons as they are added, the homing_override seems to have no effect anymore.
I take a look into the code, but I'm not quite familiar with everything yet, maybe you habe an idea.

Regards

KevinOConnor, 2018-01-25:
On Thu, Jan 25, 2018 at 08:26:09PM +0000, mediumo wrote:
> Everything worked with the bed_tilt until I added the x_adjus and y_adjust to to config :-).
> As soons as they are added, the homing_override seems to have no effect anymore.
> I take a look into the code, but I'm not quite familiar with everything yet, maybe you habe an idea.

If anything undesirable happens like the above, issue an M112, and
post the log here as described at:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

mediumo, 2018-01-25:
sure,

find the log attached.
As soon as x_adjust and y_adjust <> 0, the custom g-code from home-override is not executed.

Olaf
[klippy.log](https://github.com/KevinOConnor/klipper/files/1665670/klippy.log)

KevinOConnor, 2018-01-25:
On Thu, Jan 25, 2018 at 09:09:48PM +0000, mediumo wrote:
> sure,
> 
> find the log attached.
> As soon as x_adjust and y_adjust <> 0, the custom g-code from home-override is not executed.

Okay, thanks - there was a bug in the code.  Can you retest with the
latest code:

cd ~/klipper ; git fetch ; git checkout origin/work-probe-20170609 ; sudo service klipper restart

Also, if your probe triggers at a Z of zero, then you'll likely need
to allow the probe to go below zero - update your config with:
stepper_z.position_min=-1 and probe.z_position=-1

-Kevin

stoto, 2018-01-26:
@KevinOConnor the BLTouch has an extending arm which deploys when it's probing. That way the Z0 point for the probe means the nozzle is still a few mm above the bed. When the probing is finished the arm is retracting going above the nozzle height. That's why we need to be able to set the probe-nozzle offset in X-Y-Z.

mediumo, 2018-01-26:
@KevinOConnor: Thanks, I will try it later. Yesterday I did a quick implementation of 2 commands for allowing and forbidding unsafe movements, which had pretty much the same effect. I will provide a pull if you like.
I implemented it analog to dragonnns bltouch branch. But I think it would be best to do the bypass in the actual kinamtics class (check_movement) instead of doing it in the toolhead class and bypass the check_movement method completly, what do you think?

@stoto: You are completely right. I think In the moment it wouldn't make a huge difference (if the probe is not to far away from the nozzle) since just the tilt of the bed is calculated. As soon as you think about something like mesh-leveling it should be considered. Z-offset can easily be applied with M206 and the config described by Kevin

@KevinOConnor: I think in the long term a mesh leveling would make sense, since you can deal with uneven beds (likely if you have a standard mk2/3 like bed without glas) or sag in the axis (likely at standard 8mm rods and long axis). I'm just getting into that topic, seems like there are several approaches like splitting the moves. With the processing power of the pi it could be interesting to add a spline for the z-correction to each movement which you can get from an interpolated heightmap of the bed etc.. Or to get an elevation-profile from the heightmap for the xy-path like navigation/GIS tools provide them .... I just don't know how how easy it would beto calculate the required step-corrections/timings for that.

Olaf

KevinOConnor, 2018-01-29:
FYI, the work-probe-20170609 branch has been merged into master.

tgiphil, 2018-01-29:
Awesome work!

w1ebr, 2018-01-29:
Kevin, thank you for your work on this! I have one big job to get done, after which I intend to really dig into this great approach! Again, thanks!

KevinOConnor, 2018-02-08:
FYI, basic bed leveling should now be supported (via the bed_tilt_calibrate command).  I don't have any immediate plans to work on mesh bed leveling.  Hopefully someone else will provide an implementation.

psy0rz, 2018-02-10:
Cool ill try this with my CR-10. Still hope someone will implement mesh leveling.

The cr10 due to its size and design has multiple possible leveling problems, even if the bed itself would he flat. Z axis can sag in the middle or tilt and the bed can tilt as well. You could compensate manually, but it changes all the time.

Alexdad76, 2018-02-13:
Hello! 
I can not complete BED_TILT_CALIBRATE.
After 1-2 points this is constantly interrupted by a error:
"Failed to home probe: Timeout during endstop homing
Failed to home probe: Timeout during endstop homing "
I use an optical sensor as a endstop and a probe.
[2018-02-13_18-55_klippy.log](https://github.com/KevinOConnor/klipper/files/1720802/2018-02-13_18-55_klippy.log)
[2018-02-13_18-40_klippy.log](https://github.com/KevinOConnor/klipper/files/1720801/2018-02-13_18-40_klippy.log)

KevinOConnor, 2018-02-13:
On Tue, Feb 13, 2018 at 08:57:34AM -0800, Alexdad76 wrote:
> Hello! 
> I can not complete BED_TILT_CALIBRATE.
> After 1-2 points this is constantly interrupted by a error:
> "Failed to home probe: Timeout during endstop homing
> Failed to home probe: Timeout during endstop homing "
> I use an optical sensor as a endstop and a probe.

In order to home the z axis with the probe, you need to know the
distance between the nozzle and the bed when the probe triggers.  This
distance needs to be programmed in both the stepper_z position_endstop
and the probe's probe_z_offset.

If unsure of that distance, start with something safe - for example
2.345mm.  Set both stepper_z position_endstop and probe probe_z_offset
to 2.345mm.  Then run the BED_TILT_CALIBRATE command.  At completion
of that command, it should tell you the actual z offset (if it's
different from 2.345mm).  You'd then update the config with the
recommended x_adjust/y_adjust parameters AND update the config with
the new position_endstop / probe_z_offset.

Separately, it's not clear if you are looking to do a "blind" Z lift
at the start of your homing procedure.  If you are, use something like
this:

[homing_override]
set_position_z: 5
gcode:
    G90
    G1 Z7 F600 ; Blindly lift the Z 2mm at start
    G28 X0 Y0
    G1 X100 Y100 F3600
    G28 Z0

-Kevin

Alexdad76, 2018-02-14:
Many thanks for the prompt reply!
I'll try as soon as I get home tonight.

> @KevinOConnor 
Separately, it's not clear if you are looking to do a "blind" Z lift at the start of your homing procedure. If you are, use something like this:

Before homing procedure, I first move the table down 5-10 mm, so as not to touch the glass holding clips.  I try to repeat the "safe homing" as in Marlin.
Maybe there is a more elegant way, but I still do not have enough experience.

Alexdad76, 2018-02-14:
@KevinOConnor, everything worked perfectly!

One more question:
My Z-axis is driven by a belt. As I understand all the axes move at the same speed:

> The speed (in mm / s) of non-probing moves during the calibration. The default is 50.

For the Z-axis, a strong jerk occurs after the probing when the table is lowered.
Is it possible to separately adjust the speed of X-Y and Z for non-probing moves?

iggut, 2018-06-07:
Any developments in this? Sadly I would have to rebuild my printer with new C-beams to use Klipper. Mine are slightly bowed and either need mesh bed leveling or to be replaced and printer rebuilt :(

psy0rz, 2018-06-07:
I just received my prusa i3 mk3. :) ill use the stock firmware for now.

3eggert, 2018-08-14:
Are there any news? I tried to implement it by my self but failed )-:

AxMod3DPrint, 2018-08-14:
@3eggert - see pull request #535 - needs to be looked at by Kevin, but hopefully should be implemented soon.

jwcrawley, 2018-08-16:
Awesome :-D 

I did have a few questions:

1. Will there be a way to manually measure a mesh via jogging per measured point
2. Can we save a mesh level, and use the saved?

Artefact2, 2018-08-16:
At first glance it seems to require a probe, which is too bad. Being able to probe mesh points manually would be nice.

jakep82, 2018-08-16:
@Artefact2 I'm sure a pull request to add this feature would be welcomed to help improve this free, open source software.

Arksine, 2018-08-17:
So while it isn't documented, In theory you manual_probing should be available in bed_mesh as it uses the same functionality to probe that bed_tilt uses.  I havent' tested it, and honestly it didn't occur to me that people would want to manually probe a minimum of 9 points.  After its merged if someone wants to test manual probing I can update the documentation to include it assuming it works okay.  The only thing that could be tricky is if you don't home toward the bed or have a delta printer.  I would likely need to update bed_mesh.py for manual probing in those instances.

Bed Mesh won't initially support dumping a mesh to a file, but it would be trivial to add.  I would just need guidance from Kevin on the best location to write the file to.

Artefact2, 2018-08-20:
> Bed Mesh won't initially support dumping a mesh to a file, but it would be trivial to add. I would just need guidance from Kevin on the best location to write the file to.

It could work similarly to `PID_CALIBRATE`: it outputs mesh data via serial, and you can add it to your `printer.cfg` file.

Artefact2, 2018-08-20:
See #555.

theopensourcerer, 2018-08-20:
Thanks for working on this feature. I just updated my klipper install and got a basic mesh probe working. One thing I would like to be able to do is to do multiple probes at the same point so the calculation uses the average. When I have been using BED_TILT_CALIBRATE I have three points at each location, e.g.

```
points:
  10,10
  10,10
  10,10
  100,10
  100,10
  100,10
  200,10
  200,10
  200,10
 ...
```
So each point is probed three times before the tilt is finally calculated. 

Thanks.

Alan

jwcrawley, 2018-08-28:
This is awesome!

There's that last bit, which I think @Artefact2 was on to. 

I would like to take a very detailed measurement of my bed (think 400 points), and be able to save that as an array. Then, I want to be able to apply that array to a bed. Then, at the beginning of the print, it would do a 3 point bed level, and apply the telemetry from the 400 points into the plane defined by the 3 point level.

The reason for the 3 point level is to determine the plane in which to calculate the large set of points on. If the machine had an uneven dual Z, the 3 point would not invalidate the data from the prior large scan.

Being able to take a very high resolution measurement is essential for determining the errors in the bed surface. However, doing that accurate of a measurement isn't feasible every time I wish to print, or even once a day.

m4gn3to, 2018-09-02:
Hi.
I can do, apparently, everything related with the bed mesh level, but after all it does not apply it, it does not corrects the z up and down where ever it should be supposed to do... What I'm missing???

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2343659/printer.cfg.txt)
[bed_mesh_output.txt](https://github.com/KevinOConnor/klipper/files/2343661/bed_mesh_output.txt)

Arksine, 2018-09-02:
@m4gn3to I don't think your [probe] z_offset is correct.  It should be the distance between the print surface and the nozzle when the probe triggers, which should never be a negative number.  Also, your fade_start and fade_end options in [bed_mesh] essentially disable z adjustment.  My suggestion is to use the defaults for fade_start and fade_end, or if you want to disable fading set fade_end to 0.  I should probably add some additional checks that don't allow such a small fade interval.

m4gn3to, 2018-09-02:
@Arksine Thank you very much. I've changed the [probe] z_offset and it seems so much better. I also disabled fade_start and fade_end following your recommendations. Now I'm waiting to see my first print... Thank you again

theopensourcerer, 2018-09-03:
@Arksine I just saw the commit for the samples addition. Tested it and it works both with BED_TILT and BED_MESH_CALIBRATE. Thanks.

m4gn3to, 2018-09-04:
Hi. I've been able to print "something". The bed mesh works, as I can see the z axis move up and down while printing, but in some places it does not downs all that is needed. If I'm not misunderstood, I need to get "bicubic_tension" a higher value, I'll try with .5 this night and keep you informed.
Thank you for your help and an amazing firmware for our printers.

Artem-B, 2018-09-08:
I've tried few prints with bed_mesh_calibrate and it works OK.
There are few things I've observed.

* This is not klipper's problem per se, but is something I didn't think about until I ran into this. In some cases one may need a fairly dense sample grid in order for probing to do its job. In my case I've got a PEI sheet that seems to have thickness changes with a period of ~ few inches along one axis. When I probed 12"x12" bed with 4x4 mesh, [the aliasing between the periodic thickness change and the probe points](https://en.wikipedia.org/wiki/Aliasing#Sampling_sinusoidal_functions) made things look as if the bed has a huge hump in the middle. I knew that it was not the case, so I've bumped sample mesh to 10x10 and that clearly showed the higher frequency pattern along one axis.

* Occasionally bed level visualizer shows crazy level 'spikes' on the edge of the bed. I'm not sure if it's something wrong in klipper's code that extrapolates the levels beyond the last sample point or if it's just an artifact of octoprint's bed level visualizer plugin. Alas, I forgot to save the data when I saw it. I'll get more details if I see this again.

Can't wait till I can save/restore the mesh and, maybe, validate few points to sanity check that it matches the bed (mine is removable and may end up being rotated).

Arksine, 2018-09-09:
Generally I find that probing an odd  number of points in each direction does a better job of mapping the surface geometry.  

I don't think the second issue is a problem with Klipper.  If it were then you would see a massive amount of toolhead movement when it reaches the "Spikes".   You can directly analyze the output of BED_MESH_OUTPUT to see if there are any strange values that look out of place.

KevinOConnor, 2018-09-17:
I'm going to close this issue as basic mesh support is not in the main branch.  I understand there are some additional feature requests, but lets track them separately.

-Kevin

closed 2018-09-17 by KevinOConnor