Klipper3d/klipper #99: Printing just stops

lerouxb, 2018-01-04:
Sometimes deep into a print klippy just stops. Nothing interesting in the logs that I can find (see attached), the hotend and heated bed still stays on, it responds to `status`, M105 commands are still returning the temperature.. Except it just doesn't move anymore. Octopi seems to think the print is done.

If I drop the speed it usually doesn't happen. I dropped the serial baud rate to 115200 and that also seemed to have reduced it. But I still have prints where it happens, usually hours in which is kinda crappy for trying to reproduce the error.

I'm on a raspberry pi zero. Not sure if that makes a difference.

Any ideas? My motors are 400 steps per revolution and I'm doing 1/32 microstepping for X and Y with 16 mm diameter pulleys (I have a custom corexy using spectra line) so that might be a lot of steps I suppose. Extruder is standard mk8 bowden setup with 1/16 microstepping and 200 steps per revolution motor, z axis lead screw is TRS 8mm pitch, also 1/16 microstepping and 200 steps per revolution. So just the x and y plane that could be a problem.

What sort of max speeds should I expect? I was hoping to do 80mm/s as that should be a reasonable/safe maximum for this kind of extruder and hotend, 0.4mm nozzle, 0.2mm layer height.

(Really loving klippy, btw. First tried it out on my mks gen 1.4 while trying to isolate an issue I was having with smoothieware on another board and I'm so not going back.)

[klippy.log](https://github.com/KevinOConnor/klipper/files/1603063/klippy.log)

lerouxb, 2018-01-04:
I see a pi zero could in fact be a problem https://github.com/KevinOConnor/klipper/issues/93#issuecomment-354669085

lerouxb, 2018-01-04:
Hmm and also https://github.com/foosel/OctoPrint/issues/450#issuecomment-342492897

KevinOConnor, 2018-01-04:
As noted, the Zero isn't a recommended setup, but the log doesn't see the classic symptoms associated with slow command input.  Instead, in the klippy.log file I see:

Extrude below minimum temp
See the 'min_extrude_temp' config option for details

What temperature did you print at?  Something seems to be going awry with your extruder temperature.

jakep82, 2018-01-04:
Perhaps unrelated, but I also see several extrude length errors.

> Extrude only move too long (50.000mm vs 50.000mm)

lerouxb, 2018-01-05:
yeah that was when I loaded filament before it started. It thinks 50mm is > 50mm, so I just went with 49 in the end as a workaround ;)

I printed at 205C. I noticed that at times the temperature graph jumps around a bit, especially when I cancel a print. I recently did a PID autotune too. Probably also related to the zero.

KevinOConnor, 2018-01-11:
This looks like Klipper responded correctly.  Please reopen with further information if necessary.

closed 2018-01-11 by KevinOConnor