Klipper3d/klipper #101: Prusa MK3 support

Richardgriff, 2018-01-05:
Hi,

With the release of the prusa mk3 which now has new controlboard which has the option of an on board raspberry pi.

I think it would be nice if slipper could support the mk3

dragonnn, 2018-01-05:
It is mostly compatible, since the board is on a AVR and is RAMPS like. I think only we are missing TMC2130 support.

hg42, 2018-01-05:
cruwaller already has a branch with tmc2130 (and a smoothie compatible, or both), I already compiled it and will probably test it at the weekend

Richardgriff, 2018-01-06:
Would like to try. Please let me know if you get this working

Diederikvanderpant, 2018-01-16:
Hi, i am also interested in testing this somewhere in the next month. Just installed tmc2130 on my megatronics 2 board. Where could i find this cruwaller branch?

simonlee, 2018-01-23:
Would be amazing if this worked with prusa mk3 + octoprint. The prusa firmware has a bunch of features which I assume would not have equivalences in klipper such as the auto leveling stuff and filament detection/crash detection...

simonlee, 2018-01-26:
actually cruwaller does not have a tmc2130 branch anymore. I think he deleted it.

the prusa mk3 does not use endstop pins to detect the min/max positions but instead rely on some other metric which i believe are voltage changes based on something interrupting the stepper motion.

hg42, 2018-01-26:
I think, at some time he merged tmc2130 into lpc176x branch, where he is testing it now, see klippy/drivers/tmc2130.py

It's latest commit also has the comment "TMC2130 debugging continues. Works with AVR now, Due to go..."

hg42, 2018-01-26:
unfortunately he did not answer my questions on his github mail address or in any of my issue comments (explicitly asking @cruwaller).

He has a bunch of changes that shpuld be impoerted to klipper, e.g. things like allowing any order of **^!** for pin definitions.

cruwaller, 2018-01-26:
@hg42 sorry :( 
I missed your email.... I will send you a response soon.

cruwaller, 2018-01-26:
Others, TMC2130 branch is merges to lpc176x (all my mods are there now). It still supports same MCU as original...
TMC2130 drivers are working great now incl sensorless homing (i  running my corexy setup on RADDS)...

NeelD87, 2018-01-31:
@cruwaller so TMC2130 should now work in ramps config?

simonlee, 2018-01-31:
@cruwaller I see there is an ss_pin for CS/SS but what about for the diag pin?

hg42, 2018-01-31:
from what I know about TMC2130 the diag pin is not needed, because you can write/read values via SPI.
The diag pin could be used to signal some condition (e.g. stall detection) to the outside world. This could allow to use a diag pin as endstop.

Which usage do you mean/want/think of?

simonlee, 2018-01-31:
my printer is the prusa mk3 which i believe uses the diag pin for the endstop. ive tried setting the endstop pin to use the diag pin but no detection. there are many other issues im having and i doubt i can get this to work correctly without expert aid.

hg42, 2018-01-31:
@cruwaller how did you connect the TMC2130?

I once had a plan to program the diag pin to signal stalls and simple connect it to an endstop input.
This way you could program the TMC at start and then forget it.
But may be there are benefits form reading values (force etc.) while printing. Also the stall detection may differ depending on speeds (e.g. homing speed). So there are definitely more usages.

hg42, 2018-01-31:
@simonlee not that I currently use the TMC2130, but did you look at the example config files?

```
[driver tmc2130_x]
type:               TMC2130
#steps_per_mm:       400
steps_per_mm:       100
# SPI
ss_pin:             driverxy:ar7
spi_mode:           3
spi_speed:          2000000
# config
current:            1500.0
microsteps:         8
#microsteps:         32
mode :              spreadCycle
diag0_out:          temp_prewarn
diag1_out:          stall
diag0_active_high:  1
diag1_active_high:  1
interpolate:        0
sensor_less_homing: 1
stealth_max_speed:  40
stall_threshold:    19
```

so you select a function for the diag pins 0 and 1.
You say one diag pin is connected to the endstop input (you don't need to configure this, it's a cable!).
You need to make sure that the diag pin that is connected to the end stop has the stall function.
If it is not diag1, you have to swap the diag0 and diag1 names in the section.

For example, if you have diag0 connected to the end stop input, then it has to be
```
diag1_out:          temp_prewarn
diag0_out:          stall
diag1_active_high:  1
diag0_active_high:  1
```

simonlee, 2018-01-31:
I gathered from this file that the max endstop pin is the diag pin:
https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/pins_Einy_0_4.h

```
#define X_TMC2130_CS           41
#define X_TMC2130_DIAG         64 // !!! changed from 40 (EINY03)
#define X_STEP_PIN             37
#define X_DIR_PIN              49
#define X_MIN_PIN            12
//#define X_MAX_PIN            30
//#define X_MIN_PIN              X_TMC2130_DIAG
#define X_MAX_PIN              X_TMC2130_DIAG
#define X_ENABLE_PIN           29
```
so my config looks like:

```
[driver tmc2130_x]
type:               TMC2130
steps_per_mm:       400
# SPI
ss_pin:             ^PG0
spi_mode:           3
spi_speed:          2000000
# config
current:            1000.0
microsteps:         32
mode :              spreadCycle
diag0_out:          temp_prewarn
diag1_out:          stall
diag0_active_high:  1
diag1_active_high:  1
interpolate:        1
sensor_less_homing: 1
stealth_max_speed:  40
stall_threshold:    19

...

[stepper_x]
driver: tmc2130_x
step_pin: PC0
dir_pin: !PL0
enable_pin: !PA7
step_distance: .01
endstop_pin: ^PK2

position_min:       0
position_max:       200
position_endstop:   0
homing_offset:      0
homing_speed:       40
homing_direction:   min
homing_retract_dist:5
```

PK2 is the x diag pin. Not sure if it needs `^` or `!` though. I will try swapping the diag0 and diag1 functions.

hg42, 2018-02-01:
I would try to find out which diag pin is used, there are 3 pins.
Eventually you can follow the copper traces.
With diagX_active_high you can change the polarity.
You can get your endstops and note them.
Then you changexxxx_active high and then send RESTART and see if an endstop changed.
Otherwise you programm another diag pin.
I think I would only configure the one you try. Remove the other.
I think it's not a pullup (no ˆ) and adding a pullup could harm.

KevinOConnor, 2018-02-12:
@TunaLatte , I'm not sure where @cruwaller is with his development, but if someone can forward me a schematic of an AVR board with the TMC2130 then I'll take a look at adding basic support for it.  (I need a link to the schematics - a simple document describing which pins on the mcu connect to which pins on the TMC2130.)  We already support the TMC2100 on the Replicape, so I don't think it's a big deal to add in basic TMC2130 support.

herrwusel, 2018-02-12:
@KevinOConnor : Check page 10 in the [datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf). Is that what you need?

simonlee, 2018-02-12:
@TunaLatte 
I've taken a break from this. I tried the above suggestions but no luck. The steppers do move, but I am not able to figure out how to get it to detect the endstops via the diag pins.

TunaLatte, 2018-02-12:
I see
Can you please post your config? For ramps I suspect the SS pin to be
different?
Thanks

simonlee, 2018-02-12:
```
[driver tmc2130_x]
type:               TMC2130
steps_per_mm:       400
# SPI
ss_pin:             ^PG0
spi_mode:           3
spi_speed:          2000000
# config
current:            1000.0
microsteps:         32
mode :              spreadCycle
diag0_out:          temp_prewarn
diag1_out:          stall
diag0_active_high:  1
diag1_active_high:  1
interpolate:        1
sensor_less_homing: 1
stealth_max_speed:  40
stall_threshold:    19

[driver tmc2130_y]
type:               TMC2130
steps_per_mm:       400
# SPI
ss_pin:             ^PG2
spi_mode:           3
spi_speed:          2000000
# config
current:            1000.0
microsteps:         32
mode :              spreadCycle
diag0_out:          temp_prewarn
diag1_out:          stall
diag0_active_high:  1
diag1_active_high:  1
interpolate:        1
sensor_less_homing: 1
stealth_max_speed:  40
stall_threshold:    19

[stepper_x]
driver: tmc2130_x
step_pin: PC0
dir_pin: !PL0
enable_pin: !PA7
step_distance: .01
endstop_pin: ^PK2

position_min:       0
position_max:       200
position_endstop:   0
homing_offset:      0
homing_speed:       40
homing_direction:   min
homing_retract_dist:5

[stepper_y]
driver: tmc2130_y
step_pin: PC1
dir_pin: !PL1
enable_pin: !PA6
step_distance: .01
endstop_pin: ^PK7

position_min:       0
position_max:       200
position_endstop:   0
homing_offset:      0
homing_speed:       40
homing_direction:   min
homing_retract_dist:5

[stepper_z]
step_pin: PC2
dir_pin: !PL2
enable_pin: !PA5
step_distance: 0.0025
endstop_pin: ^PB4

position_min:       0
position_max:       210
position_endstop:   0.15
homing_pos_x:       100
homing_pos_y:       100
homing_offset:      1.5
homing_speed:       3
homing_direction:   min
homing_retract_dist:5
homing_retract_after: true

[extruder]
step_pin: PC3
dir_pin: PL6
enable_pin: !PA4
step_distance: .003571
nozzle_diameter: 0.400
filament_diameter: 1.750
heater: heater_nozzle

[heater_nozzle]
heater_pin: PE5
min_temp: 0
max_temp: 270
control: pid
pid_Kp: 21.70
pid_Ki: 1.60
pid_Kd: 73.76
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF0

[heater_bed]
heater_pin: PG5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF2
control: watermark
min_temp: 0
max_temp: 130

[heater_fan extruder_cooling_fan]
heater_temp: 50.0
pin: PH5
heater: heater_nozzle

[heater_fan nozzle_cooling_fan]
heater_temp: 100.0
pin: PH3
heater: heater_nozzle

[mcu]
serial: /dev/ttyACM0

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 1250
max_z_velocity: 12
max_z_accel: 200
```

I did not try to swap the diag0 diag1 as @hg42 suggested though.

TunaLatte, 2018-02-12:
Thanks Ill mess around and report asap

TunaLatte, 2018-02-12:
@KevinOConnor In a basic scheme this is how mine is exactly wired now + diag1 (upper pin from the triangular pin order in the following picture orientation.):
[image]

TunaLatte, 2018-02-12:
Well using your config with pin modification worked well except sensorless homing like you said, but after changing stall_threshold from 19 to 4 (0 also works well) its worked perfectly! 
I think the value is too high since if you look at cruwaller's example, his current is also too high, which works well with a value like 19. I had the same issue when converting from 12v to 35v.
Time to test stealthgaurd skipping! Apparently, the reason why it isn't used in marlin, duet (reprapfirmware) and most firmware is that it doesn't like double or quadstepping, at least thats what Trinamic said. Klipper can save this with 8 bit mcu :) No more noise or spreadCycle.

Here is my config for RAMPS1.4 with the same wiring as above including the extra (not visible) enstop pins like this from toms3d: 
[image]
(the handwritten stuff are crap, its diag1 if u cant read it)

Config:
```
# This file contains common pin mappings for RAMPS (v1.3 and later)
# boards. RAMPS boards typically use a firmware compiled for the AVR
# atmega2560 (though other AVR chips are also possible).

# See the example.cfg file for a description of available parameters.

[driver tmc2130_x]
type:               TMC2130
steps_per_mm:       40
# SPI
ss_pin:             ^PB0
spi_mode:           3
spi_speed:          2000000
# config
current:            500.0
microsteps:         8
mode :              spreadCycle
diag0_out:          temp_prewarn
diag1_out:          stall
diag0_active_high:  1
diag1_active_high:  1
interpolate:        1
sensor_less_homing: 1
stealth_max_speed:  40
stall_threshold:    0

[driver tmc2130_y]
type:               TMC2130
steps_per_mm:       40
# SPI
ss_pin:             ^PL0
spi_mode:           3
spi_speed:          2000000
# config
current:            500.0
microsteps:         8
mode :              spreadCycle
diag0_out:          temp_prewarn
diag1_out:          stall
diag0_active_high:  1
diag1_active_high:  1
interpolate:        1
sensor_less_homing: 1
stealth_max_speed:  40
stall_threshold:    0

[stepper_x]
driver: tmc2130_x
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ar3
#endstop_pin: ^ar2
#position_endstop: 0
position_max: 284
homing_speed: 50
position_endstop:   0
homing_offset:      0
homing_speed:       40
homing_direction:   min
homing_retract_dist:5

[stepper_y]
driver: tmc2130_y
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
#endstop_pin: ^ar15
position_endstop: 0
position_max: 292
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: 0.04
endstop_pin: ^!ar18
position_endstop: 0.0
position_max: 310
homing_speed: 10.0

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: 0.00249314385
nozzle_diameter: 0.500
filament_diameter: 1.750
pressure_advance_lookahead_time: 0.010
pressure_advance: 0.05
max_extrude_only_distance: 150
pwm_cycle_time: 0.010
heater: heater_nozzle

[heater_nozzle]
heater_pin: ar10
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog13
control: pid
pid_Kp: 28.853476
pid_Ki: 1.282377
pid_Kd: 162.300802
min_temp: 0
min_extrude_temp: 60
max_temp: 257

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp: 0
max_temp: 170
#max_power: 0.6
pwm_cycle_time: 0.010
pid_deriv_time: 10

[fan]
pin: ar9
#hardware_pwm: True
cycle_time: 0.1

[mcu]
serial: /dev/ttyACM0
#baud: 38400
pin_map: arduino
#restart_method: command

#[virtual_sdcard]
#path: ~/.octoprint/uploads/

[printer]
#buffer_time_high: 30
kinematics: cartesian
max_velocity: 500
max_accel: 1500
max_z_velocity:  10
max_z_accel: 30
#junction_deviation: 0.005

```

NeelD87, 2018-02-13:
@TunaLatte what printer are you running and what speeds did you manage? with Marlin I have everything running including sensorless homing but the problem is skipped steps and have to run it at slow speeds.

NeelD87, 2018-02-13:
In Marlin you can override the SPI pins , will it be possible in klipper
 * Default pins for TMC software SPI
 */
#if ENABLED(TMC_USE_SW_SPI)
  #ifndef TMC_SW_MOSI
    #define TMC_SW_MOSI    66
  #endif
  #ifndef TMC_SW_MISO
    #define TMC_SW_MISO    44
  #endif
  #ifndef TMC_SW_SCK
    #define TMC_SW_SCK     64
  #endif
#endif

 /**
   * Use software SPI for TMC2130.
   * The default SW SPI pins are defined the respective pins files,
   * but you can override or define them here.
   */
  //#define TMC_USE_SW_SPI
  //#define TMC_SW_MOSI       -1
  //#define TMC_SW_MISO       -1
//#define TMC_SW_SCK        -1

TunaLatte, 2018-02-13:
@NeelD87 Its a custom built cartesian printer, I just ran the calibration cube at 70mm/s with 130mm/s travel, very shaky but intended for layer shifting i guess and so far there are no shifts. On marlin it skips at ~30mm/s without s3d auto speed on small areas ("Adjust printing speed for layers below"). I have a semi broken Duet and it skips immediately at around 100mm/s on travel moves if I disable double/quadstepping mode. (invert constexpr uint32_t GCONF_REV_DIR = 1 << 3; in tmc22xx.c if you want to try).

NeelD87, 2018-02-14:
@TunaLatte thanks I am going to try it on weekend.
Please keep posted on your tests

NeelD87, 2018-02-17:
@TunaLatte I am getting Recv: // Internal error on command:"G28" when homing XY 
Z homing works

TunaLatte, 2018-02-17:
@neeld87 can you post the full log? There are important info on startup related to SPI init, to determine wether the connection is working to the drivers.

NeelD87, 2018-02-17:
INFO     printer.mcu.mcu :: Configured MCU 'mcu' (633 moves)
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
INFO     printer.stepper.x.tmc2130 :: Stealth max speed: 55 (71)
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
INFO     printer.stepper.x.tmc2130 :: Stand still
ERROR    printer.stepper.x.tmc2130 :: Phase B short to ground
ERROR    printer.stepper.x.tmc2130 :: Phase A short to ground
WARNING  printer.stepper.x.tmc2130 :: Over temperature prewarning!
ERROR    printer.stepper.x.tmc2130 :: Over temperature!
ERROR    printer.stepper.x.tmc2130 :: motor stall
WARNING  printer.stepper.x.tmc2130 :: driver has switched to fullstep
INFO     printer.stepper.x.tmc2130 :: DRV_STATUS : Current (CS) = 31, stallGuard2 result (SG) = 1023
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
ERROR    printer.stepper.x.tmc2130 :: GCONF Configuration error! [was 0xFFFFFFFF expected 0x00003144]
ERROR    printer.stepper.x.tmc2130 :: CHOPCONF Configuration error! [was 0xFFFFFFFF expected 0x15028135]
ERROR    printer.stepper.x.tmc2130 :: COOLCONF Configuration error! [was 0xFFFFFFFF expected 0x000A0000]
ERROR    printer.stepper.x.tmc2130 :: PWMCONF Configuration error! [was 0xFFFFFFFF expected 0x000505FF]
ERROR    printer.stepper.x.tmc2130 :: IHOLD_IRUN Configuration error! [was 0xFFFFFFFF expected 0xFFFFF9EC]
INFO     printer.stepper.x.tmc2130 ::  init done!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
INFO     printer.stepper.y.tmc2130 :: Stealth max speed: 50 (79)
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
INFO     printer.stepper.y.tmc2130 :: Stand still
ERROR    printer.stepper.y.tmc2130 :: Phase B short to ground
ERROR    printer.stepper.y.tmc2130 :: Phase A short to ground
WARNING  printer.stepper.y.tmc2130 :: Over temperature prewarning!
ERROR    printer.stepper.y.tmc2130 :: Over temperature!
ERROR    printer.stepper.y.tmc2130 :: motor stall
WARNING  printer.stepper.y.tmc2130 :: driver has switched to fullstep
INFO     printer.stepper.y.tmc2130 :: DRV_STATUS : Current (CS) = 31, stallGuard2 result (SG) = 1023
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
ERROR    printer.stepper.y.tmc2130 :: GCONF Configuration error! [was 0xFFFFFFFF expected 0x00003144]
ERROR    printer.stepper.y.tmc2130 :: CHOPCONF Configuration error! [was 0xFFFFFFFF expected 0x15028135]
ERROR    printer.stepper.y.tmc2130 :: COOLCONF Configuration error! [was 0xFFFFFFFF expected 0x000A0000]
ERROR    printer.stepper.y.tmc2130 :: PWMCONF Configuration error! [was 0xFFFFFFFF expected 0x000505FF]
ERROR    printer.stepper.y.tmc2130 :: IHOLD_IRUN Configuration error! [was 0xFFFFFFFF expected 0xFFFFF9EC]
INFO     printer.stepper.y.tmc2130 ::  init done!

Are the drivers gone

TunaLatte, 2018-02-17:
@NeelD87 Nope, thats what I had with wrong CS/SPI pins. You can't kill these via external software anyways. The following errors are normal as they are not used:
ERROR printer.stepper.y.tmc2130 :: IHOLD_IRUN Configuration error! [was 0xFFFFFFFF expected 0xFFFFF9EC]
ERROR printer.stepper.y.tmc2130 :: COOLCONF Configuration error! [was 0xFFFFFFFF expected 0x000A0000]

When you have short to ground error and stallguard2 result > 0 then your SPI comm is not working. You should always post your klippy.log to avoid clutter this issue page I guess. Not to mention it will show the config aswell as other useful data.

KevinOConnor, 2018-02-18:
On Mon, Feb 12, 2018 at 11:19:01AM -0800, Talal A. wrote:
> @KevinOConnor In a basic scheme this is how mine is exactly wired now + diag1 (upper pin from the triangular pin order in the following picture orientation.):

Thanks.  Could you test if the work-tmc2130-20180217 branch works on
your hardware?  This doesn't implement stealthchop or sensorless
homing, but I'd like to see if basic stepper movement works.  See the
example-extras.cfg file for the config details.

cd ~/klipper ; git fetch ; git checkout origin/work-tmc2130-20180217 ; sudo service klipper restart

-Kevin

TunaLatte, 2018-02-18:
@KevinOConnor  I didn't have time to re-setup the endstops but will do, though I just issued G28 X > Emergency stop (m112) > G28 Y > Emergency stop with a slow homing speed and it works well.
 Interpolation is apparently not enabled by the sound of it. I left the sense_resistor value stock as its equivalent to my waterott specs.
Will test a print if required.
Klippy.log attached for config.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1734754/klippy.log)

andrea7376, 2018-02-18:
@KevinOConnor, sorry but I can't understand why restart with a completely new branch when @cruwaller branch is working with all register configurable and also the sensorless homing function work,
It's something I already ask to @cruwaller without an answer, isn't the opensource principle to share 
to grow up faster and better?
Maybe I'm lost somewhere?
Anyway this is the link of his fork:
https://github.com/cruwaller/klipper/tree/lpc176x

regards

KevinOConnor, 2018-02-18:
On Sun, Feb 18, 2018 at 04:40:22PM +0000, andrea7376 wrote:
> @KevinOConnor, sorry but I can't understand why restart with a completely new branch when @cruwaller branch is working with all register configurable and also the sensorless homing function work, [...]

I chatted briefly with cruwaller on irc a couple of days ago.  He
indicated that he is not planning to submit his changes to my repo.  I
can't easily extract his changes (and wouldn't without a
signed-off-by).

A number of people have been asking about the tmc2130 and it looks to
be only about a days worth of work.  So, it's not a big deal to
implement it if I need to.  :-(

-Kevin

NeelD87, 2018-02-18:
OK after playing with it further got it running, used same pin configurations which were working with marlin. 
Its homing now sensorless and all axis are moving. So time to do some actual printing.

TunaLatte, 2018-02-18:
Cruwaller branch implemented far beyond than tmc2130 support, things like temperature runaway protection, move without homing, class-like structure for heaters and stepper drivers etc in config. Though it doesn't go without saying kevin's branch is far more simpler and readable. Trying his latest branch was easy to do with the tmc2130. I prefer slow concise development over rushed ones. (esp documentation, took me a while to make cruwaller's branch work, so yeah its worth restarting assuming its simple)

TunaLatte, 2018-02-18:
@NeelD87 be sure to set stealth_max_speed:  300 to disable spreadcycle, report if you have any layer shifting. Also my current seems a bit low from the example as I'm running the drivers at 35v.

andrea7376, 2018-02-18:
@KevinOConnor Ok, now the situation it's clear for all :-( 
Thank you.

NeelD87, 2018-02-18:
@TunaLatte  just printed 20mm calibration cube and for few layers had small incremental X shift and later had big Y shift about 10mm.

NeelD87, 2018-02-18:
[printer.txt](https://github.com/KevinOConnor/klipper/files/1735087/printer.txt)

tiziano149, 2018-02-18:
Can't understand why @cruwaller wouldn't submit his changes but hey...
Really looking forward to this, I am running an i3 MEGA on the trigorilla board with TMC2130 stepsticks.

TunaLatte, 2018-02-18:
@NeelD87 Are you running your drivers over 18v? Trinamic says that you need 18v minimum for stealthChop. With Marlin at 35v i get immediate shifts at 100mm/s (kinda like stalling sound), Klipper/Duet works great, but on 12v both whine and stutter. Also if your drivers are not from waterott, they could have difference sense resistor value.

NeelD87, 2018-02-19:
Running them at 30V. Going to play with current and see. 
Waiting for @KevinOConnor implementation to test further.

NeelD87, 2018-02-19:
Seems teemuatlut has now worked on coolstep
https://github.com/teemuatlut/TMC2130Stepper

TunaLatte, 2018-02-19:
@NeelD87 that's an Arduino library completely irrelevant, and I donno but something smells fishy here.

NeelD87, 2018-02-19:
Its one used in Marlin now

TunaLatte, 2018-02-19:
> current:            500.0 #800 minimum under 18v

> Running them at 30V. Going to play with current and see.
But you mentioned before that you are on 12v, your config shows that. Your drivers are burning hot right now, and cruwaller's branch doesn't report pre-temp or overheating yet, sounds all like a lie to me, opting out.

NeelD87, 2018-02-19:
I am using 150W DC-DC 10-32V to 12-35V adjustable Step-UP 
Measured the voltage with multimeter.
Seems few others have similar issues
https://github.com/MarlinFirmware/Marlin/issues/9368#issuecomment-366426100

TunaLatte, 2018-02-19:
@NeelD87 We see what direction you are taking.

NeelD87, 2018-02-19:
Thanks for all your help it did help in getting started and debuging. 
 I cant see what lies you seem in the config and log and what you think is fishy.

NeelD87, 2018-02-19:
I am using ramps 1.4 board. Half success today , again thanks to @TunaLatte I have now reduced the current to 500.
[image]

TunaLatte, 2018-02-20:
@NeelD87 You seem to have a hardware issue, I've never seen such shifts on Marlin. 3rd time asking: are you using the waterott/official not chinese knock off?

TunaLatte, 2018-02-20:
Also, if your step-up is slow to power up it could damage and lead to similar issues. According to waterott, you should supply power to VM (motor) before VIO (driver) that is assuming that you are only running the drivers at this voltage. It wont matter if the entire system is 30v.

andrea7376, 2018-02-26:
Hi,
Today I got my TMC2130 and tested the cruwaller branch the printer (corexy) is moving and also the sensorless homing is working.
At the moment I didn't print anything. 
I have some strange things in the log file:

CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
INFO     printer :: Start printer at Mon Feb 26 21:30:10 2018 (1519680610.3 660.5)
INFO     config :: ===== Config file =====
[driver tmc2130_x]
type = TMC2130
steps_per_mm = 400
ss_pin = ar53
spi_mode = 3
spi_speed = 4000000
current = 600.0
microsteps = 32
mode = stealthChop
diag0_out = temp_prewarn
diag1_out = stall
diag0_active_high = 1
diag1_active_high = 1
interpolate = 1
sensor_less_homing = 1
stealth_max_speed = 50
stall_threshold = 2

[driver tmc2130_y]
type = TMC2130
steps_per_mm = 400
ss_pin = ar49
spi_mode = 3
spi_speed = 4000000
current = 600.0
microsteps = 32
mode = stealthChop
diag0_out = temp_prewarn
diag1_out = stall
diag0_active_high = 1
diag1_active_high = 1
interpolate = 1
sensor_less_homing = 1
stealth_max_speed = 50
stall_threshold = 2

[stepper_x]
driver = tmc2130_x
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = 0.00625
endstop_pin = ar3
position_endstop = 0
position_min = -1
position_max = 300
homing_speed = 25
homing_positive_dir = false

[stepper_y]
driver = tmc2130_y
step_pin = ar60
dir_pin = ar61
enable_pin = !ar56
step_distance = 0.00625
endstop_pin = ar14
position_endstop = 0
position_min = -1
position_max = 300
homing_speed = 25
homing_positive_dir = false

[driver drv8825_z]
type = DRV8825
microsteps = 32

[stepper_z]
driver = drv8825_z
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = 0.000625
endstop_pin = ^!ar18
position_endstop = 350.6
position_min = -10.0
position_max = 355.0
homing_speed = 5
homing_positive_dir = true

[extruder]
step_pin = ar36
dir_pin = ar34
enable_pin = !ar30
steps_per_mm = 900
nozzle_diameter = 0.300
filament_diameter = 1.750
heater = heater_nozzle
min_temp = 0
max_temp = 310
pressure_advance = 0.05
pressure_advance_lookahead_time = 0.020
max_extrude_cross_section = 99.0

[heater_nozzle]
heater_pin = ar10
min_temp = 0
max_temp = 300
control = pid
pid_kp = 10.186
pid_ki = 0.686
pid_kd = 37.816
sensor_pin = analog13
sensor_type = EPCOS 100K B57560G104F
rtd_nominal_r = 100
rtd_num_of_wires = 2
rtd_reference_r = 430
spi_mode = 1
spi_speed = 8000000
protect_period_heat = 20
protect_hysteresis_heat = 2
protect_period = 10
protect_hysteresis_runaway = 10

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog15
control = pid
pid_kp = 38.279
pid_ki = 1.994
pid_kd = 183.740
min_temp = 0
max_temp = 130
max_power = 0.75
protect_period_heat = 20
protect_hysteresis_heat = 4
protect_period = 10
protect_hysteresis_runaway = 10

[mcu]
serial = /dev/ttyACM0
pin_map = arduino
restart_method = arduino

[printer]
kinematics = corexy
max_velocity = 300
max_accel = 3000
max_z_velocity = 30
max_z_accel = 300
motor_off_time = 600
require_home_after_motor_off = false
allow_move_without_home = true

=======================
INFO     printer.stepper.x :: steps per mm 160.0 , step in mm 0.00625
INFO     printer.stepper.y :: steps per mm 160.0 , step in mm 0.00625
INFO     printer.stepper.z :: steps per mm 1600.0 , step in mm 0.000625
INFO     printer.coreXY :: Kinematic created: coreXY
INFO     printer.extruder.stepper :: steps per mm 900.0 , step in mm 0.00111111111111
INFO     printer.mcu.mcu.serial :: Starting serial connect
INFO     printer.mcu.mcu.serial :: Loaded 59 commands (v0.5.0-135-ga51d851-20180226_203056-octopi / gcc: (GCC) 4.9.2 binutils: (GNU Binutils) 2.26.20160125)
INFO     printer.mcu.mcu.serial :: MCU config: ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 MCU=atmega2560 STATS_SUMSQ_BASE=256
INFO     printer.mcu.mcu :: Sending MCU 'mcu' printer configuration...
INFO     printer.mcu.mcu :: Configured MCU 'mcu' (636 moves)
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
INFO     printer.stepper.x.tmc2130 :: Stealth max speed: 50 (197)
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
INFO     printer.stepper.x.tmc2130 :: Stand still
INFO     printer.stepper.x.tmc2130 :: DRV_STATUS : Current (CS) = 9, stallGuard2 result (SG) = 0
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.x.tmc2130 :: Reset has occurred!
ERROR    printer.stepper.x.tmc2130 :: COOLCONF Configuration error! [was 0x00000000 expected 0x00020000]
ERROR    printer.stepper.x.tmc2130 :: PWMCONF Configuration error! [was 0x00000000 expected 0x000505FF]
ERROR    printer.stepper.x.tmc2130 :: IHOLD_IRUN Configuration error! [was 0x00000000 expected 0x00001209]
INFO     printer.stepper.x.tmc2130 ::  init done!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
INFO     printer.stepper.y.tmc2130 :: Stealth max speed: 50 (197)
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
INFO     printer.stepper.y.tmc2130 :: Stand still
INFO     printer.stepper.y.tmc2130 :: DRV_STATUS : Current (CS) = 9, stallGuard2 result (SG) = 0
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
WARNING  printer.stepper.y.tmc2130 :: Reset has occurred!
ERROR    printer.stepper.y.tmc2130 :: COOLCONF Configuration error! [was 0x00000000 expected 0x00020000]
ERROR    printer.stepper.y.tmc2130 :: PWMCONF Configuration error! [was 0x00000000 expected 0x000505FF]
ERROR    printer.stepper.y.tmc2130 :: IHOLD_IRUN Configuration error! [was 0x00000000 expected 0x00001209]
INFO     printer.stepper.y.tmc2130 ::  init done!

Can someone figure out what is happening? thank you.
Andrea

TunaLatte, 2018-02-26:
They are pretty normal and exactly like my log, these are unconfigured
features and the debugging level is verbose.

cruwaller, 2018-02-27:
I fixed the errors yesterday. Those are meaningless and my typo :) 
Get a pull and try again....

[101 more replies]

closed 2018-05-29 by KevinOConnor