Klipper3d/klipper #102: printer stops due MCU shutdown

TylonHH, 2018-01-05:
Hi,
I used klipper for few days without any big trouble.
Today I had a bigger print about 6hr. Nearly exact 50% the print stops. Coincidence?
It seems that the hot parts (hotend, bed) wear still aktive and shut down by the plugin HeaterTimeout
Now I have to find out what I can do better to prevent this.

Here is my setup and all my logs.
Raspberry Pi 3
Ramps 1.4
Atmega 2560
Octoprint 1.3.6

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1607939/printer.cfg.txt) (renamed for uploading)
[gcode.zip](https://github.com/KevinOConnor/klipper/files/1607921/gcode.zip)
[klippy.log](https://github.com/KevinOConnor/klipper/files/1607922/klippy.log)

```
Read 578996.910328: 'N655899 G1 X181.374 Y165.535 E0.8396*97\n'
Read 578996.913739: 'N655900 G1 X181.155 Y165.315 E0.8489*108\n'
Read 578996.917324: 'N655901 G1 X181.045 Y165.771 E0.8629*99\n'
gcode state: absolutecoord=True absoluteextrude=True base_position=[-0.0, -0.0, -0.0, 19152.017599995204] last_position=[181.045, 165.771, 47.4, 19152.880499995204] homing_add=[0.0, 0.0, 0.0, 0.0] speed=56.0
MCU 'mcu' shutdown: Rescheduled timer in the past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.
```

and here the graphical debug of the mcu, wich looks ok for me.
[image]

Luckily I could finish my print, because the Z high was logged.

But this situation now its not reliable for me. How can I found the reason?
I also reconized one time, a print stop for an half second while I reloaded the weppage of octoprint.

dragonnn, 2018-01-05:
> I also reconized one time, a print stop for an half second while I reloaded the webpage of octoprint.
This is octoprint problem - the g-code reader/sender isn't running in separate thread. Yeah this is ridiculous but they claim they can not change it because backwards compatibly.  And this happened in my case on Marlin

TylonHH, 2018-01-05:
the print stops at this high. As you can see there was nothing special what he not printed before
[image]

TylonHH, 2018-01-05:
Added Inforamtion: I use TMC2100 without jumper, so they are in 16µ steps with interpolated 256

TylonHH, 2018-01-05:
#56 

I changed for an workaround in the ``src/avr/timer.c`` 
from ``timer_set(timer_get() + 50);`` to ``timer_set(timer_get() + 100);``

But I still have to test it

KevinOConnor, 2018-01-06:
On Fri, Jan 05, 2018 at 09:28:59PM +0000, TylonHH wrote:
> Hi,
> I used klipper for few days without any big trouble.
> Today I had a bigger print about 6hr. Nearly exact 50% the print stops. Coincidence?
> It seems that the hot parts (hotend, bed) wear still [...]

Unfortunately, you've hit a bug in the Klipper pressure advance code.
The bug was triggered by some odd g-code in your print:

G1 X182.900 Y157.467 E4.1681 F1050
; outer perimeter
G1 X182.900 Y157.468 E4.1861 F2100

In the second move above, Klipper was instructed to extrude 18 microns
of filament while moving the Y axis 1 micron.  This exceeds the
max_extrude_cross_section check, but the Klipper host code allowed it
because it is a small amount of filament (18 microns).  Unfortunately,
the pressure advance code interpretted it as a spike in extruder flow
rate and subsequently amplified the spike to "account for pressure".

When the massive extruder movement arrived at the micro-controller,
the mcu detected the excessive stepper movement and properly raised an
error.

I'll put together a fix.  In the interim, you could disable pressure
advance.

> I changed for an workaround in the ``src/avr/timer.c``
> from ``timer_set(timer_get() + 50);`` to ``timer_set(timer_get() + 100);``
> 
> But I still have to test it

The above change is not necessary (or desirable).  The mcu code worked
as expected.  I would revert to the standard code for any future
prints.

-Kevin

TylonHH, 2018-01-06:
I'm glad to hear that you found the issue. Very impressing to see that in a gcode.
I disabled pressure advance, maybe its also better with my bowden setup.

gdachs, 2018-01-06:
Seeing this bug I got the idea that I might be a good idea to collect gcode sequences like this. If I would be you, I would setup a jenkins job with klipper and simulavr that could do regression tests with gcode files like this.

TylonHH, 2018-01-06:
Today I tried to print a similar object. A case just with diffenrent letters as a relif.
And it failed too. But this time I guess its not an klipper issue.
So I review the timelaps, the gcode and the klipper log.

A picture of the mess

Here is the loadgraph
[image]
Look fine to me

[the timelaps](https://youtu.be/RuFvE-gDM94) (Youtube Video, 2 minutes)
[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/1609363/klippy.log.txt) (12MB)

What I found out so far:
it seems that the Z achse fall a little bit or moved more then it should. Check out the timelaps video and look at the bottom of the object.
But what is realy strange, is that what I found when I looked to Octoprint. I saw this screen

You see, that it seems that Octoprint hang or kinda stopped.

Im not familiar to setup a jenkins job with klipper. 

I hope it was just my Z achse. I'll remount it.

gdachs, 2018-01-06:
sorry, my post was for KevinOConnor, not for you TylonHH

KevinOConnor, 2018-01-08:
This should be fixed as of commit f6f174ab3.

C4g2, 2018-01-11:
Im getting a similar yet different error. I got everything installed and even printed up a calibration cube with Klipper! About the end of my very next print, printer just restart itself and has started throwing these errors. Now when I tell octoprint to start a file, it seems like it trys to pre-heat the bed, then just resets after a few seconds, I'm at a bit of a loss here.
```
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: !! Printer is shutdown
```

I've tried attaching klippy.log but it wont take it, next post will be the file, so sorry for long post

C4g2, 2018-01-11:
```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.5.0-74-g83e9e92-dirty'
CPU: 4 core ARMv7 Processor rev 5 (v7l)
Python: '2.7.9 (default, Sep 17 2016, 20:26:04) \n[GCC 4.9.2]'
Start printer at Thu Jan 11 05:53:11 2018 (1515671591.7 1226.2)
===== Config file =====
[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC2
position_endstop = -30
position_max = 220
position_min = -30
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC3
position_endstop = -8
position_min = -8
position_max = 220
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = !PB2
enable_pin = !PA5
step_distance = 0.0025
endstop_pin = ^!PC4
position_endstop = 0.5
position_max = 240
homing_speed = 20

[extruder]
step_pin = PB1
dir_pin = PB0
enable_pin = !PD6
step_distance = .0105
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA7
control = pid
pid_kp = 2.151492
pid_ki = 0.633897
pid_kd = 230.042965
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = PD4
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA6
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = PB4

[mcu]
serial = /dev/ttyUSB0
[... 134 lines cut ...]
Printer is shutdown

Timeout with MCU 'mcu' (eventtime=1307.693770)
Stats 1307.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=7 freq=16000630
Timeout with MCU 'mcu' (eventtime=1308.694399)
Stats 1308.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=16000630
Timeout with MCU 'mcu' (eventtime=1309.695389)
Stats 1309.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=9 freq=16000630
Timeout with MCU 'mcu' (eventtime=1310.696488)
Stats 1310.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=10 freq=16000630
Timeout with MCU 'mcu' (eventtime=1311.697175)
Stats 1311.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=11 freq=16000630
Timeout with MCU 'mcu' (eventtime=1312.697596)
Stats 1312.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=12 freq=16000630
Timeout with MCU 'mcu' (eventtime=1313.698134)
Stats 1313.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=13 freq=16000630
Timeout with MCU 'mcu' (eventtime=1314.698723)
Stats 1314.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=14 freq=16000630
Timeout with MCU 'mcu' (eventtime=1315.699442)
Stats 1315.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=15 freq=16000630
Timeout with MCU 'mcu' (eventtime=1316.700273)
Stats 1316.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=16 freq=16000630
Timeout with MCU 'mcu' (eventtime=1317.701284)
Stats 1317.7: gcodein=170 print_time=79.142 buffer_time=0.000 print_stall=0 mcu: mcu_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=1250 bytes_read=10714 bytes_retransmit=27 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.007 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=17 freq=16000630
```

KevinOConnor, 2018-01-11:
On Thu, Jan 11, 2018 at 12:01:23PM +0000, C4g2 wrote:
> Im getting a similar yet different error. I got everything installed and even printed up a calibration cube with Klipper! About the end of my very next print, printer just restart itself and has [...]

The above error is unrelated to this github issue.  The above is a
communication outage - check your cabling and power supply to the
micro-controller.

Also, not related to your error, but you do not want to use 57600 baud
for the main Klipper data protocol - you should go back to the default
250000 baud rate.

-Kevin

C4g2, 2018-01-11:
Good to know its not software related and too have a place to start looking. Also thanks for the config tip!

closed 2018-01-16 by KevinOConnor