Klipper3d/klipper #103: wired temperature while not heated, jumping around 3-5 degrees

TylonHH, 2018-01-05:
After everything is cooled down I noticed that in octoprint the temperature is not very clear
[image]

I set the ``EPCOS 100K B57560G104F`` sensor type and the PID Tuning. While heating and printing this dosnt looks so wired.

Does anybody has this (issue) too? Or is this just normal and I should ignore it?

darenschwenke, 2018-01-05:
ADC resolution near room temperature on a 100k thermistor is not very
good.
The voltage divider is 100k/4.7k at that point so the corresponding swing
in voltage for the ADC to measure is very near it's resolution and the math
then makes the resulting calculated temperature move a lot more.
Ignore it.

KevinOConnor, 2018-01-06:
On Fri, Jan 05, 2018 at 09:57:43PM +0000, TylonHH wrote:
> After everything is cooled down I noticed that in octoprint the temperature is not very clear

That's weird.  OctoPrint always scales the graph so that it is between
0 and 300 degrees for me, so these types of fluctuations aren't
usually noticeable.

I don't typically see that much noise on my printers, but I don't
think it's an active problem.  I'd guess there's a little more
electrical noise on your thermistor lines.

-Kevin

TylonHH, 2018-01-08:
This is the same setup but with Marlin
[image]
 Thats why I noticed it here.

But it's fine.

closed 2018-01-08 by TylonHH