Klipper3d/klipper #104: Static PWM fan responds to M106 and M107

sapell, 2018-01-05:
Noticed that the fan that I have configured to be static ([static_pwm_output board_fan] in the config) does not turn on when Klipper starts and also responds to M106 and M107 g-codes. As the documentation suggests that the static pwm could also be utilized for controlling e.g. stepper current, it probably shouldn't respond to M106 and M107.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1608056/klippy.log.gz)

jakep82, 2018-01-05:
I noticed this too, but realized that's not actually what's happening.  My static PWM fan stays off until anything else on the board is activated.  So the static PWM fan will also turn on when I run G28 or turn on the hotend or bed, and turns off after all heaters, fans and steppers turn off.  I personally like this behavior as I'm using it to cool the Replicape, and I only want that fan running when the printer is doing something that might generate heat.

KevinOConnor, 2018-01-06:
On Fri, Jan 05, 2018 at 10:13:16PM +0000, Simo Apell wrote:
> Noticed that the fan that I have configured to be static ([static_pwm_output board_fan] in the config) does not turn on when Klipper starts and also responds to M106 and M107 g-codes. As the [...]

As @jakep82 reports, this is due to the peculiarities of the replicape
enable pin - the outputs from the pca9685 chip are only activated when
the P9_41 enable pin is driven low.  The Klipper host code makes sure
to do that on any fan, heater, or stepper enable - but it isn't doing
that for the static_pwm_outputs.  We could change it to do that, but I
suspect it's not worth the effort.

The heater_fan code does interact with the global enable line, so you
might want to consider using that.

-Kevin

sapell, 2018-01-06:
As the Replicape steppers are not properly disabled (issue #62), the stepper drivers are running quite hot even without activity. Therefore I'd prefer the static fan to be enabled all the time. Luckily I have two Replicape independent 40mm fans blowing on top Replicape and the static fan is only blowing fresh air towards Replicape (I have the electronics underneath the build bed on my delta).

@KevinOConnor: I have some prior experience of controlling the PCA9685 (have used it at work). I think the enabling/disabling a fan output independently requires writing the PCA9685 LEDn_ON/-OFF registers values producing 0% duty-cycle to disable and appropriate values producing the desired duty-cycle to enable rather than relying on the global enable.

The heater_fan (cold end) I have configured to enable only when the heater is heating.

KevinOConnor, 2018-01-07:
On Sat, Jan 06, 2018 at 10:35:35AM +0000, Simo Apell wrote:
> As the Replicape steppers are not properly disabled (issue #62), the stepper drivers are running quite hot even without activity. Therefore I'd prefer the static fan to be enabled all the time. [...]

The code should be doing that already (see src/linux/pca9685.c).

Looking at the Replicape schematic further, I think I misunderstood
how the (many) enable lines work on the Replicape.  It looks like the
P9_41 is solely a pca9685 enable line (I originally thought it also
impacted the shift registers).  So, that makes it easier to control
the P9_41 line.

Also, looks like the stepper enable lines from the shift register
aren't configured properly - it's not one line per stepper, but a
series of enable features (stepper_xyz, servo_1, servo_0, stepper_eh,
stepper_tristate).  The Replicape really is a goofy board.  On the
upside, I think it should be okay to change the code so that the
stepper enable lines are all off until the first stepper is enabled
and then to disable them all once all steppers are disabled.

I'll try to put some code together for testing.

-Kevin

sapell, 2018-01-07:
Yes, I too think it's a bit complicated, but the I/O resources on the BeagleBone are after all quite limited. The HDMI and eMMC virtual capes take up many of the I/O:s and I believe thar the Replicape design has been done so that it wouldn't conflict with those so that users could use HDMI and eMMC if needed. Hence the shift registers and external PMW generator instead of direct I/O pins. 

Thanks a lot for the effort, looking forward to the release!

jakep82, 2018-01-08:
Looking forward to this.  I finally got my new printer (Hypercube) running with the Replicape last night.  With my steppers set at 0.6, I do notice some heat even at idle.  For now I've set my hotend fan to turn on at 10C which means it runs all the time, and that keeps the static pwm fan I have cooling the board on all the time as well.

sapell, 2018-01-09:
Tested the branch work-replicape-20180108 which should address also this issue.

The static pwm fan turns on expectedly when e.g. issuing firmware_restart, but still responds to M106 and M107 when stepper activity is not overriding the static pwm fan control.

KevinOConnor, 2018-01-11:
Should be fixed in commit 47e45821.

closed 2018-01-11 by KevinOConnor