Klipper3d/klipper #105: MCU 'mcu' shutdown: backwards adc

khseal, 2018-01-06:
I have a new bug. Here is the log
[klippy.log](https://github.com/KevinOConnor/klipper/files/1609304/klippy.log)

KevinOConnor, 2018-01-08:
This isn't a new error - it's part of the debugging added during the investigation of issue #56.  Lets continue the discussion there.

closed 2018-01-08 by KevinOConnor

KevinOConnor, 2018-01-08:
On Sat, Jan 06, 2018 at 11:03:06PM +0000, Sergey wrote:
> I have a new bug. Here is the log
> klippy.log
> Twice printed and an error occurred twice.

This log shows the same symptoms as previous logs - high variance in
ADC response times.  The debugging further indicates that the time it
takes to read the ADC values is highly variable - normally a reading
(with 8 oversamples) takes about 8 milliseconds to complete.  However,
we see in various places in the log where the cycle is taking over
250ms (see the "#output: lst" lines).

> I'll try to roll back to the old version.
> It seems the error is due to the poor power of the AVR controller.

I'm inclined to agree that this is something specific to this
particular chip or setup.

-Kevin