Klipper3d/klipper #106: Configuration for Mini Rambo and Prusa i3 MK2S

Science400, 2018-01-06:
My stepper motors just hum and don't ever move on my Prusa i3. 

Kevin already pointed out that the Mini Rambo uses a different stepper current control
mechanism than the Rambo. It uses PWM control for the current instead of a digital trimpot. 

How is this handled and how can I configure Klipper for the Prusa?
[klippy.log](https://github.com/KevinOConnor/klipper/files/1609339/klippy.log)

KevinOConnor, 2018-01-07:
Can you test with the attached config?
[generic-mini-rambo.cfg.txt](https://github.com/KevinOConnor/klipper/files/1610023/generic-mini-rambo.cfg.txt)

KevinOConnor, 2018-01-16:
Any update on this?

Science400, 2018-01-17:
Between a new semester of school starting up and having to completely disassemble my extruder, I haven't had time to test things until last night. 

Your PWM values seemed to work just fine. I'm curious how you came up with those values.

[static_pwm_output xy_current]
pin: PL3
scale: 2.0
value: 1.3
hard_pwm: 32640

[static_pwm_output z_current]
pin: PL4
scale: 2.0
value: 1.3
hard_pwm: 32640

[static_pwm_output e_current]
pin: PL5
scale: 2.0
value: 1.25
hard_pwm: 32640

Before you had posted I had come up with these values.
[static_pwm_output xy_current]
pin: PL3
value: 0.270
scale: 1
hard_pwm: 1

[static_pwm_output z_current]
pin: PL4
value: 0.830
scale: 1
hard_pwm: 1

[static_pwm_output e_current]
pin: PL5
value: 0.450
scale: 1
hard_pwm: 1

I found some forum post that said that these are the currents in amps that the normal Prusa firmware gives the motors in silent mode. I never noticed any problems with torque with either sets of values. I do, however, think that your values made the motors run just a little bit hotter.

I have a little more work to do to get things just right, but I can share my config file if that would help someone else. Thing are looking great by the way.

KevinOConnor, 2018-01-20:
On Wed, Jan 17, 2018 at 05:49:22AM +0000, Science400 wrote:
> Between a new semester of school starting up and having to completely disassemble my extruder, I haven't had time to test things until last night. 
> 
> Your PWM values seemed to work just fine. I'm curious how you came up with those values.

I pulled them from the Marlin defaults for the mini rambo board.  The
amps do look a bit high there, so it may make sense to reduce them
(the hard_pwm and scale parameters are probably fine).

I added config/generic-mini-rambo.cfg file to the repo as part of
commit 84c623e7.

-Kevin

closed 2018-01-20 by KevinOConnor

dbacon, 2018-02-10:
@Science400 I can independently confirm your pwm values xy, z, e = 0.27, 0.83, 0.45 - I started with the values sent by prusa firmware and converted for use in klipper.  Using the pwm values from generic-mini-rambo.cfg resulted in the steppers being far too hot to touch.

KevinOConnor, 2018-02-11:
On Sat, Feb 10, 2018 at 09:34:47AM -0800, Dave Bacon wrote:
> @Science400 I can independently confirm your pwm values xy, z, e = 0.27, 0.83, 0.45 - I started with the values sent by prusa firmware and converted for use in klipper.  Using the pwm values from generic-mini-rambo.cfg resulted in the steppers being far too hot to touch.

Thanks.  Could you send a pull request with an updated
generic-mini-rambo.cfg?

-Kevin