Klipper3d/klipper #109: Heater fan turned off if print was stop from octoprint

LucaBertuolo, 2018-01-07:
I've spend some time to test this issue many times, and I always get this problem:
If a print is stopped after some layer from the "cancel" button of octoprint, the heater fan configured in printer.cfg is turned off..
This cause that the heater is not cooled down to 50~

A strange things is that if the fan_speed parameter is set to 0.8, then when a print is stopped the fan will spin very slow, like about fan_speed 0.3...
but if fan_speed is set to 1.0, then when a "cancel" is pressed the fan stop at all..

Fan_spped= 1.0 -> [klippy.log](https://pastebin.com/7dDSFFfH)
Fan_spped= 0.8 -> [klippy.log](https://pastebin.com/69gJDKmg)

I take this "opportunity" to say that I'm using klipper on a Orange Pi Zero Plus without apparently no one problem during print...
i will upload some video asap :+1:

KevinOConnor, 2018-01-07:
On Sun, Jan 07, 2018 at 12:21:47PM -0800, LucaBertuolo wrote:
> I've spend some time to test this issue many times, and I always get this problem:
> If a print is stopped after some layer from the "cancel" button of octoprint, the heater fan configured in printer.cfg is turned off..

Be sure to only use each pin once in the config file.  If ar7 is used
for the heater_fan section, it must not be also used in the fan
section.

-Kevin

LucaBertuolo, 2018-01-08:
You are right, it's my mistake...
I've swap from a RAMPS to a MKS BASE that should be ramps-compatible but it use a different pinout for the fan...

Maybe it can be added to the FAQ or "bug reporting" explanation...

closed 2018-01-08 by LucaBertuolo

manu7irl, 2018-11-01:
> I've spend some time to test this issue many times, and I always get this problem:
> If a print is stopped after some layer from the "cancel" button of octoprint, the heater fan configured in [...]

can you please contact me I am trying to do the same with a Orange Pi Zero...
manu7irl@gmail.com
thanks!! 
[image]