Klipper3d/klipper #115: Move out of range error raised when moving relative Xmm from home and back

sapell, 2018-01-09:
Klipper raises movement out of range error on my delta printer when I:
1. Home to top,
2. Command relative move of Xmm (e.g. 100mm) down and
3. Command relative move of same amount up (back to home).
That movement should be allowed.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1616794/klippy.log.gz)

KevinOConnor, 2018-01-10:
On Tue, Jan 09, 2018 at 09:42:16PM +0000, Simo Apell wrote:
> Klipper raises movement out of range error on my delta printer when I:
> 1. Home to top,
> 2. Command relative move of Xmm (e.g. 100mm) down and
> 3. Command relative move of same amount up (back to home).
> That movement should be allowed.

This is normal behavior for Klipper.

The code models the available print volume as a cylinder with a cone
on top of it.  There are certainly valid moves that fall outside of
this volume, but it's too complex to verify every possible move.  In
particular, when each tower has a different arm length, different
angle, and/or different endstop position there are moves that are very
difficult to validate.

In this particular case, the homing position is close enough to an
invalid posititon that I don't see it being used in any real-world
print.  So, I don't think it's worth the additional code complexity to
support moving to that particular position.

-Kevin

sapell, 2018-01-10:
Ok, thanks for the explanation, that is understandable. I was worried if this behavior would imply of some error in the movement calculation routines. Closing the issue.

closed 2018-01-10 by sapell