Klipper3d/klipper #117: pin_map issues

agentnoise, 2018-01-10:
I am trying to configure the printer without arduino aliases. When I remove or comment out the pin_map line it get the error "Unable to translate pin name: config_soft_pwm_out oid=19 pin=ar44 cycle_ticks=160000 value=0 default_value=1 max_duration=0" when is send a STATUS command. 

I do not see that pin anywhere in the the config file the example config file

KevinOConnor, 2018-01-10:
On Tue, Jan 09, 2018 at 09:00:07PM -0800, Max Holden wrote:
> I am trying to configure the printer without arduino aliases. When I remove or comment out the pin_map line it get the error "Unable to translate pin name: config_soft_pwm_out oid=19 pin=ar44 [...]

To report an error, please include the /tmp/klippy.log file as
described in docs/Contact.md.

-Kevin

agentnoise, 2018-01-10:
Apologies, here is the log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/1620766/klippy.log)

KevinOConnor, 2018-01-10:
On Tue, Jan 09, 2018 at 09:00:07PM -0800, Max Holden wrote:
> I am trying to configure the printer without arduino aliases. When I remove or comment out the pin_map line it get the error "Unable to translate pin name: config_soft_pwm_out oid=19 pin=ar44 [...]

Your config file has the following:

[heater_fan extruder_fan]
pin = ar44

You can only use "ar44" if you use the arduino pin mappings.

-Kevin

closed 2018-01-16 by KevinOConnor