Klipper3d/klipper #118: Massive Z Axis Problem

Philjayhill, 2018-01-11:
I was able to install everything super about the Raspberry Pi, but the Z-axis doesn't work at all... I'm really desperate! G28 xy is no problem, but the G28 Z doesn't work... the stepper motor moves only for a short time and is extremely loud... does anyone know this problem? a stepper motor has already broken... My board is a MSK Base V1.5

KevinOConnor, 2018-01-11:
On Thu, Jan 11, 2018 at 09:16:06AM -0800, Philjayhill wrote:
> I was able to install everything super about the Raspberry Pi, but the Z-axis doesn't work at all... I'm really desperate! G28 xy is no problem, but the G28 Z doesn't work... the stepper motor moves [...]

In order to help, we need the klippy.log file as described at:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Further, please be sure to see the FAQ:

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-converted-my-config-from-marlin-and-the-xy-axes-work-fine-but-i-just-get-a-screeching-noise-when-homing-the-z-axis

darenschwenke, 2018-01-11:
Barring any obvious actual software issues here, I would say the step rate
being applied to Z is wrong (too fast) or your board is damaged.
If you accelerate/move faster than a stepper can handle, it will move for a
short time, and then just sit and make a loud noise until the end of the
move.

A bad output transistor within the stepper driver chip itself can also
cause a similar result though, and this is fairly easy to do if the stepper
is ever unplugged while operating.
Good news if it is the latter and you only have one extruder, is you could
re-assign the second extruder to be the Z axis.

Philjayhill, 2018-01-12:
Hello Kevin
I followed your instructions and adjusted the printer. cfg so that it works for me, I also adjusted the max_z_velocity. But unfortunately, the Z axis is still not working. Attached the. log file
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1625599/klippy.log.zip)

Philjayhill, 2018-01-12:
Which step_distance its common? Yes i use lead screw. The construction based on Flsun Cube. what do you mean by Pitch?

gdachs, 2018-01-12:
If you don't know this things, how was you able to calculate the step_distance?
Pitch is the length in millimeter the nut is moved by the lead screw if you rotate it one turn.
A common pitch is 8 mm. A typical stepper has a step angle of 1.8 °. Simple stepper drivers use 1/16 micro steps. that would mean step_distance is: 1/(((360/1.8)*16)/8) == 0.0025

Philjayhill, 2018-01-13:
I found out what the problem was, a connection was broken, a cold solder joint. But now I have another problem, my printer doesn't start anymore.
"Recv: // The klippy host software is attempting to connect.  please
Recv: // retry in a few moments.
Recv:!! Printer is not ready"
In the appendix my klippy log. hopefully someone can help me
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1628766/klippy.log.zip)

theopensourcerer, 2018-01-13:
@Philjayhill 

> Unable to open port: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

The USB Port has probably moved. Check /dev. It might be ttyUSB1 - this has happened to me,,,

@KevinOConnor It might be an idea to mention about how to add a udev rule so the same USB device is always associated with the USB ID of the board?

i.e. http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/

Philjayhill, 2018-01-13:
I have tried everything from USB0 to USB4, but unfortunately nothing works...

theopensourcerer, 2018-01-13:
Have you followed the helpful instructions in the FAQ about looking for the serial port?

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Philjayhill, 2018-01-13:
Here the USB0 is shown to me strangely enough. I'm confused!

Philjayhill, 2018-01-13:
i'm pretty much assuming i don't have to flash again to change the serial port? i just need to change that in printer. cfg

theopensourcerer, 2018-01-13:
Every time you unplug/re-plug the board into the pi, the USB device (i.e. /dev/ttyUSB0) is destroyed/re-created. Sometimes, depending on timing, it may get a different device name. [Hence my comment earlier to Kevin about using a udev rule to create a persistent name]

Yes, you only need to change the entry in printer.cfg.

KevinOConnor, 2018-01-13:
On Sat, Jan 13, 2018 at 03:06:27PM -0800, Alan Lord wrote:
> Every time you unplug/re-plug the board into the pi, the USB device (i.e. /dev/ttyUSB0) is destroyed/re-created. Sometimes, depending on timing, it may get a different device name. [Hence my comment earlier to Kevin about using a udev rule to create a persistent name]

You don't need to use udev - if one follows the directions at:

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

and uses the /dev/serial/by-id/xyz name then it should be stable.

-Kevin

Philjayhill, 2018-01-14:
Now the things we discussed here work. thanks for that. everything is installed now and I wanted to make the first print, nozzle and heatbed warm up correctly xyz homing works. when the print should start everything goes down again and cools down again.
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1629313/klippy.log.zip)

GitHub1712, 2018-01-14:
Maybe your axis are out of reference, klipper does not print with disabled
steppers if you have no homing in print gcode

@Kevin is it possible to change configuration to print without reference?

For laser engraving and milling I like to drive to or set needed
coordinates, disable steppers and move axis by hand to that point

KevinOConnor, 2018-01-14:
On Sun, Jan 14, 2018 at 04:25:06AM -0800, Philjayhill wrote:
> Now the things we discussed here work. thanks for that. everything is installed now and I wanted to make the first print, nozzle and heatbed warm up correctly xyz homing works. when the print should start everything goes down again and cools down again.

Something in your print requested the head to move past the permitted
y range and Klipper properly raised an error.

Move out of range: 41.884 -13.489 0.300 [3.000]

OctoPrint used to display a popup when an error occurred - it doesn't
seem to do that anymore - not sure why.

-Kevin

gdachs, 2018-01-14:
@KevinOConnor: sorry, but if I look at the construction of this Flsun Cube it looks like his step_distances for X and Y axes are wrong too.
@Philjayhill recalculate them please

Philjayhill, 2018-01-14:
@gdachs sorry but, i'm a bloody beginner in 3d printing, how can I recalculate that?

gdachs, 2018-01-14:
Where did you get the number from then?
I am a beginner too, but I could calculate them for you. In the calculation for the Z axes I made a lot of assumptions, because I don't know what hardware you have and you didn't even verify one of this assumptions.
Make first your homework:
- What is the step angle of your stepper motors?
- how many microsteps are the drivers of your controller doing?
- what pitch does you lead screw have?
- how many teeth does the drive gears for X an Y motors have?
- do you use GT2 belts with 2 mm teeth distance?

gdachs, 2018-01-14:
It might be even more easy, if you could provide a configuration.h file from your stock firmware.

gdachs, 2018-01-14:
Okay, I did your homework and have found this: 
`/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   {100,100,400,150}`
ergo:
- X step_distance = 1/100 = 0.01
- Y step_distance = 1/100 = 0.01
- Z step_distance = 1/400 = 0.0025
So my assumptions from above seemed to be right.
Extruder is wrong too:
- E step_distance = 1/150 = 0.00666666666

I believe the next problem you will get with the thermistors. I doubt that these Chinese printers really use thermistors from EPCOS.

Philjayhill, 2018-01-14:
@gdachs you are great! thanks for your help! I will try the the calculation

gdachs, 2018-01-14:
Looks like I have found the same, so my calculations should be okay

Philjayhill, 2018-01-14:
i'm a little desperate and frustrated right now. i don't have any ideas how i can solve this with the move out of range: 41.884 -13.489 0.300[3.000]. has someone got a way to do it?

gdachs, 2018-01-14:
So you still get this after you changed the configuration and restarted klipper? Post a fresh log file.

Philjayhill, 2018-01-14:
@gdachs 
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/1629674/klippy.log.zip)

Philjayhill, 2018-01-14:
Yes i restart many times

gdachs, 2018-01-14:
Okay, then I am out of business. I guess @KevinOConnor would like to see the corresponding gcode file.

Philjayhill, 2018-01-14:
@gdachs you mean the gcode file I want to print?

gdachs, 2018-01-14:
Huh? What else could I mean?

Philjayhill, 2018-01-14:
@KevinOConnor  Here its the gcode file from the test print, hope it helps.... 
[incenseBurnertest.gcode.zip](https://github.com/KevinOConnor/klipper/files/1629688/incenseBurnertest.gcode.zip)

gdachs, 2018-01-14:
Look into the gcode file!
`;RAFT
G0 F9000 X41.884 Y-13.489 Z0.300`
There is your -13.489. I don't know what you told your slicer to generate that

KevinOConnor, 2018-01-14:
On Sun, Jan 14, 2018 at 07:08:55PM +0000, Philjayhill wrote:
> @KevinOConnor  Here its the gcode file from the test print, hope it helps.... 
> [incenseBurnertest.gcode.zip](https://github.com/KevinOConnor/klipper/files/1629688/incenseBurnertest.gcode.zip)

It looks like this g-code was sliced for a delta printer.  Do you have
a delta printer?  If so, you'll need to update the printer config file
to use delta kinematics.  If not, you'll need to slice the model with
the correct slicer settings.

-Kevin

Philjayhill, 2018-01-14:
No i have a Flsun Cube

gdachs, 2018-01-14:
Then you configured your Slicer for the wrong printer.

gdachs, 2018-01-14:
It is absolutely clear now, that this is not a klipper issue

Philjayhill, 2018-01-14:
shame on me! Sorry for that. I have now slice it with Cura and the right profile now it works, thanks to all who have helped me.

closed 2018-01-15 by Philjayhill