Klipper3d/klipper #120: Controller Fan

amaximchuk, 2018-01-12:
Marlin supports controller fan to cool down the stepper drivers and MOSFETs. The fan automatically turns on when stepper is enabled and turn off after all steppers are turned off. 
Is it possible to get this function in Klipper?

KenshiHH, 2018-01-12:
Honest question, why?
Why not let the Fan run all the time? One less source of possible failures while printing.

gdachs, 2018-01-12:
I don't think that this is necessary. Look at this document: https://github.com/KevinOConnor/klipper/blob/master/docs/Protocol.md
It shows you how to use extra pins. Add the commands, that switches on/off the pin for the fan, to your start/stop gcodes and you are done.

KevinOConnor, 2018-01-12:
On Fri, Jan 12, 2018 at 12:39:34PM +0000, amaximchuk wrote:
> Marlin supports controller fan to cool down the stepper drivers and MOSFETs. The fan automatically turns on when stepper is enabled and turn off after all steppers are turned off. 
> Is it possible to get this function in Klipper?

I'd do this by declaring a [heater_fan] config section (see
config/example-extras.cfg).  It will turn the fan on whenever the
extruder heater is on.

If for some reason you don't want that, you could also use the
multi_pin feature (connected to one of the stepper enable lines) or
use the pwm_output feature (along with manually issuing SET_PIN in
g-code) - both are also described in config/example-extras.cfg.

-Kevin

amaximchuk, 2018-01-12:
Thank you for the answer! I'll try both variants.

closed 2018-01-13 by amaximchuk

lenne0815, 2018-11-24:
I ran into the same issue but couldnt solve it by linking an ena pin ( Apparently motor pins cant be used anymore ? ) 

My scenario is this: Stepsticks pretty much overheat instantly when not actively cooled and a lengthy bed probe, chaos ensues.

Is there any other way then gcode activation of a pin ? Problem with that it that it puts me in charge which never works out well...