Klipper3d/klipper #121: Klipper fails to open port /dev/rpmsg_pru30

sapell, 2018-01-12:
Due to the limited size of the BeagleBone Black eMMC (4G) I have chosen to start from scratch with Klipper basing on stretch-console (https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Releases). I managed to install Klipper by installing first build-essential and python-dev packages. I ran the make flash as per the guide for BeagleBone, however, I have not installed Octoprint yet. The klippy.log has error entries about not being able to open /dev/rpmsg_pru30, apparently communication issue with the klipper_pru.

Might be that I have some dependency missing or something fairly simple as I am basing on a subset of the recommended OS image in the installation guide.

Would you be able to help me with this?

Here's the [klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1627542/klippy.log.gz)

_hostnamectl tells me this:_
```
  Operating System: Debian GNU/Linux 9 (stretch)
            Kernel: Linux 4.9.74-ti-r90
      Architecture: arm
```
_klipper.service status:_
```
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated; vendor preset: enabled)
   Active: active (running) since Fri 2018-01-12 21:02:59 UTC; 29min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 366 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/klipper.service
           └─401 /home/debian/klippy-env/bin/python /home/debian/klipper/klippy/klippy.py /home/debian/printer.cfg -l /tmp/klippy.log

Jan 12 21:02:58 beaglebone systemd[1]: Starting LSB: Klipper daemon...
Jan 12 21:02:59 beaglebone klipper[366]: Starting klipper: klipper.
Jan 12 21:02:59 beaglebone systemd[1]: Started LSB: Klipper daemon.
```
_klipper_pru.service status:_
```
● klipper_pru.service - LSB: Klipper_PRU daemon
   Loaded: loaded (/etc/init.d/klipper_pru; generated; vendor preset: enabled)
   Active: active (running) since Fri 2018-01-12 21:03:00 UTC; 26min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 368 ExecStart=/etc/init.d/klipper_pru start (code=exited, status=0/SUCCESS)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/klipper_pru.service
           └─433 /usr/local/bin/klipper_mcu -w -r

Jan 12 21:02:58 beaglebone systemd[1]: Starting LSB: Klipper_PRU daemon...
Jan 12 21:02:59 beaglebone klipper_pru[368]: /etc/init.d/klipper_pru: 47: /etc/init.d/klipper_pru: cannot create /sys/bus/platform/drivers/pru-rproc/unbind: Directory nonexistent
Jan 12 21:02:59 beaglebone klipper_pru[368]: /etc/init.d/klipper_pru: 48: /etc/init.d/klipper_pru: cannot create /sys/bus/platform/drivers/pru-rproc/unbind: Directory nonexistent
Jan 12 21:03:00 beaglebone klipper_pru[368]: Starting pru:/etc/init.d/klipper_pru: 53: /etc/init.d/klipper_pru: cannot create /sys/bus/platform/drivers/pru-rproc/bind: Directory nonexistent
Jan 12 21:03:00 beaglebone klipper_pru[368]: /etc/init.d/klipper_pru: 54: /etc/init.d/klipper_pru: cannot create /sys/bus/platform/drivers/pru-rproc/bind: Directory nonexistent
Jan 12 21:03:00 beaglebone klipper_pru[368]: Loading ADC module:/etc/init.d/klipper_pru: 57: /etc/init.d/klipper_pru: cannot create /sys/devices/platform/bone_capemgr/slots: Permission denied
Jan 12 21:03:00 beaglebone klipper_pru[368]: Starting klipper MCU: klipper_pru.
Jan 12 21:03:00 beaglebone systemd[1]: Started LSB: Klipper_PRU daemon.
```

BlackStump, 2018-01-13:
when I installed Debian Stretch I had to edit the uEnv.txt because of the changes to U-Boot_Overlays
It does not see the Replicape otherwise.
Here is mine got no idea if it is correct but it works for me.

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.89-ti-r130
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
#enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

sapell, 2018-01-13:
Thanks for the tip BlackStump! I'll check my uEnv. It's just that I'd need HDMI enabled.

sapell, 2018-01-14:
jessie-console did not work either, same error. So I guess I'll go back to the found working OS image 9.2 stretch-iot and live with the bloat. I'll put an SD-card for print files and map OctoPrint print files accordingly. Closing the issue as it's not really a Klipper issue.

closed 2018-01-14 by sapell

jakep82, 2018-01-29:
This information references an older kernel, but perhaps it's still useful.  I would also like to get stretch-console working to remove some of the bloat.

http://codegists.com/snippet/shell/beaglebone-44-pru-rprocsh_alexanderhiam_shell

jakep82, 2018-02-13:
I got Klipper and Octoprint running on Jessie console last night.  I ran out of time to verify everything, but it appeared everything was working.  I'll copy over my printer.cfg file tonight and do a test print.  After installing Klipper, Octoprint, and all the stuff necessary to get them working I had 1.8GB of free space on the eMMC.  Other than build-essential, python-dev and a few other packages for wifi, I also installed ti-pru-cgt-installer and everything worked fine during installation of Klipper.  I do still have the ADC warning on klipper_pru, but I also have that warning on a known working install of IoT.  The slots file is deprecated, but I verified that ADC is working on both the IoT and console builds.

sapell, 2018-02-13:
Thanks jakep82 for letting us know. I've got my setup working acceptably by using a spare SDcard and dedicating that for the print files. Also I decided to ditch the Replicape at least for now, as the TMC2100 stepper drivers on it clearly lack current driving capability to drive my direct drive extruder. At full current the drivers overheat regardless of active cooling, at lower currents there's not enough torque, skipping steps in either case. Decided to keep BeagleBone though, just using Arduino/RAMPS with it.

jakep82, 2018-02-13:
That's too bad.  I'm running a fairly large 59Ncm stepper on a Bondtech BMG extruder in a bowden setup, so I have no issues running my extruder stepper in spreadcycle at 0.6A.  I haven't tried running a small pancake stepper with the Replicape which is what I'm guessing you're using.  Anyways, if you want to take advantage of the eMMC on the BBB it is possible with Jessie console.  I'm guessing by the time I'm done setting up my webcam and a few other things I'll have around 1.5GB of free space.  I might also try installing Stretch console tonight just to see if I can get that working.

sapell, 2018-02-13:
Actually I have an ACT motor Nema17 motor, would have to check the torque of that, but definately not a pan cake afaik. Also have a bowden and a mk7 extruder gear. I have the webcam on a separate RasPi as I had issues with it running on the BBB concurrently with Klipper.

jakep82, 2018-02-13:
Hmm, I've not had that problem.  I've been running a Logitech C270 at 720p with no issues on my Jessie IoT installation (haven't tested webcam on console yet).  I will say I had a lot of issues even without the webcam before Kevin implemented the virtual SD feature.  Now that I'm using that I'm seeing about 70% idle while printing and recording a timelapse at 720p, and I've had zero buffer issues even on prints with a lot of short segments.

sapell, 2018-02-13:
Ok, good to know. My experience was prior to the virtual sd feature, I have Logitech C525 at 720p an 15fps. Must try the virtual sd, have not updated and configured it yet.

jakep82, 2018-02-13:
I highly recommend it.  I had a lot of host buffer issues when printing round holes and bosses.  With virtual SD those issues have all disappeared.  You do lose a couple non-essential features in Octoprint, but it's well worth the trade off.

sapell, 2018-02-13:
Since this went OT already, might as well continue, closed issue anyway. I have 48Ncm holding torque (ACT Motor 17HS5425), I had the current set at 0.8A and spreadcycle 16. With the RAMPS I have the DRV8825 drivers, extruder set at ~2A current, no issues with overheating or lack of power. When I had the tower motors (have a delta) set at 0.6A with the Replicape, I noticed that when returning the effector to the center from the edge of the print area (200mm diameter), the motor(s) lost steps causing the nozzle to not move in plane and hit the bed (not during print luckily). Therefore I increased their currents to 0.8A and had no issues after with the movement. Based on my previous working settings with the RAMPS, I didn't think the extruder motor could work reliably with lower or equal current as the movement motors. Should probably look in to getting a stronger (longer) nema 17 for the extruder if I'm to continue experimenting with the Replicape. Off the record, for the price tag the Replicape is not worth it imho. Especially their official SW/FW is poorly documented (scattered with obsolete and confusing info), neither did they help me get Redeem properly working. Klipper redeemed (pun intended) the Replicape in my case. We'll see maybe it'll get a second shot some day.

jakep82, 2018-02-13:
I 100% agree. I bought my Replicape almost a year ago, and I completely gave up on running it with Redeem.  I had all kinds of trouble getting it running initially, and to this day Redeem still has some major issues.  I recently went back and got Redeem working on the develop branch, but it still has some major issues.  The documentation is also incomprehensible at this point, and they're now in the process of making the configuration even more complicated.  I've completely lost interest in following it any more, especially now that I have Klipper running so well.

I am however curious to see what happens with their new Revolve board.  If you haven't seen it, it's basically a Replicape and BBB (Octavo SiP) combined on a single board with 6 TMC2130 stepper drivers.  They're aiming for a $99 retail price which would make it compelling as long as I can run Klipper on it.