Klipper3d/klipper #125: Support Repetier Server for RaspberryPi

amaximchuk, 2018-01-16:
I installed the Repetier Server next to Octopi, but I can't connect it to the Klipper. I went through all possible serial ports from list (like this - /dev/serial/by-path/platform-3f980000.usb-usb-0:1.5:1.0-port0, /dev/serial/by-path/platform-3f980000.usb-usb-0:1.5:1.0-port0, /dev/ttyUSB0), the server tries to connect but it fails. 

I tried to use the TouchUI interface for Octopi - it's a real hell compared to the Repeater Server Touch.

KevinOConnor, 2018-01-16:
On Tue, Jan 16, 2018 at 12:04:02AM +0000, amaximchuk wrote:
> I installed the Repetier Server next to Octopi, but I can't connect it to the Klipper. I went through all possible serial ports from list (like this - [...]

In order to replace OctoPrint, you'll need to convince the Repetier
Server software to connect to /tmp/printer.  (Only the Klipper host
software talks directly with the micro-controller - Repetier Server
needs to talk directly with the Klipper host software via
/tmp/printer.)

More generally though, if you want to explore an advanced topic like
this, then you'll need to figure out all the steps necessary to make
it work.

-Kevin

amaximchuk, 2018-01-16:
Sorry for the noob question, where I can find more information about /tmp/printer port?

amaximchuk, 2018-01-17:
I just received answer from Repetier Server developers about connection to Klipper:
"Repetier-Server connects to the device given in config. It will not list it but if it can be controlled like a serial port it should work if you put the path to it manually. So first question is, if it is like a serial port or more like a socket. A socket requires other functions to talk as it has no baud rate etc.

Second thing is selecting the right firmware. I think Marlin would be the correct start to get it going. Also you can copy the firmware description /usr/local/Repetier-Servers/firmwares to adjust to special answer format. But most important is to get a general communication running and repetier-firmware uses a binary format that klipper surely does not speak. "

If you don't mind - could you comment?

amaximchuk, 2018-01-17:
Kevin if you will have a little time may you answer for some questions from Repetier Server developer?
[https://forum.repetier.com/discussion/4795/connect-repetier-server-to-klipper-firmware](url)

P.S. And I want to thank you for implementing such a cool idea! I have a completely different quality and speed of printing with the same hardware!

TunaLatte, 2018-01-26:
Using tty0tty as "tmash" pointed out works, but it requires kernel module.

amaximchuk, 2018-01-28:
Yes. I installed the tty0tty. It works. Thanks!

closed 2018-01-29 by amaximchuk