Klipper3d/klipper #127: Aborting a just-started print.

rewolff, 2018-01-16:
Occasionally (two or three times in the last two days), I decide I've made a mistake in slicing while the printer is still warming up. Hit cancel in octoprint, go back to slic3r and upload again. This works... except that now I get "error: Must home first" from KLIPPER. This is difficult to recover from. Octoprint disconnects and the "connect" does not immediately work. Manually issuing a home from CONTROL or TERMINAL tab works. 

You'll probably say that I have something in my start-gcode that triggers this. You're probably right. My main gripe is that this causes the printer to end up "heated up", but not connected to octoprint. 

Is it possible to programmatically find out if Octoprint is connected to Klipper? If yes, how about a timeout. If after 60 seconds no connection has been made, stand down and cool things down?

Is it octoprint that issues the "what's the temp?" GCODE commands? If yes, than that's what you can detect.

KenshiHH, 2018-01-16:
Look at the FAQ
[https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#why-cant-i-move-the-stepper-before-homing-the-printer](url)

rewolff, 2018-01-16:
I know that. But my start GCODE does not move the head before homing. Still, aborting and restarting quickly causes this error. 

OK. I have (bad memory for the actual gcodes...)  
"heat up the bed and wait". 
"start heating the hotend"
"home"
"move to X0 Y0 Z100"
"heat the hot end and wait". 

Could it be that octoprint has already sent the "move X0 ... " gcode when it aborts and issues the "disable motors" somehow managing to get that into the queue _before_ the MOVE command? 
The thing is: I'm pretty sure I hit the "cancel" before the bed was at temp. 

Ideally, I would suggest that hitting cancel would be able to even abort the "heat the bed and wait" GCODE command, right?

KenshiHH, 2018-01-16:
you can set a custom gcode script in octoprint for aborting a print

KevinOConnor, 2018-01-16:
On Tue, Jan 16, 2018 at 11:27:28AM -0800, Roger Wolff wrote:
> Occasionally (two or three times in the last two days), I decide I've made a mistake in slicing while the printer is still warming up. Hit cancel in octoprint, go back to slic3r and upload again. [...]

If you're having an issue with Klipper, please provide the klippy.log
file as described at:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

geodave810, 2018-01-18:
This is the same problem I ran into last month & Kevin pointed out the setting in Octoprint I needed to change.  Under Octoprint Settings, look at GCODE Scripts section and look in the portion "After print job is cancelled" and see if you have:
; disable motors
M84
Comment that out & see if that fixes your problem.

rewolff, 2018-01-18:
I am pretty sure that I have that there. And in 90% of the cases, I abort a print because things are going wrong (That used to happen a lot, but a small tweak now has my printer printing neat things with high reliability!). Then the "disable motors" is entirely reasonable. So I don't need to do the experiment that removing this "disable motors" will leave the motors enabled and allow a quick restart of the print. 

But the logical course-of-events is that:
* start print
* cancel print
* disable motors
* start new print
* home 
* first movment. 

But somehow octoprint and klipper are conspiring in this case to trigger a movement just after the motors are disabled. Is it possible in G-code to issue GCODE commands with "do it now" while still waiting for a previous one? In that case, aborting during the heat-up (even without starting the second print) would be triggering this: Start heating the bed, start heating the hotend home, move. wait-for hotend to finish heating All that is queued, and then the "do it now" version of M84, disable motors is issued. This would then finish heating the hotend and then process the move that was still queued....

KevinOConnor, 2018-01-21:
Closing this pending a log with M112 that shows the issue.

closed 2018-01-21 by KevinOConnor