Klipper3d/klipper #128: Servo support for BLTouch

RickMcConney, 2018-01-16:
Has anyone figured out a way to get a BLTouch probe working as the Z end stop. I can home X and Y then move to the center of the bed but I need to deploy the probe before I can home Z. If the Servo Gcode (M280) was supported I think I could then deploy the probe and have it act as a normal end stop.

KevinOConnor, 2018-01-17:
On Tue, Jan 16, 2018 at 02:19:55PM -0800, RickMcConney wrote:
> Has anyone figured out a way to get a BLTouch probe working as the Z end stop. I can home X and Y then move to the center of the bed but I need to deploy the probe before I can home Z. If the Servo [...]

This looks similar to issue #46.  See the config/example-extras.cfg
file for information on manipulating servos.

-Kevin

RickMcConney, 2018-01-17:
Thanks. I see from issue #46 that there is another issue after using the servo to deploy the probe. The BLtouch will stay triggered after it is hit and a backoff is performed. Which seems to cause the home to fail with the error in #46. Is there a work around for this.

KevinOConnor, 2018-01-21:
Closing this in favor of the existing issue #46.

closed 2018-01-21 by KevinOConnor