Klipper3d/klipper #129: Recv: !! Endstop x still triggered after retract -- Even after unplugging the endstop?

recuso, 2018-01-17:
I had been waiting for a part but got back to my klippy-running maker select v2 (melzi board). It had previously been running well, but now when I try to print or home the X/y, it "jumps" a tiny bit away from the end and says:

Here is my klippy.log:

Start printer at Wed Jan 17 07:56:53 2018 (1516175813.9 876.4)
===== Config file =====
[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .0125
endstop_pin = ^!PC2
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .0125
endstop_pin = ^!PC3
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = !PB2
enable_pin = !PA5
step_distance = 0.0025
endstop_pin = ^!PC4
position_endstop = 0.5
position_max = 200

[extruder]
step_pin = PB1
dir_pin = !PB0
enable_pin = !PD6
step_distance = .010
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PA7
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = PD4
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PA6
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = PB4

[mcu]
serial = /dev/ttyUSB0
baud = 115200

[printer]
kinematics = cartesian
max_velocity = 300
[... 45 lines cut ...]
Stats 911.1: gcodein=94 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000061 bytes_write=2029 by                                                                                                             tes_read=6326 bytes_retransmit=0 bytes_invalid=0 send_seq=164 receive_seq=164 re                                                                                                             transmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994122
Stats 912.1: gcodein=94 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000059 bytes_write=2035 by                                                                                                             tes_read=6440 bytes_retransmit=0 bytes_invalid=0 send_seq=165 receive_seq=165 re                                                                                                             transmit_seq=0 srtt=0.009 rttvar=0.005 rto=0.028 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994125
Stats 913.1: gcodein=94 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000059 bytes_write=2041 by                                                                                                             tes_read=6569 bytes_retransmit=0 bytes_invalid=0 send_seq=166 receive_seq=166 re                                                                                                             transmit_seq=0 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994126
Stats 914.1: gcodein=94 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000059 bytes_write=2047 by                                                                                                             tes_read=6670 bytes_retransmit=0 bytes_invalid=0 send_seq=167 receive_seq=167 re                                                                                                             transmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994128
Stats 915.1: gcodein=94 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000059 bytes_write=2053 by                                                                                                             tes_read=6771 bytes_retransmit=0 bytes_invalid=0 send_seq=168 receive_seq=168 re                                                                                                             transmit_seq=0 srtt=0.009 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994130
Stats 916.1: gcodein=99 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000059 bytes_write=2059 by                                                                                                             tes_read=6900 bytes_retransmit=0 bytes_invalid=0 send_seq=169 receive_seq=169 re                                                                                                             transmit_seq=0 srtt=0.009 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994130
Stats 917.1: gcodein=99 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=2065 by                                                                                                             tes_read=7014 bytes_retransmit=0 bytes_invalid=0 send_seq=170 receive_seq=170 re                                                                                                             transmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994129
Stats 918.1: gcodein=99 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=2071 by                                                                                                             tes_read=7115 bytes_retransmit=0 bytes_invalid=0 send_seq=171 receive_seq=171 re                                                                                                             transmit_seq=0 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994129
Stats 919.1: gcodein=99 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000070 mcu_task_stddev=0.000059 bytes_write=2077 by                                                                                                             tes_read=7244 bytes_retransmit=0 bytes_invalid=0 send_seq=172 receive_seq=172 re                                                                                                             transmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 f                                                                                                             req=15994128
Stats 920.1: gcodein=99 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: mc                                                                                                             u_awake=0.006 mcu_task_avg=0.000 [... cut ...]
Stats 935.1: gcodein=114 print_time=8.930 buffer_time=0.000 print_stall=0 mcu: m                                                                                                             cu_awake=0.006 mcu_task_avg=0.000071 mcu_task_stddev=0.000060 bytes_write=2173 b                                                                                                             ytes_read=9039 bytes_retransmit=0 bytes_invalid=0 send_seq=188 receive_seq=188 r                                                                                                             etransmit_seq=0 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0

BlackStump, 2018-01-17:
I would check wiring! As a test remove your X endstop wires from the board end, you eluded that it was working but was waiting on part , was the part anyway related to the X endstop or involved messing with the wiring ?

recuso, 2018-01-17:
Ah, I've unplugged both from the board itself and I still get: Recv: !! Endstop x still triggered after retract

The part wasn't related to the endstop itself but it was swapping out my extruder end. So yes, there was wiring on the extruder side. I didn't touch the endstop microswitch wire since the mounting wasn't changed.

I'll try to grab my big ole fluke 8025b from the garage, thank you for the thoughts/direction. I'm not sure how I might be able to check the board itself, though, now that I know the switches aren't at "fault" (as I said, it still says X triggered after unplugging the wires from the board itself).

cmock, 2018-01-17:
I would first try to exclude a hardware problem -- flash your previous firmware and check whether the endstops (still unconnected) show up as open or triggered with M119. If they show up as triggered, you can be pretty sure you have a hardware problem.

If not, it's a software problem...

recuso, 2018-01-17:
Curious! Well, BOTH X and Y are interpreted as always triggered right now. I was able to home it by using the "Z" microstop pin for all 3 and manually triggering it -- just to see what was up.

Cmock: Thank you, that's a good suggestion and kind of where my mind was going. Problem: I don't have the original firmware. I looked around online and have found a couple of files, but nothing that purports to be the original maker select firmware for sure. If I flashed something compatible or similar (say, for the wanhao di3 which is the same machine) but the pins are different I'm kind of back to square one on figuring out what pins are what -- with two pins already potentially broken. Yuck.

It seems like my only option might be to just flash some sort of arduino program onto it and see if I can read the inputs on those pins? I'm sorry that my particular understanding in this respect is a bit limited, but it seems like that'd be a safe-ish route to go. After confirming it's good or bad, re-flash klippy.

I'm using a Melzi (hybrid 1.0) type board  -- maker select v2 http://reprap.org/wiki/Melzi

Time to do a bit of digging it seems

cmock, 2018-01-17:
Right, arduino would be the way to go in this case. Start with the "DigitalInputPullup" example and change the port number. As the endstops are inverted, it should report "1" for a disconnected or untriggered switch, and "0" for a triggered switch.

closed 2018-01-18 by recuso