Klipper3d/klipper #132: Use Socat to link /tmp/printer to a virtual serial port

andrea7376, 2018-01-20:
Hi,
I'm not very familiar with linux but I would like test klipper with other software that support only real serial port.
Is it possible to use Socat to link /tmp/printer to a virtual serial port ?
How can I do?
Thank you.
Andrea

andrea7376, 2018-01-20:
I tested something like that: sudo socat -d -d pty,/tmp/printer,raw,echo=0 pty,link=/dev/ttyS0,raw,echo=0
but doesn't seem to work.

cmock, 2018-01-20:
How is your software determining what a "real serial port" is? Because /tmp/printer is a symlkink to a pseudoterminal (/dev/pts/N), so AFAIK it should accept all relevant ioctls.

If it is about filenames, just symlink /dev/pts/N to whatever path your software is happy with.

KevinOConnor, 2018-01-21:
As noted - Klipper already produces a "virtual serial port" and /tmp/printer is a link to that port.  If you're having a problem with some software communicating with Klipper, please open an issue describing that problem (along with the klippy.log file for it).

closed 2018-01-21 by KevinOConnor