Klipper3d/klipper #134: Color Mixing Config

RobMink, 2018-01-21:
I am having trouble configuring Kilpper for the diamond 3-1 print head.  After an update, it is complaining that I am using the same analog pin for all three extruders.  If I remove the offending line it says I have no sensor.  
[klippy.log](https://github.com/KevinOConnor/klipper/files/1649569/klippy.log)

[printerwith3extruders.cfg.txt](https://github.com/KevinOConnor/klipper/files/1649570/printerwith3extruders.cfg.txt)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1649571/printer.cfg.txt)

I am sure I am just doing something wrong.  Thanks for reading this.

RobMink, 2018-01-21:
Wow, you are awsome!  Thanks!

RobMink, 2018-01-22:
Tested - Fixed

KevinOConnor, 2018-01-22:
Great!  I committed this to the master branch (commit 09eec37).

closed 2018-01-22 by KevinOConnor