Klipper3d/klipper #135: How to get Z-probe status?

ronindev, 2018-01-21:
Hello! 
I'm trying to config my delta, but I can't figure out how to get status of Z-min (Z-probe) pin. M119 give only a,b,c endstops statuses

KevinOConnor, 2018-01-21:
Closing this in favor of the existing issue #19.

closed 2018-01-21 by KevinOConnor

ronindev, 2018-01-21:
I do not think It's related to #19 because M119 should print Z-min status, like in marlin.

KevinOConnor, 2018-01-22:
On Sun, Jan 21, 2018 at 05:23:15PM +0000, Igor Shuvalov wrote:
> I do not think It's related to #19 because M119 should print Z-min status, like in marlin.

Sorry, I misunderstood your issue.  The work-probe-20170609 branch has
a QUERY_PROBE command that reports the status of the probe - it should
be used instead of M119/QUERY_ENDSTOPS.

-Kevin

dodasch, 2021-07-09:
what if my Probe and connection is okay (LED on Probe works) but the QUERY_PROBE command is always showing "OPEN" ?