Klipper3d/klipper #138: Question: Any way to divorce Klipper from Octo*

kjcole, 2018-01-23:
HacDC [the second oldest hackerspace in the US] has a non-microprocessor server serving several 3D printers via pronsole (the CLI-only portion of Printrun / Pronterface). No need for Cura, and the other baggage that appears to come with Octo*.

Any guidance on setting up Klipper sans OctoPrint / OctoPi?  Or are they joined at the hip?

Thanx.

hg42, 2018-01-23:
I currently use pronterface on my PC with klippy.py also running on the same PC.
pronterface is set to use `/tmp/printer` as the serial, I guess pronsole will work the same.
klippy.py is just like a normal firmware communicating over a serial called `/tmp/printer`.
Most host software will not detect `/tmp/printer`. If you cannot set it manually (or add it like in octoprint) then you can link `/tmp/printer` to `/dev/ttyANYNAME` or wherever it is found by the host software.

However, I'm not sure, what exactly "with non-microprocessor server serving several 3D printers via pronsole" means.
Where is pronsole running?
Where should klippy.py run?
From your description I assume pronsole is running on the server (if it has python?) then it probably uses a serial cable to connect to the printer. In this case you would need to use a program to establish a bidirectional connection on the PI between the serial to the server `/dev/ttyXXX` and `/tmp/printer`.
I found an example with socat, something like:
```
sudo socat -d -d pty,link=/tmp/printer,raw,echo=0 pty,link=/dev/ttyXXX,raw,echo=0
```
(untested) if that works with a pty, the original example connected two real serial devices.

Theroretically klippy.py could also run on the server (depending on how it behaves).
Or eventually klippy could be changed to use a real serial device instead, I guess you would have to change this line in klippy.py:
```
        input_fd = util.create_pty(options.inputtty)
```
The name `/tmp/printer` can already be changed on the klippy command line, but from what I see in the code it is always used to create a pseudo terminal device.
Instead it could probably use a device directly, if it already exists.

kjcole, 2018-01-23:
I guess I'm displaying my ignorance: We have a computer connected to two Lulzbots and a Rostock Max. I've never dug into what processors drive those beasties. I was thinking that Klipper would massage g-code and feed optimized g-code to Pronsole. It sounds like I have it backwards / sidewards and must go Pronsole --> Kiipper --> printers?

hg42, 2018-01-23:
I remember another issue where socat was mentioned -- mom -- found it: #132
@andrea7376 used a similar command line:
```
sudo socat -d -d pty,/tmp/printer,raw,echo=0 pty,link=/dev/ttyS0,raw,echo=0
```
and said it doesn't work. But one link= was missing.
But I think it's another problem, the concept was wrong.
If the sequence (for one direction = sending a byte from host to klippy) would have been
```
host software -> /dev/ttyS0 --cable--> /dev/ttyS1 --socat--> /tmp/printer -> klippy
```
it might have worked, but
```
host software -> /dev/ttyS0 -> empty output pin
                                                empty input pin -> /dev/ttyS0 --socat--> /tmp/printer -> klippy
```
cannot work.

In case of the server this would be:
```
host software on server --cable--> /dev/ttyS0 --socat--> /tmp/printer -> klippy
```
So if socat can connect an existing(!) pty with a real serial device, this should do the job.

socat has many options, so I think it can be done, if mastering socat.

But the easier way would be changing klippy.py to simply use the existing tty device.

hg42, 2018-01-23:
> We have a computer connected to two Lulzbots and a Rostock Max

ok, then it's probably a normal PC, right? or a bigger machine?

> I was thinking that Klipper would massage g-code and feed optimized g-code to Pronsole

no, the klippy/klipper pair is essential the same like Marlin etc.
So klippy gets gcode from the host software (pronsole, cura, etc.) and calculates the step times and other events like pwm for heaters and fans and sends them to the MCUs (e.g. Mega/RAMPS), which do the final IO.

The difference for your case is that the "serial" port of klippy is not a physical port like /dev/ttyS0 but a virtual port (a pseudo terminal device, /dev/pty/X which is linked to /tmp/printer).
So there's no connector for a cable.

> It sounds like I have it backwards / sidewards and must go
> Pronsole --> Kiipper --> printers?

the full sequence would be:
`pronsole --> klippy --> klipper --> steppers etc.`

with octoprint it would be like this instead:
`web-browser --> octoprint --> klippy --> klipper --> steppers etc.`

The web browser can be on the PC or on a smart phone etc.
octoprint usually runs on the PI and is connected via the virtual port to klippy which runs on the same PI. The low level klipper runs on the MCUs.

pronsole replaces browser+octoprint.

pronsole can run on the PC and klippy can also run on the PC (one klippy for each printer).
Or klippy runs on a PI and the communication from pronsole must be routed to `/tmp/printer` somehow.
That would be the job of socat.

hg42, 2018-01-23:
I did some tests with socat, but I think I am missing something. Way too many options...and probably too late (03:53)

hg42, 2018-01-23:
ok, a last (very simple) test seems to do something I would expect:
```
socat -d -d -d -d 'open:/tmp/printer,raw,echo=0' 'open:/dev/ttyS0,raw,echo=0'
```
so both "devices" are simply opened, which sounds reasonable.

Unfortunately I have only one serial on my PC, so in this configuration I cannot test, if it really works.

kjcole, 2018-01-23:
Thanks for all the help. I'll digest that for a bit and see how far I get. 

(There may need to be instructions in the docs/Installation.md for folks who won't be running scripts/install-octopi.sh or scripts/installl-beaglebone.sh and/or possibly another script.)

hg42, 2018-01-23:
JFYI:
I just tried to create a virtual serial port `/tmp/virtual` connected by socat to `/tmp/printer` which works like a charm with pronterface connected to `/tmp/virtual`. So I am quite sure it will also work with a real serial using open:/dev/ttyS0....
```
socat -d -d -d -d 'open:/tmp/printer,raw,echo=0' 'pty,link=/tmp/virtual,raw,echo=0'
```
Some other quick tests failed because of permission problems, e.g. I have no clue why sudo socat cannot access `/tmp/printer`...

hg42, 2018-01-23:
> There may need to be instructions in the docs/Installation.md for folks who won't be running scripts/install-octopi.sh

yes...but you can always look at the scripts.
I also use install-octopi.sh on my linux machine. This seems to be the same procedure.

It basically
* installs apt packages for compiling klipper and python
* creates a python virtual environment to run klippy inside
* installs an init script for klipper (shouldn't this be called klippy?)
* installs settings for klippy in /etc/defaults
* starts klippy

this is all very dependent on debian linux, so no chance for now to install it on other machines.
Though there are similar mechanisms on other unix systems and may be even on macs and windows systems via cygwin etc.

KevinOConnor, 2018-01-23:
On Tue, Jan 23, 2018 at 12:02:46PM -0800, Harald wrote:
> I just tried to create a virtual serial port `/tmp/virtual` connected by socat to `/tmp/printer` which works like a charm with pronterface connected to `/tmp/virtual`. So I am quite sure it will also work with a real serial using open:/dev/ttyS0....

FYI, the socat utility does the exact same thing that Klipper does -
it creates a pseudo-tty device (ie, a pty).  So, there is no gain in
running socat to create a pty device hanging off of /tmp/printer.

That said, one could use socat to connect the Klipper host software to
an instance of OctoPrint (or similar) running on a different desktop
machine.  However, if one were to do that, I'd guess one would want to
use a tcpip channel between the two computers instead of an actual
serial port..

-Kevin

hg42, 2018-01-23:
@KevinOConnor yes, this command above does only create /tmp/virtual as a pty (this was a test, if communication through socat works at all), but I am quite optimistic, it can open a real serial device, too. See the other command line I posted above:
```
socat -d -d -d -d 'open:/tmp/printer,raw,echo=0' 'open:/dev/ttyS0,raw,echo=0'
```
I think this will do the job.

Unfortunately I have no equipment available to test this right now (cables, adapters, serial devices, ...).
I know it can open the serial device, but there is no answer, because nothing is connected.

At least it is able to open the existing /tmp/printer. So there is little difference to opening a real serial device (it's a file with the same properties).

However, my **preferred solution** would be to change the behavior, when klippy is started with something like
`--input-tty /dev/ttyS0`

Then klippy.py could detect, that the given device already exists and just open it instead of creating a pty and link to that name. This would be intuitive and simplify such setups very much.

KevinOConnor, 2018-01-23:
On Tue, Jan 23, 2018 at 01:39:52PM -0800, Harald wrote:
> However, my **preferred solution** would be to change the behavior, when klippy is started with something like
> `--input-tty /dev/ttyS0`

I wouldn't be comfortable taking input g-code commands from a real
serial port - they're notoriously susceptible to line noise and there
is no error checking on the g-code input.  That said, using a socket
as input could certainly be done.

-Kevin

hg42, 2018-01-23:
I think, it's not Klipper's business to decide if the connection is bad.
Such a file can be anything...
If Klipper can open a file, it can be connected to many more things.

I also think, serial lines being noisy is not the whole truth. It's more because todays serial devices don't use proper voltages and such. In former times they were used for many things and simply worked.
But you are right about the missing error checking. However while printing you have a sequence number and crc. Some hosts can use this all the time. It's the user's responsibility.
You may add a comment in the docs.

hg42, 2018-01-23:
Also, in those rare cases like that one from this issue, they already use the serial.
So it is known to work. People using this, usually know what they do.
Another non-critical variant is a serial USB cable with the adapter on klippy's side.
If you think this is a problem, then you cannot use a serial on the MCUs either.

closed 2018-02-02 by KevinOConnor