Klipper3d/klipper #139: Anet A8 Issue

JimJamUrCode, 2018-01-24:
I am having a little problem here. I have tried following the steps to make this work multiple times, but run into the same problem everytime.

I have attached the output from my avrdude flash. It seems to have flashed properly. I rebooted the avr board, and the Anet LCD shows a bunch of white squares, I have read this is normal.

I reboot my Pi just to make sure everything is running good.

I make sure klipper service is running:

`sudo service klipper start`

I then go to my OctoPrint connection screen and select my /tmp/printer, set my baud rate to 115200, and then hit connect. OctoPrint acts like it connects, but when I go to the "terminal" tab and type "status" I get the following message:

> // The klippy host software is attempting to connect.  Please
> Recv: // retry in a few moments.
> Recv: !! Printer is not ready

No matter how long I wait, the same message is always returned.

When I initially press the connect button this gets spit out on the terminal:

> Connecting to: /tmp/printer
> Changing monitoring state from 'Offline' to 'Opening serial port'
> Connected to: Serial<id=0x6bdf36b0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
> Changing monitoring state from 'Opening serial port' to 'Connecting'
> Send: N0 M110 N0*125
> Connection closed, closing down monitor
> Recv: ok
> Changing monitoring state from 'Connecting' to 'Operational'
> Send: N0 M110 N0*125
> Recv: ok
> Send: N1 M115*39
> Recv: ok FIRMWARE_VERSION:v0.5.0-108-g1b3ef8a FIRMWARE_NAME:Klipper
> Send: N2 M21*18

I have tried building the firmware with 250000, 115200, and 57600 baud rates to no avail. Each time I recompile the firmware, I make sure to update my printer.cfg file with the new baud rate and when I attempt to connect, I make sure to select the new baud rate in OctoPrint (I just noticed in the documentation that this isn't really supposed to matter, but I did go back and try at the recommended default 250000).

I have attached my printer.cfg file, it is nearly identical to the config example for the anet a8, it just specifies a baud rate of 115200.

running ls -l /dev/serial/by-id gives me this output:

> lrwxrwxrwx 1 root root 13 Jan 24 04:39 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0

In my configuration file I made sure that the [mcu] "serial" setting was set to /dev/ttyUSB0

Finally, my Octoprint version is 1.3.6, I really hope that's not the issue :)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1658648/printer.cfg.txt)
[avrdude output.txt](https://github.com/KevinOConnor/klipper/files/1658646/avrdude.output.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/1658647/klippy.log)

cmock, 2018-01-24:
I see the following in the avrdude output:

```
avrdude: verifying ...

avrdude: verification error, first mismatch at byte 0x0002
         0x46 != 0x6f
avrdude: verification error; content mismatch
```

That means flashing the new firmware failed. 

How did you try to flash it? The example config (anet-a8-2017.cfg) says to use

`avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex`

mediumo, 2018-01-24:
Interestingly I experienced more or less the same yesterday. I checked out Kevins probe branch.
Everything worked fine, could built and flash the MCU and on the first try everything looked good.
Got correct log entries for communication in klippy.log (which I sadly deleted), and saw temperature readings etc. in OctoPi.

Then I changed the config with the servo information for my bltouch sensor, and restarted klipper.
After that, only: 
`Timeout on serial connect`

I reverted the config, built and flashed the MCU binary again, everythging went fine, but still: 
`Timeout on serial connect`

Also tried to see what is going on directly on the serial-port: I see some scrambled characters, which should be fine because it's the klipper serial protocol, so there is some correct looking transmission from the MCU....

I will try to flash Marlin back tonight and see if it works, after that I will try to flash the klipper-mcu binary again. Unless somebody has a different idea :-).

Regards
Olaf

JimJamUrCode, 2018-01-24:
@cmock
Good eye! I can't believe I didn't see that error message.

When I try

> avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex

I get:

> avrdude: ser_open(): can't open device "/dev/ttyUSB0": No such file or directory
> avrdude done.  Thank you.

My avrisp mkII programmer does not show up as a usb device under /dev.

When I use:

>  avrdude -v -p atmega1284p -c avrispmkII -b 57600 -D -U /home/pi/klipper/out/klipper.elf.hex

avrdude can at least find the programmer and attempts to program the device, but ultimately fails on verification as you picked up on.

So I just tried adding the -e command to avrdude to erase the chip before programming and avrdude passed through the verification process! I attached the avrdude log from that flash. I still have the same issue...

> avrdude -e -v -p atmega1284p -c avrispmkII -b 57600 -D -U flash:w:/home/pi/klipper/out/klipper.elf.hex:i

@mediumo - I may try pulling the probe branch down and see if that changes anything for me.

[avrdude output 2.txt](https://github.com/KevinOConnor/klipper/files/1660077/avrdude.output.2.txt)

KevinOConnor, 2018-01-24:
On Tue, Jan 23, 2018 at 09:15:16PM -0800, James Harding wrote:
> I am having a little problem here. I have tried following the steps to make this work multiple times, but run into the same problem everytime.

If I'm understanding your report correctly, you are using an external
programmer to flash your board.  That's not the best way to flash
Klipper.  Generally, the external flashing tool is used to put a
bootloader on the chip (in the case that a bootloader isn't already
installed), and then Klipper is flashed using the bootloader.

So, I'd recommend doing the following:

- compile the Klipper microcontroller code with a baud rate of 250000

- flash a bootloader using your external flashing device.  For
  example, see:
  http://www.instructables.com/id/Flashing-a-Bootloader-to-the-CR-10/

- then disconnect the external flasher and flash klipper using:
  make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

- update your printer.cfg file to use the same serial device (and use
  the default baud of 250000):
  [mcu]
  serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

-Kevin

mediumo, 2018-01-24:
OK, kind of embarrassing, but your comment of the default baud-rate was the right hint.
Since the CH340G USB-to-UART converter on the Anet boards can only support max. 115200, this has to be defined in the config. I can provide an updated config as pull.

Also the flashing via make doesn't work since the baud-rate of the original boot-loader is 57600.
So it should work with optiboot flashed, otherwise it hast do be flashed manually since the baud-rate for firmware and bootloader connection differs.

@KevinOConnor: Would it make sense to define different boards for anet/melzi clones or is this case to rare and they should be flashed with avrdude manually? Another option would be to provide an additional parameter in the config via menuconfig or cli-parameter  to provide an additional baud-rate for flashing the mcu.

KevinOConnor, 2018-01-24:
On Wed, Jan 24, 2018 at 11:06:28AM -0800, mediumo wrote:
> OK, kind of embarrassing, but your comment of the default baud-rate was the right hint.
> Since the CH340G USB-to-UART converter on the Anet boards can only support max. 115200, this has to be defined in the config. I can provide an updated config as pull.

This is not correct - there have been several reports of the Anet
board working well with a baud of 250000.  A baud rate of 250000 is
the recommended rate.

> Also the flashing via make doesn't work since the baud-rate of the original boot-loader is 57600.

With the default Anet board, one should follow the instructions at the
top of config/anet-a8-2017.cfg to flash the board.  If one has
externally flashed the optiboot bootloader to the board, then the
generic "make flash" instructions should work for the board.

-Kevin

mediumo, 2018-01-24:
Hi Kevin,

thank you for the feedback. I will try with 250000 and test if it is stable.

JimJamUrCode, 2018-01-25:
Hi guys,
Thanks for the information. I did read in one of these issues that people were trying to burn bootloaders on  to their boards so that they could easily update the boards at a later date. I did not know that it was required :(

I just burned the bootloader onto the ATMEGA1284p. What a pain in the neck. Figuring out the fuses was out of control, it seemed like a huge guessing game. Ultimately I installed Atmel Studio because it has a visual tool and verification tools to make sure you don't improperly set your fuses:

[image]

After getting the bootloader flashed, I was able to successfully use flash the Klipper firmware with this command:

> avrdude -p atmega1284p -c arduino -b 115200 -P /dev/ttyUSB0 -U out/klipper.elf.hex

and then connect with OctoPrint and get a 

> //Printer is ready

Wahoo!!!!

closed 2018-02-02 by KevinOConnor