Klipper3d/klipper #140: Allow G92 to 'home' an axis

davidmpye, 2018-01-25:
Hi!

I've got an issue/feature request to add please.

I like to home my Z axis myself rather than automatically.  Once I've got the Z height where I want it, I like to be able to run the gcode command G92 Z0 (or other position I feel like!)

Could Klipper allow G92 <axis> <location>  to mark that axis as 'homed' to that location rather than insist on a separate/additional home command please?

Thanks

David

KevinOConnor, 2018-01-25:
G92 works fine once the printer is homed.  Moving without homing first is similar to issue #111.

Closing this in favor of the existing issue #111.

closed 2018-01-25 by KevinOConnor

davidmpye, 2018-01-25:
Indeed but this actually seems like a neater solution to the other issue. 

If you want to move an axis before homing it, use G92 to set it to a known location, then you can move it.  Just a thought...

hg42, 2018-01-26:
I support this M92 idea...
The only reason why klipper doesn't want to move is because the position is unknown.
If the user knows the position and sets it, this should be sufficient.

darenschwenke, 2018-01-26:
G92 is not used to set the actual position though.  It sets an offset from the homed position.  
It is useful for tweaking things like tooling (nozzle height) and should be persistent, but this is not the correct place to tell the machine it is homed.

hg42, 2018-01-26:
Hmm, I don't think so, G92 says "here is X,Y,Z,E".
If no XYZE is given, it sets them to 0. Otherwise it sets only the given position(s) to the value(s).
E.g. it isregularly used to set E to 0 at start, which is similar to homing.

I sometimes move the toolhead to a certain position and use G92 to set it to whatever I want it to be (usually all 0).

There are definitely machines out there, that cannot home because they have no end switches.
Which command would you use for these?

Also, for many machines homing doesn't make sense, because you have to set home to a certain position, e.g. for laser engravers or CNC you would chose a corner of the part or wherever you want to place the origin, may be a cross mark or a adjustment hole etc.
The end switches (if any) are used for protection in these cases.

davidmpye, 2018-01-26:
I agree.  I use G92 to home my cnc mill - it specifies absolute position.
(And therefore the machine could then allow that axis to move as it knows a
definitive location.)

David

darenschwenke, 2018-01-26:
Well I can tell you that coming from a CNC background G92 does not set the position.  This is how you align fixtures to your machine without screwing up your homing.
G92 sets an offset from a known position (and is stored as such), not the position itself.  Without a known position, what are you setting an offset from?  Homing is and should be separate.
The reprap page definition is wrong.  It's the dumbed down version of what actually happens in any real Gcode interpreter.

G92.x resets the offsets, and they got that definition right.

darenschwenke, 2018-01-26:
Concrete example:  You have a printer where you can swap out the print head.  
You store the centering and height of each print head as G92 offsets.
Now you can change print heads and have your coordinate system adapt, without affecting your homing or the global position of the machine.

davidmpye, 2018-01-26:
Is there a gcode command which *does* do what I want then?

I want to tell the axis where the damn thing is, without relying on end
stops.

David

darenschwenke, 2018-01-26:
I would say it would be logical to just use G28.
If the machine has no homing routine specified. then don't wait for a homing switch trigger and set the home to 'here'.
If it does have a routine, then that runs instead.
G28 lets you specify an axis to home so you can do just one.
I have not looked to see if this does or will work.

**Then** if you still don't like where your coordinate system is at, you can tweak it via an offset with G92.

hg42, 2018-01-27:
but if you want to home manually, this is exactly the same.
You do not set an offset but the absolute postition, right?
An offset would be added to something.
But I still assume, that after using G92 the position is at the values you used in the command?

hg42, 2018-01-27:
from the NIST standrad document:
http://ws680.nist.gov/publication/get_pdf.cfm?pub_id=823374

"You can offset the current coordinate system using G92 or G92.3. This offset will then apply to allnine program coordinate systems. This offset may be cancelled with G92.1 or G92.2."

so it is like you said, but:

"To make the current point have the coordinates you want (without motion), program
G92 X...Y... Z... A... B... C...
, where the axis words contain the axis numbers you want"

so you give the absolute position.
Internally this is stored as offsets to the underlying coordinate system (determined by homing or fixed, e.g. using encoders). Because of this, the offsets can be cleared and you get back to the original system.
But you can use M92 as often as you want without resetting in between. That's not like adding an offset. And with many 3D-printer firmwares you simply don't have the reset commands.

As a conclusion, it is exactly what @davidmpye wants...dialing the tool to the desired position and setting the coordinates he wants to have at this point with G92.

darenschwenke, 2018-01-27:
Yes, if you can home the machine manually, or just tell the machine it is already homed, then you can use G92 to move the coordinate system to wherever you need.
But the point here is:
1) homing itself shouldn't be part of G92 as that could break other things
2) G92 should not affect the absolute position, only the relative one (as an offset).
One thing I can think of in this regard... is tripod kinematics.  If you moved your absolute position instead of applying an offset, your bed would suddenly not be flat.  Not that this supports tripod kinematics atm, but if I have my way it will eventually.

hg42, 2018-01-27:
@darenschwenke yes...kind of...

I am not an expert in machining, I come from developing measurement systems (vision/video/laser/mechanical).

That's my view:

In general you have a coordinate system that's related to the hardware ( = machine coordinate system, something like the towers for a delta, etc.) and you transform this into a logical system via a kinematics module which relates to the part ( --> user coordinate system).
Then you can overlay more systems as transformations of the logical system (for certain sections of the part, referenced by measurements etc.).

Homing works on the hardware system and may even calibrate the kinematics transformation if enough values are measured (auto bed leveling etc.).
G28 does some simple homeing. G29 does calibration.
Both set parameters of the transformation machine->logical.

You say G92 sets an offset. But it does not. It sets the absolute logical coordinates of a location.
In simple cases this may be stored as an offset. It **can** also be stored as an offset to the logical system. But this is not mandatory. It could also use the reverse transformation to set some of it's parameters.
It totally depends on the transformation, what this means according to the homing. Especially if homing is necessary at all. E.g. in the simple case of a cartesian machine G92 can replace homing.

And yes, I agree, in the general case G92 does not replace homing.
So, the kinematics module has to decide if G92 can replace homing or not.
In many cases like cartesian or corexy G92 is sufficient.
In complex cases it could even be dependent on the actual values.

So, let's delegate G92 to the kinematics module (set_position).