Klipper3d/klipper #141: SeeMeCNC Rostock Max V2 Delta Printer Config for Klipper <tested & working>

FEsmondeWhite, 2018-01-25:
I created and tested this configuration for the [SeeMeCNC Rostock Max V2 printer.](https://www.seemecnc.com/products/rostock-max-v2-desktop-3d-printer-factory-assembled). My printer has the RAMBo v1.3 controller.

[WorkingRostockMaxV2.zip](https://github.com/KevinOConnor/klipper/files/1664376/WorkingRostockMaxV2.zip)

This setup is for the stock SeeMeCNC Bowden extruder setup that came with versions 1 and 2 of the Rostock Max. It took me a while to find [the correct thermistor settings](https://www.seemecnc.com/products/thermistor-100k-ntc) (the hot end and heat bed both use the Semitec 104-GT2), and to figure out that the extruder barrel fan on this printer is plugged into the extruder 2 heater port by default, rather than into the fan 2 port.

I'll be upgrading my printer with the latest [HE280 print head](https://www.seemecnc.com/products/he280-hotend-with-accelerometer-probe), then adding [the 2-into-1 single extruder dual filament feed setup, similar to the cyclops print head](https://www.seemecnc.com/collections/parts-accessories/products/2-into-1-dual-filament-feed-adapter-kit). I'll attach updated configs for each of the updates as I complete and test them.

It's unfortunate that Klipper doesn't update the LCD interface. I may try to pull and adapt the LCD driving code from [the Rostock Max Repetier code.](https://github.com/seemecnc/Repetier-091-ROSTOCKMAX)

Thanks for the awesome project!
Francis

KevinOConnor, 2018-01-25:
On Thu, Jan 25, 2018 at 03:03:24PM +0000, FEsmondeWhite wrote:
> I created and tested this configuration for the [SeeMeCNC Rostock Max V2 printer.](https://www.seemecnc.com/products/rostock-max-v2-desktop-3d-printer-factory-assembled). My printer has the RAMBo [...]

Thanks.  What year (roughly) is this printer from?  I can add a sample
config file to the config/ directory for this printer based on your
config.

-Kevin

FEsmondeWhite, 2018-01-25:
It's roughly 2015. They're still selling, though the latest version has a
hot end with an accelerometer probe, and an improved bed.

Pretty soon I'll have my system upgraded with the latest hot end.

I'm not sure how to do a proper bed calibration... Matter control had a
great multipoint bed leveling offset correction. Is there a guide on this
for Klipper/octopi? Or do I need to do some kind of calibration through
cura?

Thanks again,
Francis

KevinOConnor, 2018-01-25:
On Thu, Jan 25, 2018 at 11:50:11AM -0800, FEsmondeWhite wrote:
> I'm not sure how to do a proper bed calibration... Matter control had a
> great multipoint bed leveling offset correction. Is there a guide on this
> for Klipper/octopi? Or do I need to do some kind of calibration through
> cura?

There is a development branch (work-probe-20170609) that implements a
DELTA_CALIBRATE command - see the config/example-delta.cfg on that
branch for details:

cd ~/klipper ; git fetch ; git checkout origin/work-probe-20170609 ; sudo service klipper restart

-Kevin

KevinOConnor, 2018-01-29:
FYI, I committed your sample config (commit 03bcae8a).

closed 2018-02-02 by KevinOConnor