Klipper3d/klipper #142: Jerk settings

wizhippo, 2018-01-26:
I run the same gcode on Marlin 1.1.8 and kilpper.  Both are set at the same acceleration, steps/mm etc.

The klipper result has bulged corners and blobs where the top surface meets the inner walls.  I have tried adjusting acceleration(reducing) and pressure advance(as documented and used all .05 steps to .20) and have not been able to reproduce as clean a part as Marlin,  I'm guessing this is due to Marlin's jerk control where I cannot find a similar setting.   Is there a similar jerk setting that I may try to correct this issue?

KevinOConnor, 2018-01-26:
On Fri, Jan 26, 2018 at 02:43:02PM +0000, Douglas Hammond wrote:
> I run the same gcode on Marlin 1.1.8 and kilpper.  Both are set at the same acceleration, steps/mm etc.
>
> The klipper result has bulged corners and blobs where the top surface meets the inner walls. [...]

In Klipper the cornering speed is controlled by the junction_deviation
setting.  Though, your description sounds more like the printer's e
step_distance may need to be calibrated.

-Kevin

wizhippo, 2018-01-26:
I've gone through the whole process described in Pressure_Advance.md Prerequisites and even when closer to under extruding I get the same artifacts.

So if my e step is correct and I am getting this then increasing the junction_deviation should have the nozzle spend less time at the corners and help reduce this correct?

KevinOConnor, 2018-01-26:
On Fri, Jan 26, 2018 at 03:43:59PM +0000, Douglas Hammond wrote:
> I've gone through the whole process described in Pressure_Advance.md Prerequisites and even when closer to under extruding I get the same artifacts.
>
> So if my e step is correct and I am getting this [...]

It's hard to say.  If the underlying problem is due to ringing, then
decreasing junction_deviation would be appropriate.  If the underlying
problem is oozing then increasing it might help.

Earlier you said that Marlin was performing better - it may help if
you post both your Klipper and Marlin configs.

-Kevin

hg42, 2018-01-26:
the bulged corners and the blobs at ends of infill are typical for low jerk settings.

The extruder still has pressure and with a slow decelerate and accelerate phase this pressure is released as a blob.

That's why Marlin simply switches direction at jerk speed. It is true, that this improves those blobs.
Junction deviation can do something similar, but has a more complex calculation (not necessarily better than jerk in my opinion).

As an alternative there are also algorithms that keep the pressure in the nozzle as constant as possible, which obviously also results in more jerk.

All these are especially useful for bowden extruders, because the time constant is increased with the length of the tube.

Non-null jerk and junctions deviation settings effectively increase acceleration. So if there is an acceleration limit and the steppers (or better the masses) are moved at this limit, the steppers will skip.

I think every printer has different conditions. For some the acceleration is the main problem, for others using jerk algorithm is the better. This may depend on the sturdiness of the printer, the moved masses, the extruder, the length between extruder and hotend, the length of the heated chamber in the hotend, the filament, so lots of parameters. I think there is no best way.

For my printers it's still undecided...
Klipper is set to a low acceleration of 1000 mm/s^2 but high speeds around 100-300mm/s, so the pressure has enough time to decrease. I also use a good teflon tube (but unicorn isn't necessary) and try to keep it as short as possible while also keeping a big bending radius.
So for my current configuration Klipper is significant slower for short infill lines but also significantly faster for long lines than Marlin and Smoothiware were able to do.

hg42, 2018-01-26:
note, you need to find a corresponding junction deviation setting to have a similar movement to the Marlin jerk. That's not easy to find and will not be exactly the same (Marlin drives a real sharp corner, while junction deviation leads to a curve, eventually an infinite junction deviation would be equivalent?).

The pressure problem is the subject of the linear advance algorithm and should solve the problem. But it also leads to more extruder movements and some (cheap or bad) extruders do not like that.

wizhippo, 2018-01-26:
@KevinOConnor Here are my configs https://gist.github.com/wizhippo/16cd1dbed1388d8c1e069304c3869219  The image at the bottom has the klipper print on the left.  You can see how the blobs along the letter Z are more pronounced and if you look closely the upper corners on the left are more rounded.  Sorry couldn't get a good shot of the sides to show the bulging better.

@hg42 I'll try a few prints changing the junction deviation setting. I have a Tronxy X5S, long bowden tube.  Just for fun I'll try printing faster too and see if that helps.  The test is done at 80mm/s.

The prints overall are not bad, but I would like to get the corners sharper with klipper.

hg42, 2018-01-26:
I have a X5S, too, but mine is a bit different. E.g. I use a geared extruder at the front, so the tube starts in the middle of the front, my heated bed is totally different, I have some modifications for sturdiness etc.
This looks like you are using my changes to Marlin I posted to the facebook group (BOARD_MELZI_TRONXY) before my board was damaged.

I originally thought you were talking of the outside corners, but those seem to be ok.
The two pictures look quite similar to me.
I guess the difference is mostly filament squeezed out of the surface?
Looks a bit like a slightly too high extrusion. Perhaps you give 0.01 mm step_distance a try (I originally had 99 steps/mm).
The path at the corners seems to be rounded in both. Could the temperature be too high and the filament dragging around the corner?
I found that my X5S is very sensitive to temperature calibration (and regulation) and speeds, also depending on the filament.
The original hotend cooling (if you still use it) has bad effects. If you want, we can discuss X5S specifics (print head cooling etc.) via mail (you find my email in my profile).

And yes, I understand you used the same material etc. in Marlin and Klipper.
I didn't compare every setting, but some comments according from the things I checked:
* you should probably calibrate PID in Klipper instead of copying the values from Marlin
* I am not sure if Marlin uses 1000 or 3000 with those settings.
* did you check, that you used the values in the EEPROM instead of those in Configuration.h (because my X5S patches enable the EEPROM).

To test if jerk makes the difference you could also try the effect of reducing the jerk settings for Marlin.
This would probably make them more similar than trying to find a junction deviation that matches Marlins jerk. With jerk = 0 and junction deviation = 0 they are exactly equal (if I am not mistaken).

KevinOConnor, 2018-01-27:
On Fri, Jan 26, 2018 at 11:04:10AM -0800, Douglas Hammond wrote:
> @KevinOConnor Here are my configs https://gist.github.com/wizhippo/16cd1dbed1388d8c1e069304c3869219  The image at the bottom has the klipper print on the left.  You can see how the blobs along the [...]

That picture looks to me like too much filament is coming out.  It
does not look to me like a jerk, pressure, acceleration, or
junction_deviation issue.  My first impression is that the
extruder.step_distance needs to be increased.  Or perhaps something is
off on the Z (causing squishing of the layers).  It's also possible
there is some kind of oscillation in the bed or nozzle heater
temperatures causing squishing of the Z - try calibrating and using
the PID for both the bed and the nozzle.

I can't explain why you're seeing different results on Marlin.  As
@hg42 said, is it possible Marlin has a different config in eeprom?
Also, what speed did you print the cube at?  What infill percent is it
at?

-Kevin

wizhippo, 2018-01-27:
Infill 40%, 80mm/s.  I PID tuned. I have increased the e step, used a higher temp and bumped junction deviation up to 0.04   The corners are sharper now.  I still have some surface boogers but I am making progress here thank you.  Just to be clear the prints were ok before but like most of use I'm learning and trying to get the best performance I can.  Thanks for the great work and be so responsive.

closed 2018-01-27 by wizhippo