Klipper3d/klipper #145: Automated MCU restart wont work.

kapperi, 2018-01-27:
After changing config, the restart command wont work.

kapperi, 2018-01-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/1670080/klippy.log)

KevinOConnor, 2018-01-28:
Although this may have a slightly different symptom, it appears to be due to host to micro-controller communication outages and errors.  Closing this in favor of issue #147.

closed 2018-01-28 by KevinOConnor

kapperi, 2018-01-28:
Working after changing restart method to command.

KevinOConnor, 2018-01-29:
On Sun, Jan 28, 2018 at 05:41:27PM +0000, kapperi wrote:
> Working after changing restart method to command.

Ah, yes, the native USB AVR chips need to use a restart method of
'command'.  I've updated the code to use that as the default on those
chips (commit 077a56c2).

Is the board stable now with the shorter USB cable?  (The retransmit
counter in the log is no longer incrementing?)

-Kevin