Klipper3d/klipper #146: Move exceeds maximum extrusion (6.926mm^2 vs 0.640mm^2) Firmware was halted, trying to reconnect.

amaximchuk, 2018-01-27:
Looks as the same issue with #32.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1670093/klippy.log)

KevinOConnor, 2018-01-28:
On Sat, Jan 27, 2018 at 05:20:31AM -0800, amaximchuk wrote:
> Looks as the same issue with #32.
> [klippy.log](https://github.com/KevinOConnor/klipper/files/1670093/klippy.log)

Can you provide the g-code file you were printing at the time?  What
tool did you use to produce the g-code file?

-Kevin

amaximchuk, 2018-01-28:
The first attempt failed, but the second one was normal.
I used Simplify3D to produce g-code.
[bird_bowl.zip](https://github.com/KevinOConnor/klipper/files/1671151/bird_bowl.zip)

KevinOConnor, 2018-01-29:
On Sun, Jan 28, 2018 at 06:22:32AM -0800, amaximchuk wrote:
> The first attempt failed, but the second one was normal.
> I used Simplify3D to produce g-code.

This appears to be a bug in Repetier Server (or at least Repetier
Server is incompatible with Klipper).  The log provided shows Klipper
received:

Read 55661.971765: 'N155733 G1 X110.322 Y114.350 E0.0154*101\nN155732 G1 X110.298 Y113.909 E1.1561*103\n'

Which means that Repetier Server sent line 155733 before it sent line
155732.  Klipper does not honor the ancient (and ineffective) g-code
line numbering scheme.  Klipper must be sent each command in the order
they are intended to be executed in.

-Kevin

amaximchuk, 2018-01-29:
Thank you for the answer! I'll be asking Repetier guys about that.

amaximchuk, 2018-02-11:
I have the same error messages two more times. I asked Repetier guys about that, his answer:

"I'm pretty sure server does not reorder lines. All firmwares would complain a missing line number. I also think it might be tty0tty reordering the 2 lines send.
Only way to be 100% sure is when I have next release running with pipe support.

Line numbers are not ancient btw, they are a very good method for firmwares to communicate errors. How else could a firmware tell which line had a defect. During internal communcation there is of course no chance for com errors so line numbers and resends can be skipped."

First: [klippy.zip](https://github.com/KevinOConnor/klipper/files/1714750/klippy.zip)
Second: [klippy.log](https://github.com/KevinOConnor/klipper/files/1714749/klippy.log)

KevinOConnor, 2018-02-11:
On Sun, Feb 11, 2018 at 07:26:17PM +0000, amaximchuk wrote:
> I have the same error messages two more times. I asked Repetier guys about that, his answer:
> 
> "I'm pretty sure server does not reorder lines. All firmwares would complain a missing line number. I also think it might be tty0tty reordering the 2 lines send.

Okay, it's a tty0tty bug then.

I strongly recommend you stop using tty0tty.  Reordered input lines
can cause all sorts of havok and potentially damage your printer.  A
failed print is the least of your worries.

-Kevin

amaximchuk, 2018-02-12:
Got it. Thank you.

closed 2018-02-12 by amaximchuk