Klipper3d/klipper #147: Timer too close

kapperi, 2018-01-27:
This issue comes on shortly after starting to print. Starts just fine but suddenly stops.

Mcu is sav mk1

[klippy.log](https://github.com/KevinOConnor/klipper/files/1670282/klippy.log)

kapperi, 2018-01-27:
Something to do with this? MCU load seems to be a little bit to high couple of times?
[image]

KevinOConnor, 2018-01-28:
On Sat, Jan 27, 2018 at 09:02:29AM -0800, kapperi wrote:
> This issue comes on shortly after starting to print. Starts just fine but suddenly stops.
> 
> Mcu is sav mk1

A "Timer too close" error is generally indicative of an intermittent
communication failure between micro-controller and host.  And, indeed,
the log backs that up - if you look at the stats messages you will see
there are bursts of retransmits occuring and lost connections between
the host and micro-controller.  I'd double check the USB cable and try
to reduce the electrical noise in the environment.

-Kevin

kapperi, 2018-01-28:
Mcu restart problem was solved with the command method of restart? Can it be?

I'll try to find better cable but Im using cable from external harddisk and I think that it is pretty good but long, It even have these ferrulas to noise reduction.

kapperi, 2018-01-28:
No luck with this problem. Changed cable to 15cm short cable. Even put raspberry behind ups.

Nothing, perhaps somekind internal problem of sav mk1. 

Back to marlin and rattling sound of linear bearings...

kapperi, 2018-01-29:
> Is the board stable now with the shorter USB cable?  (The retransmit
counter in the log is no longer incrementing?)

I copied your answer from another issue... Actually its not stable, Its something to do board fets, I think.

Is there anyway to adjust pwm freguencys?

KevinOConnor, 2018-01-30:
On Mon, Jan 29, 2018 at 07:55:34AM -0800, kapperi wrote:
> > Is the board stable now with the shorter USB cable?  (The retransmit
> counter in the log is no longer incrementing?)
> 
> I copied your answer from another issue... Actually its not stable, Its something to do board fets, I think.
> 
> Is there anyway to adjust pwm freguencys?

The pwm_cycle_time can be used to adjust the heater pwm frequency.

-Kevin

khseal, 2018-01-31:
> No luck with this problem. Changed cable to 15cm short cable. Even put raspberry behind ups.
Do you have a ground in the power supply?
Try to connect the ground devices with a separate wire.
Before these measure an AC voltage between the 3d printer and the device with the Klipper.

kapperi, 2018-02-01:
I actually noticed that my psu is grounded but there was no connection between ground and negative. I did a little loop there. I also changed my power cord from psu to mainboard to shielded one.

I measured 0.05V of Dc voltage between raspberry usb connector and mainboard connector without usb cable... No measureable AC voltage...

Latest log attached...

[klippy.log](https://github.com/KevinOConnor/klipper/files/1687128/klippy.log)

kapperi, 2018-02-01:
.... Now it does not want to give me new log file anymore...

khseal, 2018-02-02:
Do you have a webcam connected to usb?

kapperi, 2018-02-02:
No I dont.

Is there way to slow down usb speed between mcu and raspberry?

Could it be that raspberry usb cant provide enough power to mcu? Altought I looked logs from raspberry that should not be problems with usb power.

KevinOConnor, 2018-02-02:
On Thu, Feb 01, 2018 at 08:44:36PM +0000, kapperi wrote:
> I actually noticed that my psu is grounded but there was no connection between ground and negative. I did a little loop there. I also changed my power cord from psu to mainboard to shielded one.
>
> I [...]

Log shows the same problem - retransmit statistic keeps increasing.
It indicates unreliable mcu to host communication channel.

-Kevin

kapperi, 2018-02-02:
How do you see that from the logs?

kapperi, 2018-02-04:
Apparently there's no fix for this problem, and I ordered Mks Sbase to fix the problem.

The Sav Mk1 stays in old printer eventually and im trying to find the problem with klipper. Ä

closed 2018-02-04 by kapperi