Klipper3d/klipper #148: MCU shutdown: Stepper too far in past

com9, 2018-01-27:
Im getting a Stepper too far in past error message lately. Everything was working quite nice until i added a second Extruder, since then i get this message every print exactly after it travels to the start point und starts extruding.

In Octopie all Axis including Stepper are working.

[klippy.log](https://github.com/KevinOConnor/klipper/files/1670417/klippy.log)

com9, 2018-01-27:
Edit: It´s working if i disable pressure advanced!

KevinOConnor, 2018-01-28:
On Sat, Jan 27, 2018 at 10:45:44AM -0800, com9 wrote:
> Im getting a Stepper too far in past error message lately. Everything was working quite nice until i added a second Extruder, since then i get this message every print exactly after it travels to the start point und starts extruding.
> 
> In Octopie all Axis including Stepper are working.

A pressure_advance of 2.9 does not make sense.  The pressure_advance
is generally set to something between 0.0 and 0.20.  For instructions
on calibrating pressure advance, see:

https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md

The hint with the error message ("this generally occurs when the
micro-controller has been requested to step at a rate higher than it
is capable of obtaining") appears correct - the massive amount of
movement that a pressure_advance of 2.9 would require is more steps
than the mcu can accomplish.

-Kevin

com9, 2018-01-28:
I know that there is a problem with my Extruder. I guess my Extruder mount isn´t stiff enough and acts like a spring when Retracting/Extruding. Whats weirdis, that it worked almost two weeks until this message came, thats why i thought it may be a bug.

closed 2018-02-02 by KevinOConnor