Klipper3d/klipper #149: PAUSE or FILAMENT_CHANGE implementation (M600)

LemarinelNet, 2018-01-29:
Hi,

I was about to enable filament change on my marlin firmware, and just discovered klipper firmware !
I installed it this morning, I now have to test it !

So I was wondering if the firmware does implement M600 and it seems it's not yet there.

I'm willing to help implementing it: I know python, GCODE, but need some guidelines to help me decide correctly where it has to be setup.

So the goal is to have a M600 gcode working as in Marlin : you can call M600 without parameters , or with some parameters. It will stop extruding, move the head to a "parking position", and wait for the user to change the filament.
Once the filament has been changed, the user has to push a button to resume the print. Since we don't have a physical UI, we could wait for a special GCODE to resume the print (surely a MXXX code !)

What about this ?

KevinOConnor, 2018-01-30:
On Mon, Jan 29, 2018 at 02:16:42AM -0800, LemarinelNet wrote:
> Hi,
>
> I was about to enable filament change on my marlin firmware, and just discovered klipper firmware !
> I installed it this morning, I now have to test it !
>
> So I was wondering if the firmware does [...]

I'd recommend using the existing M0 logic (which is implemented in
OctoPrint and likely others).  It will cause OctoPrint to pause the
print, allow the user to make a modification, and then allow the user
to resume the print via the OctoPrint continue button.

We could add an alias capability to Klipper so that a single command
(eg, "PARK_MY_PRINT_HEAD") will run a series of g-code commands.  That
may help reduce the complexity of specifying long g-code parking
commands within the slicer.

-Kevin

LemarinelNet, 2018-01-30:
I'm half ok with you ;)
the problem with the "PARK_MY_PRINT_HEAD" macro, is that we must keep a track of where the head comes from, to go back to this particular position, else we will miss a segment !

In facts, I've not yet tried M0 in octoprint...

KevinOConnor, 2018-01-30:
On Tue, Jan 30, 2018 at 04:11:00PM +0000, LemarinelNet wrote:
> I'm half ok with you ;)
> the problem with the "PARK_MY_PRINT_HEAD" macro, is that we must keep a track of where the head comes from, to go back to this particular position, else we will miss a segment !

Ah, okay.  I'm not sure I fully understand how you use M600.  How does
the command get into the g-code stream?  Can you describe a real world
example where you would use the functionality and what the steps would
be?

-Kevin

LemarinelNet, 2018-01-31:
The M600 GCode is used to change filament while printing, either because of end-of-filament, or because the user want to change color.
When M600 code is met, the process is : 
- Save the actual axis positions (x,y,z, extruder)
- Retract some filament (default "park retract" value, or specified with M600)
- Go to a "park nozzle position"
- Wait for the user to change the filament. On MARLIN, this implies that the user have a LCD display, to be able to see what's going on, and confirm that the new filament has been set-up. Once the user has changed the filament, he has to push the button... (in our case, we could wait for a MXXX code to resume the print)
- Reload the filament (either a specified value or default...)
- Go back to the saved position and resume the print !

This is in facts an advanced pause gcode, that moves the head away the print so it is not damaged while waiting for the filament change, and then set it back to its original position !

TunaLatte, 2018-01-31:
If you pause, you can jog the axis around, Octoprint paus saves position, so when you resume regardless of what moves you made it will resume in the correct position (including z axis).

But I would agree on implementing it for one reason which is: extrusion pressure. Having to do it manually would be bit difficult. M600 extrudes a bit while moving to the resume location, else you will have gaps on ~0.3mm layer heights etc.

khseal, 2018-01-31:
The function can be useful if there are any problems with the filament and you need to quickly replace or fix the problem.

KevinOConnor, 2018-02-25:
I'm going to close this issue as it doesn't appear anyone is actively working on developing it.

closed 2018-02-25 by KevinOConnor

derbengale, 2018-11-26:
that would be great 👍

NedalLive, 2018-12-02:
+1 vote for adding pause management for filament replacement :-)

m4gn3to, 2018-12-04:
Another vote, that would be great. So we can "easily" change colors while printing... Thanks for the great work.

Foxbiker, 2019-01-02:
Hi guys, I´m looking that this issue is already close but without some solve. I made some research and find the way how to run multicolor prints very similar way like M600 but without M600. First step is installing of Octorpint plug-in Multi Colors:

https://plugins.octoprint.org/plugins/multi_colors/

Next step is make you own gcode for filament change with using M0. This is my tested code for 750mm long bowden extruder setup and contain: retraction, swipe nozzle, move nozzle away, unload filament, beep, pause, load filament, clean nozzle, retraction, back to position, de-retraction, continuing:

; FILAMENT CHANGE
M117 Color changing...
G92 E0
G1 E-2.5 F3000
G92 E0
G91
G1 X+60 Y-60 F6000
G1 Z+20 F480
G90
G1 E-760 F3000
G92 E0
M300
M117 Change filament
M0
M400
G1 E+750 F3000
G1 E+800 F180
G1 E+797 F3000
G92 E0
G91
G1 X-60 Y+60 F6000
G1 Z-20 F480
G90
G1 E+3 F3000
M117 Printing...
; PRINT CONTINUE

Next is use the code in Multi Colors plug-in setup. Find line (regular expression) set on - G1 Z{layer}.*?

Now you are able to make gcode multicolor modification by setting layer or layers heights and one click.

Whole this setup running perfect only one think is important to keep on you mind. Moving of the nozzle to change position is fix and in some cases it can make some troubles.

bumbalu, 2019-02-17:
What would happen if we insert @Pause command in the GCODE? Will Octoprint understand it and send Klipper command to pause the print? And what is bettter: to use @Pause command for Octoprint or PAUSE command for Klipper?

Foxbiker, 2019-02-18:
When insert Pause command to the gcode Octoprint recognize command, pause the print and wait for resume or cancel. If you send Pause command directly to the Klipper nothings happens. Klipper don't know this command. From these reason is not usable macro function in Klipper for this kind of operations. So here is only one way, use pause command for Octoprint only and be faster like the Steppers switch off delay timer in Klipper.

MillerGregor, 2019-04-24:
@Foxbiker 's gcode worked well for me (thanks!).

I got an error:  `extrude only move too long`
I had to change my config to allow for such long extruder-only commands.

In **printer.cfg**, under the `[extruder]` seciton, add:
```
max_extrude_only_distance: 800.0
#Maximum length (in mm of raw filament) that a retraction or
#extrude-only move may have. If a retraction or extrude-only move
#requests a distance greater than this value it will cause an error
#to be returned. The default is 50mm.
```

totalitarian, 2019-05-05:
Just tried the M600 macro and get this error

`Unknown command:"PAUSE"`

This is the macro

```
[gcode_macro M600]
default_parameter_X: 50
default_parameter_Y: 0
default_parameter_Z: 50
gcode:
	PAUSE
	G91
	G1 E-.8 F2700
	G1 Z{Z}
	G90
	G1 X{X} Y{Y} F3000
	G91
	G1 E-50 F1000
	G90
```

mjedlinski, 2019-05-05:
You must add:[pause_resume]
#recover_velocity: 50.
to your printer.cfg

totalitarian, 2019-05-05:
> You must add:
> [pause_resume]
> #recover_velocity: 50.
> to your printer.cfg

So it's klipper capturing the position rather than octoprint?

spiderdab, 2021-02-04:
Hi, I'm quite new to Klipper, but had success in installing and configuring on a semi-custom delta.
Trying to setup M600 to change filament in case of run-out, I put the suggested Macro in the config file setting my personal change position as this:
`# M600 Filament change:
[pause_resume]

[gcode_macro M600]
default_parameter_X: 50
default_parameter_Y: -80
default_parameter_Z: 50
gcode:
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    G91
    G1 E-.8 F2700
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
    G91
    G1 E-50 F1000
    RESTORE_GCODE_STATE NAME=M600_state`
but it doesn't save the Z position. only X and Y, so it goes on printing at +50 Z until next Z Code.
How can I save also the Z position?

MarcinKnD, 2021-02-22:
It remembers Z's position

If you are using Octoprint add in the settings, in the After_resume Gcode section

RESUME

then when you resume printing on octoprint, it will return to printing nicely