Klipper3d/klipper #151: SPI config support

mentaluproar, 2018-01-30:
Is it possible to make Klipper talk to the SPI connectors on the TMC2130 so they can be configured?

KevinOConnor, 2018-01-30:
On Tue, Jan 30, 2018 at 07:58:42PM +0000, mentaluproar wrote:
> Is it possible to make Klipper talk to the SPI connectors on the TMC2130 so they can be configured?

It's possible to send some basic SPI messages with a little bit of
python code - see the klippy/extras/ad5206.py for an example.

Do you have a particular board in mind?  Do you have a link to the
schematics for that board (that shows all the pin mappings to the
tmc2130 driver)?

-Kevin

andrea7376, 2018-01-30:
https://hackaday.com/2016/09/30/3d-printering-trinamic-tmc2130-stepper-motor-drivers-shifting-the-gears/

If you look at this link you can find very useful information and link to some C library, please will be a dream, I' m waiting for five of this.

mentaluproar, 2018-01-30:
I’m looking at using an MKS gen 1.4 board. While I think it would be best to communicate with them through the board, we could probably do it via GPIO on a pi too.

Here’s more on that board if it helps.

mentaluproar, 2018-01-30:
Forgot the link: http://reprap.org/wiki/MKS_GEN

archi, 2018-02-02:
I think @cruwaller is doing some stuff in that direction. But I don't know if/when that gets ready to merge.
See for example this commit to his lpc176x branch, in which he mentions sensorless homing:
https://github.com/cruwaller/klipper/commit/c6d8e0fccf4632dfd301752090f81a6240cb35e3

(But seems like something to look forward to!)

NeelD87, 2018-02-02:
@mentaluproar Did you check this link its already added by @cruwaller in his branch.
https://github.com/KevinOConnor/klipper/issues/101

KevinOConnor, 2018-02-08:
Closing this in favor of the similar issue #101

closed 2018-02-08 by KevinOConnor