Klipper3d/klipper #153: MCU shutdown after sending command

mentaluproar, 2018-01-31:
Any time I send a command to the machine, klipper throws a big tantrum.  The thermistors are at room temperature and reading correctly.  I cant move steppers, check endstops, anything.  I'm running a corexy machine on an MKS gen 1.4.

Send: QUERY_ENDSTOPS
Recv: // MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: !! Printer is shutdown
Changing monitoring state from 'Operational' to 'Error:  Printer is shutdown

KenshiHH, 2018-01-31:
Read this
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

always upload the log file on an issue

mentaluproar, 2018-01-31:
Sorry about that.  Log uploaded.

KevinOConnor, 2018-01-31:
You need to issue a FIRMWARE_RESTART command (navigate to the OctoPrint terminal tab and type that command into the terminal).  That command is needed to fully reset the micro-controller and host code.  If the problem persists after issuing a FIRMWARE_RESTART then please attach an updated log at that point.  (We need the log from the session that resulted in the error - the log provided is from a subsequent session and it only reports that the micro-controller error has yet to be cleared.)

mentaluproar, 2018-01-31:
The issue survives restarts.  I tested with repetitier and it works so it’s not a hardware issue. 

I’ll do it again tomorrow and post. Any particular action you want to issue? It sends sensor data to me but if I send anything to it, it loses it’s mind.

mentaluproar, 2018-01-31:
Well what do you know, today it works!  After several restarts, it still works fine.  I notice that ever error will cause a disconnect. It's annoying, but I don't see myself resetting this once it gets started.

closed 2018-02-02 by KevinOConnor