Klipper3d/klipper #154: Will it support SCARA？

skylli, 2018-01-31:
I  wan build scara use klipper, and there scara in klipper todo list. When will scara start ?

KevinOConnor, 2018-02-02:
On Wed, Jan 31, 2018 at 07:28:34AM +0000, skylli wrote:
> I  wan build scara use klipper, and there scara in klipper todo list. When will scara start ? 

I don't know of anyone working on SCARA.  I don't have any immediate
plans to implement it.

-Kevin

closed 2018-02-25 by KevinOConnor

evenmelnik, 2019-05-24:
I am interested in the SCARA theme, they have a number of options, but their popularity is not ensured by the lack of good firmware

evenmelnik, 2019-05-24:
Do you have additional documentation for adding new kinematics?

KevinOConnor, 2019-05-24:
See the documentation at https://www.klipper3d.org/Code_Overview.html#adding-new-kinematics .  Also note that initial support for polar kinematics have been added (commit ec9cb3a1).

-Kevin

KevinOConnor, 2019-10-25:
If someone has a scara machine and is willing to run tests, I may be able to help with the kinematic implementation.  If interested, let me know.

-Kevin

antst, 2019-11-23:
I am assembling scara. But it will take most likely 2 weeks before I will last missing bit.

KevinOConnor, 2019-11-24:
FYI, I looked at the low-level step generation math for scara and it is very similar to the math for the rotary delta kinematics (currently on the work-rotary-20191119 branch).

-Kevin

Chleba, 2020-02-23:
I have MPScara with RAMPS 1.4 on Arduino Mega, so I could test something for you guys. I'm just not familiar with Klipper, but I have some time to do some testing for you guys.

KevinOConnor, 2020-02-24:
If you're up for running tests, start by installing Klipper on the machine (use cartesian kinematics to start).  Run the checks described at https://www.klipper3d.org/Config_checks.html .  Then attach the resulting Klipper log file here.

It may also be able to print using the `Gcode_Translator.py` script that comes with the MPScara software.

Once you have that working, we can try tests with scara kinematics implemented in Klipper.

-Kevin