Klipper3d/klipper #155: Can cause a fire!

ffnull, 2018-01-31:
Printer:Prusa i3
Board: Anet a8
While starting printing. Watchdog didnt work.
My Hotend burnt. Over 400 celsius. There were fire
[image]

KevinOConnor, 2018-01-31:
On Wed, Jan 31, 2018 at 06:55:19PM +0000, ffnull wrote:
> Printer:Prusa i3
> Board: Anet a8
> While starting printing. Watchdog didnt work.
> My Hotend burnt. Over 400 celsius. There were fire

If you want us to take a look at it, you need to attach the klipper
log file from the incident.  See:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

ffnull, 2018-01-31:
The problem that when its fire. Im cut off power. Log is gone. Sorry

NickRimmer, 2018-01-31:
I think we need some protections for temperature in Host and MCU both.

jakep82, 2018-01-31:
This looks like it was caused by a thermistor, wiring, or board issue.  The temperature was bouncing all over the place before instantly jumping from 150C to 450C.  From there the temperature quickly drops which is probably when the MCU shut down because it was way past the max temperature.  This should probably be addressed with a safety shutoff if the hotend doesn't have a reasonable temperature rise during heat up.

Papsnoek, 2018-02-01:
This does not look like Klipper's fault. Temperature cannot Rise from 150
to 400 instantly... So this would mean the Temperature is faulty and not
reporting the correct temperatures to Klipper.

I think you also might be over reacting about the fire part. According to
the Graph, the Temperature was kept at a stable 240 degrees for over 10
minutes AFTER the fire moment...

Only thing that should come out of this is as jakep82 says, The Shutdown
works fine as this did happen, but maybe some added protection that if
heater is on full power and temperature doesn't rise by a few degrees each
second, then shutdown as this would mean temperature sensor is not
reporting back the correct Temps.

ffnull, 2018-02-01:
Shutdown doest work. Im shut down board manualy. Then were 240 for 10 minutes becouse im dont reflash page. Im starting heating again after replace hot barier and nozle. Quick dropdown were after shutting off 12 power and manual cooling

KevinOConnor, 2018-02-01:
On Wed, Jan 31, 2018 at 12:15:05PM -0800, jakep82 wrote:
> This looks like it was caused by a thermistor, wiring, or board issue.  The temperature was bouncing all over the place before instantly jumping from 150C to 450C.  From there the temperature [...]

It's hard to tell without a log, but I agree with your analysis.

-Kevin

KevinOConnor, 2018-02-01:
On Thu, Feb 01, 2018 at 10:25:45AM +0000, ffnull wrote:
> Then were 240 for 10 minutes becouse im dont reflash page.

The klipper log file is stored on the host machine.  If you didn't
power off the host machine, then the log should still be present in
/tmp/

-Kevin

ffnull, 2018-02-01:
Please,read again. No log bicouse im shutting off board. Graphic there becouse im doesnt refrash page. No problems with thermocuple end board. Swithch back to marlin and work good

TunaLatte, 2018-02-02:
I had this type of curve before when the X axis (extruder) was in homing position, it turned out the internal cable was damaged on frequent position (especially homing position) but didn't while printing. I would avoid pretinned wires, solid core copper (like ethernet cables), and recommend twisting the entire cable, would also help against stepper back-emf. The cable bend radius should be minimal when the head moves (especially homing) twisting will also help against sharp bends.

psy0rz, 2018-02-08:
just close this post. one should always make sure the printer isn't near burnable materials. (or just never leave your printer alone)

TunaLatte, 2018-02-08:
@psy0rz agreed, but having thermal runway detection could still help bring down the cost of replacing the entire extruder assembly maybe? especially on hobby printers, e3d heatblock sensor cartidge ceramic bead holder is known to crack after extended periods for example (due to metal contraction) and using a zip tie can pull the sensor slowly overtime, always keep a slack for the wire etc, its still helpful to detect, especially for people that use cork or similar material under heatbed.

wizhippo, 2018-02-08:
Thermal runway detection would be great but if we set the max temps for both the heatbed and the hotend should they not ever be exceeded?  So if I set my max_temp to 210 under extruder and I hit 400 should it not have shutdown?

jakep82, 2018-02-08:
It will only shut down if it knows it's above 210.  In this case the MCU didn't know it was above the max temp until it hit 400, or whatever temperature it actually reached.  This is where minimum thermal rise requirements could help prevent damage.

ffnull, 2018-02-08:
After 400 
[image]

KevinOConnor, 2018-02-09:
On Thu, Feb 08, 2018 at 09:24:18PM +0000, jakep82 wrote:
> It will only shut down if it knows it's above 210.  In this case the MCU didn't know it was above the max temp until it hit 400, or whatever temperature it actually reached.  This is where minimum thermal rise requirements could help prevent damage.

Agreed.

FYI, I've been working on "first order plus dead time" calibration
code.  My plan is to implement thermistor checks around this fopdt
model.

-Kevin

unverbraucht, 2018-03-01:
To be totally honest thermal runway protection is what's been keeping me from using this for my Anet. These acryllic printers with mediocre connectors and wiring are fire hazards anyway, and this would give me some peace of mind 👍

KevinOConnor, 2018-03-11:
FYI, commit 5208fc38 adds basic heater and thermistor verification.

closed 2018-03-18 by KevinOConnor