Klipper3d/klipper #159: Error Unknown pin chip name 'replicape' when starting Klipper

sapell, 2018-02-03:
I had a halted print issue (no log of that) and decided to update Klipper after that by issuing 'git pull'.
After updating I get the error ' Unknown pin chip name 'replicape' ' when launching Klipper.
I updated from Jan 12th 2018 commit to the latest and in between Replicape has been moved from chipmisc.py to extras/replicape.py. Is there something that should be changed in config due to that? Could not right away see any difference in the example config for Replicape.

Log:
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1691922/klippy.log.gz)

KevinOConnor, 2018-02-03:
On Sat, Feb 03, 2018 at 08:27:30AM -0800, Simo Apell wrote:
> I had a halted print issue (no log of that) and decided to update Klipper after that by issuing 'git pull'.
> After updating I get the error ' Unknown pin chip name 'replicape' ' when launching [...]

Yes - the order of the config became important for some sections.  Any
section that defines a pin should now be present in the config above
any section that uses that pin.  I updated the generic-replicape.cfg
example config accordingly.

I don't currently have a good way to announce config file changes like
this.  I'm not sure what's the best way to do that.

-Kevin

sapell, 2018-02-03:
Thanks Kevin! Once again incredibly quick response and problem solved!

closed 2018-02-03 by sapell